Koenkk/zigbee2mqtt #20716: Lidl Switch with power meter model HG08673 only reports 0 current and 0 power

MathijsG, 2024-01-10:
### What happened?

I added a newly bought Lidl power switch, whom I know the models are supported now in z2m, however when I looked carefully today I saw that current and power keep being zero.

Only the total used sum is calculated towards the energy property.

[image]
[image]

I own two of these newly bought devices, and both act the same. If I look at an earlier issue it should be compatible (#14356)

Has something changed maybe that it doesn't work anymore?

### What did you expect to happen?

I expected the power and current consumption to also be reported, as they are supposed to be supported if I look at the compatibility page.

### Zigbee2MQTT version

 1.35.1

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Add-on within a virtual Home Assistant VM within Proxmox

iPAL45-80, 2024-01-11:
I have the same problem

RoadRunner-, 2024-01-15:
Same problem here with exact the same model. Same version of Zigbee2MQTT. Coordinator type is zStack3x0.

I updated the switch via OTA update to the latest version before using it. So i can't say if it worked with the delivered version.

```
Received Zigbee message from 'PM_PLUG01', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID 0

MQTT publish: topic 'zigbee2mqtt/PM_PLUG01', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":93,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":232}'

Received Zigbee message from 'PM_PLUG01', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0

Received Zigbee message from 'PM_PLUG01', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
```

Please let me know, if you need more details to analyse the problem.

RoadRunner-, 2024-02-07:
With the latest version (Zigbee2MQTT-Version: 1.35.3 commit: fe0742a) this issue is solved for me.

Thank you!

MathijsG, 2024-02-12:
They seem to work normal here as well.

fgeck, 2024-04-19:
I am still running into that issue with my 2 switches (Model: 
HG08673, Manufacturer: Lidl).  They are only reporting Voltage and Energy.

Used Zigbee2mqtt version is 1.36.1-1
And OTA says there are no new updates.

```json
{
    "child_lock": "UNLOCK",
    "current": 0,
    "energy": 0,
    "indicator_mode": "off/on",
    "linkquality": 99,
    "power": 0,
    "power_outage_memory": "off",
    "state": "ON",
    "update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },
    "update_available": false,
    "voltage": 233
}
```

```json
{
    "child_lock": "UNLOCK",
    "current": 0,
    "energy": 9.63,
    "indicator_mode": "off/on",
    "linkquality": 42,
    "power": 0,
    "power_outage_memory": "off",
    "state": "OFF",
    "update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },
    "update_available": false,
    "voltage": 234
}
```

Do you have any ideas how to further debug?