Koenkk/zigbee2mqtt #20718: Home Assistant: zigbee2mqtt + mosquitto MQTT failed to connect: connect ECONNREFUSED 127.0.0.1:1883

speedAmaster, 2024-01-10:
### What happened?

I am using Home Assistant OS (2024.1.2 - OS11.3) with both add-ons for Mosquitto (6.4.0) and zigbee2mqtt (1.35.1-1).

Home Assistant and Mosquitto are working greatly since months.
Trying to setup zigbee2mqtt with SMLIGHT SLZB-06 LAN, I achieved to connect to the device, but MQTT is not able to get connected! Both add-ons are running obviously on the same Home Assistant machine.

I am changing zigbee2mqtt/configuration.yaml directly (not via GUI), do a restart of Home Assistant after each change.

I tried server: 'mqtt://172.30.32.2' server: 'mqtt://172.30.32.2:1883' server: 'mqtt://localhost:1883' server: 'mqtt://homeassistant.fritz.box:1883' server: 'mqtt://:1883'

Strangely I always get a "MQTT error: connect ECONNREFUSED 127.0.0.1:1883"

I am at the end of my ideas and googling to find a resolution - I need an expert :-)

Any ideas?

### What did you expect to happen?

working MQTT connection to a MQTT broker on the same IP address

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

0.9.9

### Adapter

SMlight  	SLZB-06   

### Setup

Home Assistant OS on Proxmox 2024.1.2 - OS11.3

### Debug log

My current configuration.yaml looks like:

    homeassistant: true
    socat:
    enabled: false
    master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
    slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
    options: "-d -d"
    log: false
    mqtt:
    server: 'mqtt://172.30.32.2'
    base_topic: zigbee2mqtt
    user: mqtt
    password: ....
    client_id: 'MY_CLIENT_ID'
    serial:
    port: tcp://zigbee.fritz.box:6638
    baudrate: 115200
    disable_led: false
    advanced:
    transmit_power: 20
    frontend:
    port: 8099

zigbee2mqtt log:

    [11:21:37] INFO: Preparing to start...
    [11:21:37] INFO: Socat not enabled
    [11:21:37] INFO: Starting Zigbee2MQTT...
    Zigbee2MQTT:info 2024-01-09 11:21:39: Logging to console and directory: '/config/zigbee2mqtt/log/2024-01-09.11-21-39' filename: log.txt
    Zigbee2MQTT:info 2024-01-09 11:21:39: Starting Zigbee2MQTT version 1.35.1 (commit #unknown)
    Zigbee2MQTT:info 2024-01-09 11:21:39: Starting zigbee-herdsman (0.30.0)
    Zigbee2MQTT:info 2024-01-09 11:21:41: zigbee-herdsman started (resumed)
    Zigbee2MQTT:info 2024-01-09 11:21:41: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
    Zigbee2MQTT:info 2024-01-09 11:21:41: Set transmit power to '20'
    Zigbee2MQTT:info 2024-01-09 11:21:41: Currently 0 devices are joined:
    Zigbee2MQTT:info 2024-01-09 11:21:41: Zigbee: disabling joining new devices.
    Zigbee2MQTT:info 2024-01-09 11:21:41: Connecting to MQTT server at mqtt://localhost:1883
    Zigbee2MQTT:error 2024-01-09 11:21:41: MQTT error: connect ECONNREFUSED 127.0.0.1:1883
    Zigbee2MQTT:error 2024-01-09 11:21:41: MQTT failed to connect, exiting...
    Zigbee2MQTT:info 2024-01-09 11:21:41: Stopping zigbee-herdsman...
    Zigbee2MQTT:error 2024-01-09 11:21:42: MQTT error: connect ECONNREFUSED 127.0.0.1:1883
    Zigbee2MQTT:error 2024-01-09 11:21:43: MQTT error: connect ECONNREFUSED 127.0.0.1:1883
    Zigbee2MQTT:error 2024-01-09 11:21:44: MQTT error: connect ECONNREFUSED 127.0.0.1:1883
    Zigbee2MQTT:error 2024-01-09 11:21:45: MQTT error: connect ECONNREFUSED 127.0.0.1:1883
    Zigbee2MQTT:error 2024-01-09 11:21:46: MQTT error: connect ECONNREFUSED 127.0.0.1:1883
    Zigbee2MQTT:error 2024-01-09 11:21:47: MQTT error: connect ECONNREFUSED 127.0.0.1:1883
    Zigbee2MQTT:error 2024-01-09 11:21:48: MQTT error: connect ECONNREFUSED 127.0.0.1:1883
    Zigbee2MQTT:error 2024-01-09 11:21:49: MQTT error: connect ECONNREFUSED 127.0.0.1:1883
    Zigbee2MQTT:error 2024-01-09 11:21:50: MQTT error: connect ECONNREFUSED 127.0.0.1:1883
    Zigbee2MQTT:info 2024-01-09 11:21:51: Stopped zigbee-herdsman

In Mosquitto log I see
    2024-01-09 11:21:14: New connection from 192.168.178.107:58457 on port 1883.
    2024-01-09 11:21:14: New client connected from 192.168.178.107:58457 as 2JmMD77KAcHdUWcnxSArdg (p2, c1, k60, u'mqtt').
    2024-01-09 11:22:47: New connection from 172.30.32.2:60480 on port 1883.
    2024-01-09 11:22:47: Client closed its connection.
    2024-01-09 11:24:47: New connection from 172.30.32.2:49888 on port 1883.
    2024-01-09 11:24:47: Client closed its connection.

danikong, 2024-01-10:
try setting your server to `mqtt://core-mosquitto:1883`
Had the exact same issue and came across this post https://community.home-assistant.io/t/mqtt-error-connect-econnrefused-127-0-0-1-1883/555500/3
which solved it for me

speedAmaster, 2024-01-10:
@danikong : you are genius - issue resolved - never came accross that configuration!

closed 2024-01-10 by speedAmaster

nivek099, 2024-07-18:
> try setting your server to `mqtt://core-mosquitto:1883` Had the exact same issue and came across this post https://community.home-assistant.io/t/mqtt-error-connect-econnrefused-127-0-0-1-1883/555500/3 which solved it for me

THANK YOU!!! 

I had the same issue and this immediately resolved it