Koenkk/zigbee2mqtt #20720: TV02-Zigbee clock is getting out of sync

wlcrs, 2024-01-10:
### What happened?

My TRV with id clock is getting out of sync. 

My TRV is identified by Z2M as "[TV02-Zigbee](https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee)" with its Zigbee manufacturer id `_TZE200_hue3yfsn`

### What did you expect to happen?

Looking at the code I would expect this to already be fixed with https://github.com/Koenkk/zigbee-herdsman-converters/pull/2124 .

Looking in the logs, I'm seeing the following suspicious log lines:

> debug 2024-01-10 10:57:00: Received Zigbee message from 'TRV_Bureau', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":52268}' from endpoint 1 with groupID 0
> debug 2024-01-10 10:57:00: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":52268}'

This seems to indicate that this clock sync message is not understood?

I've been able to get the following traffic logs. Do the lines in bold mean that an answer is sent back to the device? If so, then maybe `fz.ignore_tuya_set_time` must be added to the device config? (disclaimer: It's not exactly clear to me what the relation is between `onEvent` and the function producing the 'No converter available' log line. )

> 2024-01-10T14:00:27.384Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
> 2024-01-10T14:00:30.617Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,0,239,249,148,1,1,0,90,0,142,157,128,0,0,5,25,106,36,44,72,249,148,29,209]
> 2024-01-10T14:00:30.618Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,0,239,249,148,1,1,0,90,0,142,157,128,0,0,5,25,106,36,44,72,249,148,29,209]
> 2024-01-10T14:00:30.618Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,0,239,249,148,1,1,0,90,0,142,157,128,0,0,5,25,106,36,44,72,249,148,29] - 209
> 2024-01-10T14:00:30.618Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":38137,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":90,"securityuse":0,"timestamp":8428942,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[25,106,36,44,72]}}
> 2024-01-10T14:00:30.619Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":106,"manufacturerCode":null,"commandIdentifier":36},"Payload":{"payloadSize":18476},"Command":{"ID":36,"parameters":[{"name":"payloadSize","type":33}],"name":"mcuSyncTime"}},"address":38137,"endpoint":1,"linkquality":90,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
> 2024-01-10T14:00:30.620Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
> Zigbee2MQTT:debug 2024-01-10 15:00:30: Received Zigbee message from 'TRV_Badkamer', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":18476}' from endpoint 1 with groupID 0
> 2024-01-10T14:00:30.624Z zigbee-herdsman:controller:endpoint Command 0xb4e3f9fffe0a1197/1 manuSpecificTuya.mcuSyncTime({"payloadSize":8,"payload":[101,158,162,255,101,158,177,15]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
> 2024-01-10T14:00:30.624Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xb4e3f9fffe0a1197:38137/1 (0,0,1)
> 2024-01-10T14:00:30.625Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":38137,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":39,"options":0,"radius":30,"len":13,"data":{"type":"Buffer","data":[1,25,36,8,0,101,158,162,255,101,158,177,15]}}
> Zigbee2MQTT:debug 2024-01-10 15:00:30: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":18476}'
> **2024-01-10T14:00:30.625Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,1,249,148,1,1,0,239,39,0,30,13,1,25,36,8,0,101,158,162,255,101,158,177,15,83]**
> 2024-01-10T14:00:30.633Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
> 2024-01-10T14:00:30.633Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
> 2024-01-10T14:00:30.633Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
> 2024-01-10T14:00:30.633Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
> 2024-01-10T14:00:30.633Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
> 2024-01-10T14:00:30.721Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,39,225]

### How to reproduce it (minimal and precise)

Wait for the TRV to send an 'commandMcuSyncTime' message

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

'coordinator revision' not visible in ' frontend -> settings -> about' ?

### Adapter

SONOFF ZBDongle-P with zStack3x0 20210708

### Setup

Add-on on Home Assistant OS on x86-64. HA runs inside proxmox VM

### Debug log

Log from `/config/zigbee2mqtt/logs`: [filtered-log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13889485/filtered-log.txt)

Logs from the Z2M docker container, captured and filtered as follows: `docker logs 71 2>@1 | grep -B 25 -A 25 mcu`: [traffic-logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/13889667/traffic-logs.txt)

Note that I already tried to debug the situation by changing the device config to `onEvent: tuya.onEvent({timeStart: '1970'}),` but the same errors keeps appearing in the logs.

I'm including the "external converter" I made for completeness (but only 'onEvent' changed compared to the current production code): [trv.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/13889511/trv.js.txt)

n1tehawk, 2024-01-24:
I have three Zigbee TRVs also identified as "TV02-Zigbee" that show the same behavior. They're **VALE TV05-ZG** that look and identify (manufacturer *_TZE200_mudxchsu*, model *TV0601*) the same as the [Zonnsmart model](https://zigbee.blakadder.com/Zonnsmart_TV05-ZG.html).

I can confirm these messages occur in 1-hour intervals. An interesting observation is that the payload size increases between messages, in increments of 256 bytes. (Can also be seen in *wlcrs*' filtered-log.txt.) I suspect that this could indicate that the TRVs are awaiting some kind of confirmation / actual "setting" that never occurs. I had experimented with 'local' control of (eco/comfort) temperature via week schedule at some point in the past, but wasn't able to get it working in any meaningful way.
___

Zigbee2MQTT version: 1.35.1 (commit: 3c962042)
Adapter and firmware version: Sonoff ZBDongle-E, NCP 7.3.2.0 build 212 (EZSP v12)
Setup: plain on Debian 12 inside a LXC container

```
debug 2024-01-24 05:10:56: Received Zigbee message from 'Room_2', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":55562}' from endpoint 1 with groupID 0
debug 2024-01-24 05:10:56: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":55562}'
debug 2024-01-24 05:33:57: Received Zigbee message from 'Room_3', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":31757}' from endpoint 1 with groupID 0
debug 2024-01-24 05:33:57: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":31757}'
debug 2024-01-24 05:48:55: Received Zigbee message from 'Room_1', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":6923}' from endpoint 1 with groupID 0
debug 2024-01-24 05:48:55: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":6923}'
debug 2024-01-24 06:11:28: Received Zigbee message from 'Room_2', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":55818}' from endpoint 1 with groupID 0
debug 2024-01-24 06:11:28: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":55818}'
debug 2024-01-24 06:34:31: Received Zigbee message from 'Room_3', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":32013}' from endpoint 1 with groupID 0
debug 2024-01-24 06:34:31: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":32013}'
debug 2024-01-24 06:49:23: Received Zigbee message from 'Room_1', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":7435}' from endpoint 1 with groupID 0
debug 2024-01-24 06:49:23: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":7435}'
debug 2024-01-24 07:12:01: Received Zigbee message from 'Room_2', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":56330}' from endpoint 1 with groupID 0
debug 2024-01-24 07:12:01: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":56330}'
debug 2024-01-24 07:35:05: Received Zigbee message from 'Room_3', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":32525}' from endpoint 1 with groupID 0
debug 2024-01-24 07:35:05: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":32525}'
debug 2024-01-24 07:49:51: Received Zigbee message from 'Room_1', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":7947}' from endpoint 1 with groupID 0
debug 2024-01-24 07:49:51: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":7947}'
debug 2024-01-24 08:12:33: Received Zigbee message from 'Room_2', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":56586}' from endpoint 1 with groupID 0
debug 2024-01-24 08:12:33: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":56586}'
debug 2024-01-24 08:35:37: Received Zigbee message from 'Room_3', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":32781}' from endpoint 1 with groupID 0
debug 2024-01-24 08:35:37: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":32781}'
debug 2024-01-24 08:50:19: Received Zigbee message from 'Room_1', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":8203}' from endpoint 1 with groupID 0
debug 2024-01-24 08:50:19: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":8203}'
debug 2024-01-24 09:13:04: Received Zigbee message from 'Room_2', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":56842}' from endpoint 1 with groupID 0
debug 2024-01-24 09:13:04: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":56842}'
[... cut ...]
debug 2024-01-24 11:37:16: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":33549}'
debug 2024-01-24 11:43:14: Received Zigbee message from 'Room_1', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":9227}' from endpoint 1 with groupID 0
debug 2024-01-24 11:43:14: No converter available for 'TV02-Zigbee' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":9227}'
```

closed 2024-08-22 by github-actions[bot]

MrManor, 2024-11-25:
I just bought a ZigBee Tuya TRV identified as TS0601 from _TZE200_mudxchsu. I think this has the same problem, as the schedule I have programmed occurs on another time of the day. I guess the time/date in the unit is wrong, which led me to this report. Would you please consider reopening and looking in to this problem?

wlcrs, 2024-11-26:
I'm still facing this problem too, but there is no way for me to reopen this issue.

Please keep me updated if you manage to find a solution.

MrManor, 2024-12-15:
Yeii.. With the latest update version the problem has disappeared: I dont see the "No converter available" message in the log and the Thermostat now changes temperature within a minute from the set time. 
Zigbee2MQTT version: 1.37.1