Koenkk/zigbee2mqtt #20722: Z2M constantly offers an OTA update while the module is up to date.

tadikwa, 2024-01-10:
### What happened?

It was noted that Z2M does not report the correct value when it comes to the firmware version.
Indeed, the latter constantly indicates that the module will be in V1 while it is in V13.
Therefore, home assistant constantly offers the OTA update.

In the "About" tab it correctly displays the firmware value 4000-0013.
On the other hand, in "Status", it indicates the firmware in version 1.

It seems to me that the latter lists the "hwVersion" field instead of the "swBuildId" field

### What did you expect to happen?

The expected behavior is that Z2M considers the Lixee module up to date

### How to reproduce it (minimal and precise)

To reproduce the phenomenon, simply associate the Lixee Zlinky module with a zigate platform so that it is available in Z2M and note in the OTA tab that an update is available, and also check the tab "Status" which states the wrong firmware version.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

a003

### Adapter

LiXee ZLinky_TIC

### Setup

HASS.io on ESX Server

### Debug log

Read result of 'genBasic': {"hwVersion":1,"swBuildId":"4000-0013"}

tadikwa, 2024-01-10:
[image]
[image]
[image]

MilesTEG1, 2024-03-18:
Hello,
I've got a similar problem, except I have a limited firmware version (ZLinky_router_v13_limited.bin), I flashed this WE over the legacy version.
The ZLinly swBuildId is `4001-0013` whereas the hwVersion is `1`.
[image]

And Z2M see an update that can't be done via OTA.

ANd what's strange is that the "About" page doesn't show the right version of the firmware :
[image]
It should be seen as `4001-0013` not `4000-0013`.

Can this be fixed?
And OTA update disable if we have an OTA firmware while OTA for limited firmware isn't supported ?

MilesTEG1, 2024-06-03:
Is someone has been able to fix this ?

closed 2024-12-30 by github-actions[bot]