Koenkk/zigbee2mqtt #20729: [New device support]:  TS0601 _TZE204_3t91nb6k 2 gang switch

VR-VIDEO, 2024-01-11:
### Link

https://www.aliexpress.us/item/3256801436204038.html?spm=a2g0o.order_detail.order_detail_item.3.4eb9f19c6NyHS0&gatewayAdapt=glo2usa

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0xa4c1381d158606d0","nwkAddr":58732,"manufId":4417,"manufName":"_TZE204_3t91nb6k","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"o�1-io�1-i��1-i��1-i�1-i�1-i","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_3t91nb6k","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1704935314878,"defaultSendRequestWhen":"immediate"}

### Comments

I tryed do it by manual, but every time i got error with no define state and   value 0, expected true or false
`Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of true,false}`

i got this in log 
```
Debug 2024-01-10 20:20:32Received Zigbee message from '0xa4c1381d158606d0', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":7}],"seq":40704}' from endpoint 1 with groupID 0
Debug 2024-01-10 20:20:32Received Zigbee message from '0xa4c1381d158606d0', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":8}],"seq":40960}' from endpoint 1 with groupID 0
```
`Debug 2024-01-10 20:22:35Received Zigbee message from '0xa4c1381d158606d0', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":2}],"seq":42240}' from endpoint 1 with groupID 0
Debug 2024-01-10 20:22:35Datapoint 2 not defined for '_TZE204_3t91nb6k' with value true
Debug 2024-01-10 20:22:35Received Zigbee message from '0xa4c1381d158606d0', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":42496}' from endpoint 1 with groupID 0
Debug 2024-01-10 20:22:35Datapoint 1 not defined for '_TZE204_3t91nb6k' with value false
Debug 2024-01-10 20:22:35Received Zigbee message from '0xa4c1381d158606d0', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":42752}' from endpoint 1 with groupID 0
Debug 2024-01-10 20:22:35Datapoint 1 not defined for '_TZE204_3t91nb6k' with value true
Debug 2024-01-10 20:22:36Received Zigbee message from '0xa4c1381d158606d0', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":2}],"seq":43008}' from endpoint 1 with groupID 0
Debug 2024-01-10 20:22:36Datapoint 2 not defined for '_TZE204_3t91nb6k' with value false
Debug 2024-01-10 20:22:37Received Zigbee message from '0xa4c1381d158606d0', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":43264}' from endpoint 1 with groupID 0
Debug 2024-01-10 20:22:37Datapoint 1 not defined for '_TZE204_3t91nb6k' with value false`

### External defintion

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_3t91nb6k',
        },
    ],
    model: 'TS0601_switch_2_gang',
    vendor: 'SHAWADER',
    description: '2 gang switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
    ],
    meta: {
        // All datapoints go in here
        multiEndpoint: true,
        tuyaDatapoints: [
            [7, 'state_l1', tuya.valueConverter.onOff],
            [8, 'state_l2', tuya.valueConverter.onOff],
        ],
    },
    endpoint: (device) => {
        return { 'l1': 1, 'l2': 1 };
    },
};

module.exports = definition;
```

alejovea9, 2024-01-15:
Please, I also need to add this switch, I have several without being able to add them

Koenkk, 2024-01-15:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/1f0463a1a3ec64233560433db11c8587): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

VR-VIDEO, 2024-01-15:
works, can u pls point me what was wrong if u will get a minute.

Koenkk, 2024-01-17:
Integrated the changes, you can remove the external converter after updating to the next release.

closed 2024-01-17 by Koenkk

dems7up, 2026-04-25:
i have a TS0601 _TZE204_3t91nb6k which gets detected, but whenever i press the physical button to turn the light off, home assistant doesn't detect my bulbs anymore.... it's so frustrating, i bought this switch exactly to avoid this

i get this error: `z2m: Failed to configure '0xa4c1383b94cafbbd', attempt 4 (Error: Bind 0xa4c1383b94cafbbd/1 genOnOff from '0xbc026efffeaac04b/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@10.0.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:719:23) at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@26.27.0/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:48:9) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@26.27.0/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:5387:13) at Configure.configure (/app/lib/extension/configure.ts:135:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:45:25) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))`