Koenkk/zigbee2mqtt #20730: Bosch BMCT-SLZ Light/shutter control unit II: Failed to read state after reconnect genOnOff UNSUPPORTED_ATTRIBUTE

L3nny5, 2024-01-11:
### What happened?

I had some issues with my BMCT-SLZ shutter control. So I updated from version 1.35.0 to 1.35.1 (non dev). After the update I removed the BMCT-SLZ, rebooted Z2M and added the BMCT-SLZ via Tools -> Add Install Code. The device immediately connected to Z2M and is working. However, I always get this error message in the front end, which you can also see in the logs. Sometimes mutliple times in a few seconds, sometimes no message in 10-15 min at all.

`Failed to read state of '0xf4b3b1fffe403772' after reconnect (Read 0xf4b3b1fffe403772/1 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))`

So I tried changing over to Z2M Dev to see if the bug fixes maybe fixed it. It seems better (less messages). But I also received these messages a lot during OTA updates, which took ages (IKEA RCs). Guess during that time Z2M is polling a lot of devices at once ?!?

The BMCT-SLZ is fully controllable via Z2M, so there is not really an issue. It is just annoying and is spamming the logs.
However there is one thing which I think is not the way it is supposed to (reason why I initially upgraded to 1.35.1):

You have to configure/calibrate the BMCT-SLZ in order to use the physical Buttons for the shutter, otherwise they're inactive. 
There are two tabs to configure the device: Exposes and Settings(specific). If I just use Exposes the device works just fine. However it is showing a lot of information which is only relevant if you use the BMCT-SLZ as a light control. 
If you set the device to shutter in Settings(specific) a lot of unused settings disappears from Exposes (I guess intended). However it also removes settings which you still need in order to configure the device, that you can use the physical buttons: Device mode and Switch type.
Also, once you set Device mode in Settings(specific) you cannot switch it back to nothing. You can only switch between light and shutter, which does not help configure the device to the correct settings in order to use the physical buttons.

A workaround for the configuration was for me to configure Exposes first and then Settings(specific). Then everything worked with the BMCT-SLZ, but the error message is appearing... 
Also I just noticed that the State changes after some time from STOP to nothing, which wasn't the case with 1.35.0.

**Settings(specific) not configured**
[image]

**Exposes, if Settings(specific) is not configured**
[image]

**Settings(specific) configured**
[image]

**Exposes, if Settings(specific) is configured**
[image]

### What did you expect to happen?

Device is configured correctly:
I can configure Switch type (and Device mode(no idea if it is needed)) also when Device mode in Settings(specific) is configured to shutter.

Also I think configuration for the Switch Type should be in Settings(specific). You don't change it very often.

### How to reproduce it (minimal and precise)

1. Add BMCT-SLZ via Install code - you already get the state read error message at the end of the config
2. Change configuration in Settings(specific) 
3. You cannot set Switch Type in Exposes anymore

### Zigbee2MQTT version

1.35.1-dev

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

Docker: koenkk/zigbee2mqtt:latest-dev

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13896596/log.txt)

closed 2024-02-18 by Koenkk

L3nny5, 2024-02-19:
Thank you! I haven't tested it yet as I'm back on stable. I will test it as soon as this will be released to stable. However after multiple reinstallations of Z2M at some point the UNSUPPORTED_ATTRIBUTE error only appeared once. I also updated the FW of the Bosch BMCT-SLZ with the Bosch hub. Could be related to it as well.
For the configuration, if you set everything before you switch the device mode, it was also usable. But once you changed the mode of the actor you had to remove the device to reconfigure.

ring0r, 2024-02-22:
Hi everyone,

I'm experiencing the same issues with my BMCT-SLZ :|

I understand there's an open pull request (#7087) to address this.  Does anyone have experience manually applying the changes from the PR while we wait for an official update? I'm relatively new to Home Assistant (only 10 days) and could use some guidance.

Thanks in advance!

L3nny5, 2024-02-24:
Hey ring0r,

this fix is for Zigbee2Mqtt (Z2M) not for Home Assistant. I guess you are using the plugin zigbee2mqtt?
Or are you using ZHA?

If you are using the plugin Z2M then have a look here: https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-friendly-name-set 

To send the message you can use a MQTT client like [MQTT Explorer](https://mqtt-explorer.com/)
Check out this [tutorial](http://www.steves-internet-guide.com/using-mqtt-explorer/)

Lenny