Koenkk/zigbee2mqtt #20732: [Wrong device]: AVATTO TRV06 is recognized as AVATTO ME168

ziermmar, 2024-01-11:
### Link

https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html

### Model

TRV06

### Description

Thermostatic radiator valve

### Vendor

AVATTO

### Picture (link)

https://avatto-smart.com/wp-content/uploads/2023/11/O1CN010gra4f26Gshf8mpzD_2209780547635-0-cib-1.jpg

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c13887a88d3353","nwkAddr":28501,"manufId":4417,"manufName":"_TZE200_p3dbf6qs","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��0-\u0011","65506":55,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_p3dbf6qs","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x048727fffe22c760","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1704957575695,"defaultSendRequestWhen":"immediate"}

### Notes

scheduling doesn't work as intended

Klepth, 2024-01-12:
Hi,  #20271 #20732 #20338 #19787 I think it's the same issue between ME167 et ME168 error.

Please Dev Help us and thanks for your job.

ziermmar, 2024-01-15:
The suggested workaround (https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs/tree/main) by @sychu in #19787 does indeed work

Koenkk, 2024-01-15:
@sychu would you mind creating a PR with your fix?

sychu, 2024-01-15:
@Koenkk sure, but it is not only issue with detection. [Current definition](https://github.com/Koenkk/zigbee-herdsman-converters/blob/42cb4ddfe285b2468767483a7a3199735de1b924/src/devices/tuya.ts#L3032) match 4 devices. My fix works for _TZE200_p3dbf6qs i have no idea if it it will work for 3 other devices. Should I create new definition only for _TZE200_p3dbf6qs?

Koenkk, 2024-01-16:
@sychu yes please

MartinBachmannHD, 2024-02-07:
@sychu I bought (according to the box-description) a TRV06, but it shows (after the workaround-fix which solves the 156 degrees surprise - thanks for this) now a ME167. The TRV06 offers additional options (like an OTA possibility or window detection), so I am wondering if I have received a wrong device - they all look the same? Model is TS0601 and Manufacturer is _TZE200_p3dbf6qs

lhajek, 2024-02-26:
I also have an AVATTO TRV06. Reported as ME168, now repaired as ME167. But I have the same confusion as @MartinBachmannHD.

Cooluck, 2024-03-06:
same issue with AVATTO TRV06

Cooluck, 2024-09-03:
comment