Koenkk/zigbee2mqtt #20733: TS0601_thermostat_1 - Reset Heating Setpoint

TwoPlayer, 2024-01-11:
### What happened?

Everytime i try to set a current heating setpoint, the setpoint is reseted. In this example I try to set it to 24°, but it is reseted to 22° instantly.

### What did you expect to happen?

I set the current heating setpoint and the set value stays

### How to reproduce it (minimal and precise)

Change the current heating setpoint of a TS0601_thermostat_1

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

Add-On on Home Assistant OS on Raspberry Pi 3

### Debug log

`debug 2024-01-11 09:49:41Received MQTT message on 'zigbee2mqtt/Wohnzimmer Thermostat/set' with data '{"current_heating_setpoint":24}'
debug 2024-01-11 09:49:41Publishing 'set' 'current_heating_setpoint' to 'Wohnzimmer Thermostat'
info 2024-01-11 09:49:41MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Thermostat', payload '{"alarm_switch":null,"battery":95,"boost_heating":"OFF","boost_time":null,"child_lock":"LOCK","current_heating_setpoint":24,"friday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","heating":"OFF","linkquality":102,"local_temperature":20.1,"local_temperature_calibration":-0.5,"max_temperature":30,"min_temperature":5,"monday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","position":100,"preset":"on","running_state":"heat","saturday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","schedule_friday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_monday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_saturday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_sunday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_thursday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_tuesday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_wednesday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","sunday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","system_mode":"heat","thursday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","tuesday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","wednesday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","window":"CLOSE","window_detection":"OFF"}'
debug 2024-01-11 09:49:42Received Zigbee message from 'Wohnzimmer Thermostat', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,202],"type":"Buffer"},"datatype":2,"dp":3}],"seq":54528}' from endpoint 1 with groupID 0
info 2024-01-11 09:49:42MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Thermostat', payload '{"alarm_switch":null,"battery":95,"boost_heating":"OFF","boost_time":null,"child_lock":"LOCK","current_heating_setpoint":24,"friday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","heating":"OFF","linkquality":98,"local_temperature":20.2,"local_temperature_calibration":-0.5,"max_temperature":30,"min_temperature":5,"monday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","position":100,"preset":"on","running_state":"heat","saturday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","schedule_friday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_monday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_saturday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_sunday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_thursday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_tuesday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_wednesday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","sunday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","system_mode":"heat","thursday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","tuesday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","wednesday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","window":"CLOSE","window_detection":"OFF"}'
debug 2024-01-11 09:49:42Received Zigbee message from 'Wohnzimmer Thermostat', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,220],"type":"Buffer"},"datatype":2,"dp":2}],"seq":54784}' from endpoint 1 with groupID 0
info 2024-01-11 09:49:42MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Thermostat', payload '{"alarm_switch":null,"battery":95,"boost_heating":"OFF","boost_time":null,"child_lock":"LOCK","current_heating_setpoint":22,"friday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","heating":"OFF","linkquality":102,"local_temperature":20.2,"local_temperature_calibration":-0.5,"max_temperature":30,"min_temperature":5,"monday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","position":100,"preset":"on","running_state":"heat","saturday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","schedule_friday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_monday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_saturday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_sunday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_thursday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_tuesday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","schedule_wednesday":"06:00/24.0 12:00/24.0 18:00/24.0 22:00/24.0","sunday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","system_mode":"heat","thursday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","tuesday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","wednesday_schedule":" 6h:0m 20°C, 12h:0m 15°C, 18h:0m 22°C, 22h:0m 15°C ","window":"CLOSE","window_detection":"OFF"}'`

b2un0, 2024-01-12:
check your other temperature settings like: `eco`, `comfort` and `setpoint auto`.

had the same issue with `_TZE200_zion52ef` and `_TZE200_chyvmhay`.

one of them kill your setpoint configuration.

TwoPlayer, 2024-01-15:
> check your other temperature settings like: `eco`, `comfort` and `setpoint auto`.
> 
> had the same issue with `_TZE200_zion52ef` and `_TZE200_chyvmhay`.
> 
> one of them kill your setpoint configuration.

Thank you, that was the problem. When the Preset was set to 'on', the Thermostat just stays on and donÄt look on the setpoint. I had to use the Preset 'manual' to achieve this.

closed 2024-01-15 by TwoPlayer