Koenkk/zigbee2mqtt #20738: TuYa Generic TS011F_plug_1 _TZ3000_gjnozsaz : can't pair them anymore

Gjuju, 2024-01-11:
### What happened?

In august 2023 i bought 8 TuYa smart plugs like this :
[image]
[image]

they were instantly recognized as TS011F_Plug_1 _TZ3000_gjnozsaz and are perfectly working until that, perfect reporting of power, current, etc, etc ...

IMPORTANT TO SAY :
as it was allowed i did upgrade all of them with OTA from v77 to v192 then removed 4 from my setup for later.

here is the report for a working one  :
`
{
    "child_lock": "UNLOCK",
    "current": 0.64,
    "energy": 74.5,
    "indicator_mode": "off",
    "linkquality": 255,
    "power": 67,
    "power_outage_memory": "restore",
    "state": "ON",
    "update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },
    "voltage": 233,
    "update_available": null
}
`

[image]
[image]
[image]
[image]

NOW REALLY WHAT HAPPENED :

I'd like to re-pair one of them and its recognized but it leaves just after pairing (I tried 2 on 4 to be sure it's not a faulty one)  .

I tried factory resetting them (put in pairing mode then leave them for 3 minutes to stop pair by itself, put in pairing mode then unplug and re-plug to AC...).

Every time i Try it's the same:

- device recognized,
- interviewed with success (i can at this point control them with Z2M),
- device stay in pairing mode,
- device leaves ... 
- device recognized,
- interviewed with success,
- and it loops until device pairing mode stops by itself (3 minutes).

here you can see my database.db registeries for the 4 working one :

`
{"id":4,"type":"Router","ieeeAddr":"0xa4c138c45bd3d2eb","nwkAddr":29902,"manufId":4417,"manufName":"_TZ3000_gjnozsaz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��%-\u0012","65506":54,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_gjnozsaz","zclVersion":3,"appVersion":192,"modelId":"TS011F","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":235,"rmsCurrent":627,"activePower":71,"acVoltageDivisor":1,"acVoltageMultiplier":1,"acPowerDivisor":1,"acPowerMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,7453]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":0,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffecfaeed","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffecfaeed","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffecfaeed","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeae73e9","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeae73e9","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeae73e9","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1310285083},"lastSeen":1704988111781,"defaultSendRequestWhen":"immediate"}
{"id":5,"type":"Router","ieeeAddr":"0xa4c1387a9eb9cfa0","nwkAddr":49634,"manufId":4417,"manufName":"_TZ3000_gjnozsaz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0007�%-\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_gjnozsaz","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":233,"activePower":35,"rmsCurrent":266,"acVoltageDivisor":1,"acVoltageMultiplier":1,"acPowerDivisor":1,"acPowerMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,11727]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":0}},"genOta":{"attributes":{"currentFileVersion":192}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffecfaeed","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffecfaeed","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffecfaeed","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeae73e9","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeae73e9","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeae73e9","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1310285083},"lastSeen":1704988071046,"defaultSendRequestWhen":"immediate"}
{"id":7,"type":"Router","ieeeAddr":"0xa4c1387e7116e2d6","nwkAddr":3078,"manufId":4417,"manufName":"_TZ3000_gjnozsaz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��%-\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_gjnozsaz","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":234,"activePower":0,"rmsCurrent":0,"acVoltageDivisor":1,"acVoltageMultiplier":1,"acPowerDivisor":1,"acPowerMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,11052]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffecfaeed","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffecfaeed","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffecfaeed","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeae73e9","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeae73e9","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeae73e9","endpointID":1}],"configured [... cut ...]
### What did you expect to happen?

I'd like to see pairing finishes with success.

### How to reproduce it (minimal and precise)

put my Z2M in pairing mode and my module wait.

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

 7.3.1.0 build 0 (RCP 4.3.1 firmware from Darkxst used with silabs multiprotocol server)

### Adapter

Itead Sonoff Dongle-E (EZSP) 

### Setup

Z2M addon on Home Assistant Generic X64 

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13905552/log.txt)

Barneybaer84, 2024-01-21:
Same problem. Switch from ZHA to Z2M and the device is not found.

mrmaximas, 2024-01-30:
how far away from the coordinator is this outlet?

mrmaximas, 2024-01-30:
> Perhaps problem come from silabs multiprotocol on my home assistant system

I don't have problems with pairing on 2652P but TUYA ZTU module on Telink chip is very cheap has a rather weak transmitter.

Gjuju, 2024-01-30:
Does CC2652P support multiprotocol plug-in in HA ? 

It really seems to be related to this plug-in.

mrmaximas, 2024-01-30:
> It really seems to be related to this plug-in.

sorry, i'm not using HA at all

davosje, 2024-04-27:
Hi, I have two of these labeled as NAS-WR01B but they look identical. They both stopped working a couple of months ago, was able to repair them once or twice but that also doesn't work anymore. If I hold the power button on the device the led starts blinking as it should in pairing mode. Anyone that was able to fix this or do get these device break often?

davosje, 2024-04-27:
> For me the problem was resolved 2 week ago when I revert back from silabs multiprotocol to ezsp.
> 
> Do you, use multiprotocol?

Not sure what you mean. Home Assistant Add-on: Silicon Labs Multiprotocol?

Gjuju, 2024-04-27:
Yes, home assistant multiprotocol plug-in.

davosje, 2024-04-27:
Didn't before, just installed. Now figuring out :) This works with Z2Mqtt?

closed 2024-11-25 by github-actions[bot]

Flash3dee, 2024-12-28:
Have several of these plugs, the existing in my network all working without a problem. But I have 2 new spare ones (which I bought at the same time as the working ones) which only connect and work close to the coordinator. 

Using HA, Z2Mqtt and latest firmware on the plugs. So basically they were working fine when I added them some time ago but as of this moment new plugs seem to act weird.

If I check the Z2M map one thing I notice: The working plugs are all routing through all other routers and coordinator but the newly added is only connected to **one** other router (also not connected to coordinator). I my case a same model located in other corner of my house. 

So something seems off with the routing.

zhuykovkb, 2025-01-06:
I have the same issue with _TZ3000_gjnozsaz or _TZ3000_zloso4jk. I’m not entirely sure which revision this specific socket is, but the other 4 _TZ3000_gjnozsaz and 2 _TZ3000_zloso4jk are working fine, and I’m not ready to test them just to figure out which one has the issue :)