Koenkk/zigbee2mqtt #20739: TS0502B/ _TZ3210_nvaik6gk flood the network

slima, 2024-01-11:
### What happened?

I recently purchased a 48V track light that pairs as TS0502B. The control works, but I've noticed it heavily spams the network, attaching logs.

### What did you expect to happen?

Zigbee2MQTT:debug 2024-01-11 17:40:58: No converter available for 'TS0502B' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[8,90,10,10,240,72,3,0,3,3],"type":"Buffer"}'

### How to reproduce it (minimal and precise)

stop spamming

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12

### Setup

docker

### Debug log

Zigbee2MQTT:info  2024-01-11 17:40:58: MQTT publish: topic 'zigbee2mqtt/0xa4c1381c84e41936', payload '{"brightness":254,"color":{"hue":0},"color_mode":"color_temp","color_temp":454,"last_seen":"2024-01-11T17:40:58+01:00","linkquality":184,"state":"ON"}'
Zigbee2MQTT:debug 2024-01-11 17:40:59: Received Zigbee message from '0xa4c1381c84e41936', type 'raw', cluster 'lightingColorCtrl', data '{"data":[8,91,10,10,240,72,3,0,3,3],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-11 17:40:59: No converter available for 'TS0502B' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[8,91,10,10,240,72,3,0,3,3],"type":"Buffer"}'
Zigbee2MQTT:info  2024-01-11 17:40:59: MQTT publish: topic 'zigbee2mqtt/0xa4c1381c84e41936', payload '{"brightness":254,"color":{"hue":0},"color_mode":"color_temp","color_temp":454,"last_seen":"2024-01-11T17:40:59+01:00","linkquality":184,"state":"ON"}'
Zigbee2MQTT:debug 2024-01-11 17:41:00: Received Zigbee message from '0xa4c1381c84e41936', type 'raw', cluster 'lightingColorCtrl', data '{"data":[8,92,10,10,240,72,3,0,3,3],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-11 17:41:00: No converter available for 'TS0502B' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[8,92,10,10,240,72,3,0,3,3],"type":"Buffer"}'
Zigbee2MQTT:info  2024-01-11 17:41:00: MQTT publish: topic 'zigbee2mqtt/0xa4c1381c84e41936', payload '{"brightness":254,"color":{"hue":0},"color_mode":"color_temp","color_temp":454,"last_seen":"2024-01-11T17:41:00+01:00","linkquality":184,"state":"ON"}'
Zigbee2MQTT:debug 2024-01-11 17:41:01: Received Zigbee message from '0xa4c1381c84e41936', type 'raw', cluster 'lightingColorCtrl', data '{"data":[8,93,10,10,240,72,3,0,3,3],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-11 17:41:01: No converter available for 'TS0502B' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[8,93,10,10,240,72,3,0,3,3],"type":"Buffer"}'
Zigbee2MQTT:info  2024-01-11 17:41:01: MQTT publish: topic 'zigbee2mqtt/0xa4c1381c84e41936', payload '{"brightness":254,"color":{"hue":0},"color_mode":"color_temp","color_temp":454,"last_seen":"2024-01-11T17:41:01+01:00","linkquality":184,"state":"ON"}'
Zigbee2MQTT:debug 2024-01-11 17:41:02: Received Zigbee message from '0xa4c1381c84e41936', type 'raw', cluster 'lightingColorCtrl', data '{"data":[8,94,10,10,240,72,3,0,3,3],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-11 17:41:02: No converter available for 'TS0502B' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[8,94,10,10,240,72,3,0,3,3],"type":"Buffer"}'
Zigbee2MQTT:info  2024-01-11 17:41:02: MQTT publish: topic 'zigbee2mqtt/0xa4c1381c84e41936', payload '{"brightness":254,"color":{"hue":0},"color_mode":"color_temp","color_temp":454,"last_seen":"2024-01-11T17:41:02+01:00","linkquality":184,"state":"ON"}'

terentev, 2024-05-13:
i have the same problem

closed 2024-12-30 by github-actions[bot]

devkaiwang, 2025-12-25:
Same issue here! Can we fix this problem?