Koenkk/zigbee2mqtt #20740: Aqara T1 (model WSDCGQ12LM) returns null for temperature, humidity and pressure

pauldcomanici, 2024-01-11:
### What happened?

After installing the sensor, value for temperature, humidity and pressure is null

### What did you expect to happen?

Get correct reported values.

### How to reproduce it (minimal and precise)

Install device model WSDCGQ12LM, check exposed values (see N/A).
.
Details:
I'm using a raspberry Pi4 with 8Gb RAM where I have a 1.5m USB extended cable where I plugged Sonoff Zigbee 3.0 USB Dongle Plus as coordinator. zigbee network has over 30 devices with ~ 15 routers.
While installing this device to the Aqara G3 or Aqara M1S hub I received instantly values for all 3 sensors.
I even changed the battery to make sure is not a problem with that.

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Add-on on Home Assistant OS on raspberry Pi4 with 8Gb RAM

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13906436/log.txt)

pauldcomanici, 2024-01-12:
I've downgraded to previous version of z2m 1.35.0 and the issue persists.

ramonvanbezouw, 2024-01-12:
Having the exact same issue here.

Might be related to the firmware running on the CC2652 soc of the Sonoff Zigbee 3.0 USB Dongle Plus:
https://www.zigbee2mqtt.io/devices/WSDCGQ12LM.html#adapter-firmware

Koenkk, 2024-01-13:
Indeed caused by an outdated firmware, update and re-pair the device.

closed 2024-01-13 by Koenkk

pauldcomanici, 2024-01-13:
Thank you for the support, I upgraded the coordinator FW to 20230507 and now everything is reported.

jzt308, 2025-07-03:
I'am struggeling with the same issue. Setup consists of SLZB-MR3 device with Radio 1 running as the zigbee coordinator. firmware is 20250212.
Then zigbee2Mqtt (version 2.4.0 running in docker) to publish messaged to mosquitto (also in docker) and they end up in home assistant (you guessed it inside docker)

[image]

A bunch of ecobright dimmers are working without a hitch but the Aquara WSDCGQ12LM (Temperature and humidity sensor T1) is added to the network but wont send data.
Have been trying with 3 different Aquara devices and they show up fine

[image]
But no data.
[image]

One of them did seem to work after adding and re-adding ... and again (removing battery) but only for a couple of minutes

[image]

I don't see any issues when adding it. 
`z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature and humidity sensor T1","exposes":[{"access":1,"description":"Measured temperature value","label":"Temperature","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","label":"Humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"The measured atmospheric pressure","label":"Pressure","name":"pressure","property":"pressure","type":"numeric","unit":"hPa"},{"access":1,"category":"diagnostic","description":"Temperature of the device","label":"Device temperature","name":"device_temperature","property":"device_temperature","type":"numeric","unit":"°C"},{"access":1,"category":"diagnostic","description":"Remaining battery in %, can take up to 24 hours before reported","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"category":"diagnostic","description":"Voltage of the battery in millivolts","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"category":"diagnostic","description":"Number of power outages","label":"Power outage count","name":"power_outage_count","property":"power_outage_count","type":"numeric"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"WSDCGQ12LM","options":[{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","label":"Temperature calibration","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Temperature precision","name":"temperature_precision","property":"temperature_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the humidity value (absolute offset), takes into effect on next report of device.","label":"Humidity calibration","name":"humidity_calibration","property":"humidity_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for humidity, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Humidity precision","name":"humidity_precision","property":"humidity_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the pressure value (absolute offset), takes into effect on next report of device.","label":"Pressure calibration","name":"pressure_calibration","property":"pressure_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for pressure, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Pressure precision","name":"pressure_precision","property":"pressure_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the device_temperature value (absolute offset), takes into effect on next report of device.","label":"Device temperature calibration","name":"device_temperature_calibration","property":"device_temperature_calibration","type":"numeric"}],"supports_ota":true,"vendor":"Aqara"},"friendly_name":"SOMEDATA","ieee_address":"SOMEOTHER","status":"successful","supported":true},"type":"device_interview"}'`

The only Error is: 
`zh:ember: Delivery of BROADCAST failed for '65533'.`

I've spend a couple of hours removing/adding/resetting/searching for a solution but can't seem to find any. Hoping someone can point me in the right direction..