Koenkk/zigbee2mqtt #20750: Z2M offering an update version of -1

khangwei, 2024-01-12:
### What happened?

I have a lot of Aqara switches. For some, it is offering an update version of -1 when installed version is 27.

### Zigbee2MQTT version

Zigbee2MQTT version 1.35.1 commit: unknown

### Adapter firmware version

 20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

intel nuc with proxmox running a VM with HAOS

khangwei, 2024-01-12:
[image]

velodromen, 2024-01-12:
I have the same issue for 9 of my Eco-Dim.07 devices (see issue below). It is also not possible to upgrade to this strange version "-1".
 
**Zigbee2MQTT version**: Zigbee2MQTT version 1.35.1
**Coordinator type**: EZSP v12
**Coordinator revision**: 7.3.1.0 build 0

In total I have 17 identical Eco-Dim.07 devices where the firmware differs:
9 devices on firmware 1.01 (20210802) --> These are the 9 devices that wants to be upgraded to "-1"
8 devices on firmware 3.09 (20200701) --> These units appear to have an older FW version, though no OTA updates available for these units?

_{
    "brightness": 132,
    "linkquality": 255,
    "state": "OFF",
    "update": {
        "installed_version": 16869200,
        "latest_version": -1,
        "state": "idle"
    }
}_

tradenet, 2024-01-12:
I am getting this issue interestingly enough with a Sylvania light bulb.

88fingerslukee, 2024-01-12:
I am getting this issue with an IKEA E1744

reefland, 2024-01-12:
Sengled Lights (E11-N14 )
```
{
    "brightness": 26,
    "last_seen": "2024-01-12T16:56:15.313Z",
    "linkquality": 94,
    "state": "OFF",
    "update": {
        "installed_version": 16,
        "latest_version": -1,
        "state": "idle"
    },
    "device": {
        "applicationVersion": 1,
        "friendlyName": "Back Porch Light",
        "hardwareVersion": 1,
        "ieeeAddr": "0xb0ce1814000d6f07",
        "manufacturerID": 4448,
        "manufacturerName": "sengled",
        "model": "E11-N13/E11-N13A/E11-N14/E11-N14A",
        "networkAddress": 61447,
        "powerSource": "Mains (single phase)",
        "stackVersion": 0,
        "type": "EndDevice",
        "zclVersion": 1
    },
    "update_available": null
}
```

Z2M `1.35.1` shows no firmware info about these lights on the front end:
[image]

But Home Assistant `2024.1.2` reports the "-1" firmware is available.

flynboimler, 2024-01-12:
Dito with SONOFF SNZB-02P. -1

LarsStegman, 2024-01-12:
Duplicate of #20658

StSaens, 2024-01-13:
When a new z2m is released the latest herdsman with the fix is included?

Koenkk, 2024-01-13:
Next release = 1 Feb, duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/20658

closed 2024-01-13 by Koenkk

tradenet, 2024-02-07:
This issue persists after latest update.