Koenkk/zigbee2mqtt #20751: [New device support]: SCHNEIDER WISER CCTFR6500

alexxiiis-7, 2024-01-12:
### Link

https://www.se.com/fr/fr/product/CCTFR6500/wiser-prise-connect%C3%A9e-r%C3%A9p%C3%A9teur-zigbee-13a/

### Database entry

.

### Comments

Good morning,
I am disappointed because I just switched from Zigbee to JeeZigbee and my Schneider CCTFR6500 connected plugs are not recognized under the right model, recognized under the HEIMAN HS2SK model.
I have problems with ON/OFF action.
In the list of your compatible equipment, in Schneider connected socket I find only the model CCT711119 which obviously looks but an old version of my model.
How do I use my shots with JeeZigbee as before?
Thanks a lot for your help!

### External defintion

```shell
.
```

danieledwardgeorgehitchcock, 2024-01-12:
Could you provide the fingerprint for this device from your Zigbee2MQTT database file?

Either that, or upload screenshots from your Z2M frontend of the device About tab and the Clusters tab?

alexxiiis-7, 2024-01-13:
Good morning;
Thank you for your feedback.
You will find the screenshots of the tabs "Dev Console", "About" and "Clusters".
Hopefully this will meet your request.
I saw in the Dev Console that the model noted is the one that was found automatically but is not the right one.
Here was identified a HEIMAN plug while it is a Schneider CCTFR6500 plug.

Thank you very much in advance

[image]
[image]
[image]

alexxiiis-7, 2024-01-13:
> Could you provide the fingerprint for this device from your Zigbee2MQTT database file or the auto generated converter?
> 
> Either that, or upload screenshots from your Z2M frontend of the device About tab and the Clusters tab?

alexxiiis-7, 2024-01-13:
> Could you provide the fingerprint for this device from your Zigbee2MQTT database file or the auto generated converter?
> 
> Either that, or upload screenshots from your Z2M frontend of the device About tab and the Clusters tab?

Good morning;
Thank you for your feedback.
You will find the screenshots of the tabs "Dev Console", "About" and "Clusters".
Hopefully this will meet your request.
I saw in the Dev Console that the model noted is the one that was found automatically but is not the right one.
Here was identified a HEIMAN plug while it is a Schneider CCTFR6500 plug.

Thank you very much in advance

[image]
[image]
[image]

danieledwardgeorgehitchcock, 2024-01-13:
Thanks for uploading the screenshots.

So, I can see that the device reports back that it is a Heiman plug so, there is not much we can do here.

Basically, certain manufacturers can 'whitelabel' products for other brands - this is a classic case of where this has occured. I.E. Heiman has their own brand of plug, but has also produced the same plug with branding for Schneider.

I will add the whitelabel to the Heiman device in the Herdsman Converters but, this will not change anything in the frontend / control.

The device should be fully functional with this converter.

closed 2024-01-13 by Koenkk

senna1992, 2024-10-12:
I am using some of these, also whitelabeled by schneider / wiser.

working fine but sometimes the total power does get mixed up by 1000x.
very annyoing, but seems to a firmware thing.

i am running versin 20220518"