Koenkk/zigbee2mqtt #20758: IKEA RODRET (E2201) keeps dropping off the network

aatifk, 2024-01-12:
### What happened?

After the RODRET switch connected successfully, it worked fine for about an hour and then it dropped off the network automatically. I force removed the device and added it again, and same issue happened again after about an hour. It won't maintain a connection to the network.

### What did you expect to happen?

The switch should have stayed connected.

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee Dongle P

### Setup

Native Linux Install

danieledwardgeorgehitchcock, 2024-01-13:
Hi,

What sort of LQI figures are you getting when the device is connected and functional?

Also, what does your network map say the route is back to the coordinator?

Is the device a fair distance away from the coordinator / any Zigbee routing devices?

aatifk, 2024-01-14:
I've been getting an LQI of 130+ when it's functional, and my house is covered in routers. I'd say it was about 10 feet from the coordinator though when I was trying this earlier. 

As an update, I force-paired it to a nearby router (permit join on the router only) and it's stayed connected for the past 24 hours. Will see how it holds up in the longer run.

aatifk, 2024-01-19:
Update: It kept on for about 3 days and dropped off again. The router is hardly 5 ft from the device and yet it dropped again. Any thoughts?

Taxxian, 2024-02-25:
I had a probably connected problem;

I could not even pair the Rodret to the Sonoff P stick, it dropped of the network while pairing.

Solution: I skipped the  Coordinator and paired it to a Sonoff Switch acting as a hub, and since this hub is always closer to the Rodret than the Coordinator it now works fine.

MattWestb, 2024-03-03:
The problem is some routers is not good ones and all IKEA controllers is using Zigbee 3 security (on current production firmware) and if the parent is not doing it OK the device think its OK but the router have timing it out and is deleting it in its children table so the network cant finding it.

RubenKelevra, 2024-05-03:
I bought two today to check them out, but can't connect them to my network.

Pressing the link button for >10 seconds does nothing, not even lighting up the LED on them.

I've checked the batteries I put in them, they are happily at around 1.5V.

Is there any trick or something I'm missing? 🤔

MattWestb, 2024-05-03:
4 pressing the paring bottom is resetting the controller and going in paring mode.
Long press is touch light paring mode that being used for direct paring with light without coordinator.

RubenKelevra, 2024-05-03:
@MattWestb thanks for the fast help!

Will report back if I see the same issue with my devices

RubenKelevra, 2024-07-28:
@MattWestb I can't confirm the issue you're having with the device. I got two of them and use one of them daily and one maybe biweekly to turn lights on/off. They both work fine for 3 month now with no outage or delay after pressing.

wwerther, 2024-08-03:
Hi,
I tried today and I'm also having issues with these devices. I have two of them and pairing works fine. After that no commands are issued and send when I press a button.
They work fine when I directly peer them to a TRETAKT smart plug. But when I try to peer with my coordinator no messages are transferred. 
My coordinator is a cod-m device connected via network. All other devices in my zigbee net work fine.

RubenKelevra, 2024-08-03:
@wwerther do you use fw 1.0.57?

wwerther, 2024-08-04:
> @wwerther do you use fw 1.0.57?
As I am not sure which FW version we're exactly talking about, I will just list all that I can get:

Coordinator: 	
CZC v1.0, ESP-FW: 1.0.0 (codm-v1-prod), 
CC2652P hw version:	CC2652P7
CC2652P fw rev:	20230507

Home-Assistant: 2024.7.4
addon Zigbee2Mqtt: Current version: 1.39.1-1
Coordinator-Type zStack3x0
Coordinator-Version 20230507
Frontend-Version 0.7.4
Zigbee Herdsman Konverter Version 19.72.0
Zigbee Herdsman Version 0.55.3

Rodret-Version: 1.0.47

interestingly I get "online" messages. So today I can see that coordinator sees device was last seen 3hours, 4hours ago. anyway Keypress is not recognized (no event visible in logs)

 please let me know if I can support in creating debug logs etc. 

Thanks

Minims, 2024-11-02:
Hello,

I have the same issue since 2 week. 
It has worked great for 1 year. 

To make it work again i need to put a new battery, but it works only 3 days max.
The battery is not drained, still 1.48v on the battery.

The device keeps lopping on online message

RubenKelevra, 2025-01-02:
> Rodret-Version: 1.0.47

So firmware 1.0.57 isn't shown as update for this device?

RubenKelevra, 2025-01-02:
> This issue is stale because it has been open 180 days with no activity. Remove stale label or comment or this will be closed in 30 days

@Koenkk the bot is malfunctioning. It hasn't been 180 since the last activity here

Koenkk, 2025-01-02:
@RubenKelevra the comment text was wrong, I'll fix it, thanks for letting me know!

closed 2025-03-12 by github-actions[bot]

tobiasbayer, 2026-01-14:
Was this issue closed accidentially? I have the same issue and I cannot find a solution or a reason for the observed behavior in this thread.