Koenkk/zigbee2mqtt #20760: No light flash functionality via Home Assistant any more

fintmu, 2024-01-13:
### What happened?

Light flash via Home Assistant is broken. Zigbee2MQTT gives an error: No converter available for 'flash' (2) / No converter available for 'flash' (10)

### What did you expect to happen?

Light should flash when flash (short or long) is requested in Home Assistant light.turn_on service call.

### How to reproduce it (minimal and precise)

service: light.turn_on
data:
  rgb_color:
    - 46
    - 194
    - 126
  brightness: 255
  flash: long
target:
  entity_id: light.office_desk

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

Electrolama zzh!

### Setup

docker container

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13927526/log.txt)

miihku, 2024-01-13:
Same problem here...

omega1582, 2024-06-03:
This still appears to be an issue.  Last few days I've been converting from ZHA to Z2M and this is one that I just noticed to day.  I have an automation that does flast:short and when it runs, Z2M throws the error 'No converter available for 'flash' (2)'

SteveAy, 2024-08-04:
Just realised the same here. No flash on any lights.

davidski, 2024-12-30:
This issue still exists. If there is additional information needed, happy to pitch in.

Dazzel, 2025-01-10:
Having the same issue.

A workaround is to not use the light on action from HA, but publish only the effect via mqtt:

```
action: mqtt.publish
metadata: {}
data:
  evaluate_payload: false
  qos: "0"
  topic: zigbee2mqtt/FRIENDLY_NAME_OF_ENTITY/set
  payload: "{\"effect\": \"blink\"}"
```

This ensures the light will be in same state after the effect finished as it was before.

Using the action of HA `"state": "ON"` will be published as well.

The workaround works, but cannot be the desired solution.

klauserber, 2025-03-16:
I have the same issue, but only with Philips bulbs. Ich have some cheap Tuya bulbs who can flash.

closed 2025-05-25 by github-actions[bot]

derDeno, 2025-08-17:
I encountered this issue too.
I was using the hue hub before, where it worked fine, but with the z2m addon, my hue bulbs and strips no longer support the flash function.