Koenkk/zigbee2mqtt #20761: Exception while calling fromZigbee converter

diaz66, 2024-01-13:
### What happened?

- After switching to version 1.35.1-1 I started to encounter the error in question.
- From this moment on, zigbee2mqtt no longer sends anything to home assistant (Mosquitto broker)
- The device generating the error is a [PJ-MGW1203](https://www.zigbee2mqtt.io/devices/PJ-MGW1203.html#tuya-pj-mgw1203)
- The system block occurs 2/3 times a day it's more , randomly

My configuration
homeassistant: true
mqtt:
  server: mqtt://core-mosquitto
  user: ***
  password: ***
serial:
  port: /dev/ttyACM0
  adapter: ezsp
frontend:
  port: 8099
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  channel: 25
  log_level: info
device_options:
  legacy: false
devices:
  '0xe0798dfffea826ff':
    friendly_name: Router ZBDongle-E
  '0x003c84fffef0ecca':
    friendly_name: ZigBee Termosifone Camera Mat
  '0xa4c1380483ecb1b5':
    friendly_name: ZigBee Termometro Sala
  '0xa4c13878a044e674':
    friendly_name: ZigBee Termometro Soggiorno
  '0xa4c138fa78091d52':
    friendly_name: ZigBee Presa Frigorifero
  '0xa4c138bab31104b2':
    friendly_name: ZigBee Luce Terrazzo Camera Matrimoniale
  '0xa4c13831eaf7dbba':
    friendly_name: ZigBee Tapparella1 Sala
  '0xa4c138c3b27cee34':
    friendly_name: ZigBee Lampada Luna
  '0xa4c138bcec51d379':
    friendly_name: ZigBee Luce Sala
  '0xa4c1383aad8d029e':
    friendly_name: ZigBee Presa Sala TV
  '0xa4c138e52b567c58':
    friendly_name: ZigBee Presa Camera Grande Play
  '0xa4c138fe60c0d996':
    friendly_name: ZigBee Presa Camera Grande PC
  '0xa4c138ddd7cffecf':
    friendly_name: ZigBee Presa Soggiorno TV
  '0xa4c1388b23a3bf79':
    friendly_name: ZigBee Porta Frigo
  '0xa4c1382e0495eafe':
    friendly_name: ZigBee Porta Congelatore
  '0xa4c138a06155b4d5':
    friendly_name: ZigBee Luce Camera Grande
    power_precision: 1
    power_calibration: 3000
    current_calibration: 100
    energy_precision: 1
    voltage_precision: 1
    current_precision: 1
    state_action: true
  '0x70ac08fffe366ce1':
    friendly_name: Zigbee Power Meter Casa
    energy_precision: 2
    energy_calibration: 1
    current_precision: 2
    power_precision: 2
    voltage_precision: 2
    current_calibration: 1
    power_calibration: 1
    voltage_calibration: 1
  '0xa4c13851ad8d6f9b':
    friendly_name: ZigBee Termometro Esterno
    temperature_precision: 1
  '0xa4c1383a8254d9ae':
    friendly_name: ZigBee Termometro Camera Grande
  '0xa4c1382c28303539':
    friendly_name: ZigBee Termometro Camera Piccola
  '0xa4c13818fe6c6cd3':
    friendly_name: ZigBee Porta Ingresso Interna
  '0xa4c13800b40af798':
    friendly_name: ZigBee Porta Ingresso Esterna
  '0xa4c1380aa558d939':
    friendly_name: ZigBee Presa Camera Piccola
  '0xa4c138af878589ac':
    friendly_name: ZigBee Presa Camera Mat
  '0xa4c138351db79811':
    friendly_name: ZigBee Sensore Acqua Esterno
  '0x4c5bb3fffe2eeaf7':
    friendly_name: Zigbee Power Meter Cucina
    description: Cappa Cucina + le due prese
    current_calibration: 1
    current_precision: 2
    power_precision: 2
    voltage_precision: 2
    energy_precision: 2
    power_calibration: 1
    voltage_calibration: 1
    energy_calibration: 1
  '0x00124b00291ffbe0':
    friendly_name: ZigBee Termometro Camera Mat
blocklist: []
availability: false

### What did you expect to happen?

I downgraded the addon version to 1.34.0-1 and for the moment, I no longer encounter these errors

### How to reproduce it (minimal and precise)

I would not know. The only way is to own the device [PJ-MGW1203](https://www.zigbee2mqtt.io/devices/PJ-MGW1203.html#tuya-pj-mgw1203) and wait for the problem to occur.

### Zigbee2MQTT version

1.35.1-1 

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

Add On Home Assistant Generic x86-64

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13927649/log.txt)

Yves-Be, 2024-01-13:
Same here, also have a [PJ-MGW1203](https://www.zigbee2mqtt.io/devices/PJ-MGW1203.html#tuya-pj-mgw1203) in my installation.

Koenkk, 2024-01-13:
Could you provide the herdsman debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

closed 2024-01-13 by Koenkk

diaz66, 2024-01-13:
```
2024-01-13T21:49:09.764Z zigbee-herdsman:adapter:ezsp:driv Close driver
2024-01-13T21:49:09.764Z zigbee-herdsman:adapter:ezsp:driv Reset connection. Try 6
2024-01-13T21:49:09.764Z zigbee-herdsman:adapter:ezsp:driv Stop driver
2024-01-13T21:49:09.764Z zigbee-herdsman:adapter:ezsp:ezsp Stop ezsp
2024-01-13T21:49:09.764Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2024-01-13T21:49:09.764Z zigbee-herdsman:adapter:ezsp:driv Close driver
2024-01-13T21:49:10.764Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
2024-01-13T21:49:10.765Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
2024-01-13T21:49:10.766Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 1 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/@serialport/stream/dist/index.js:118:18
2024-01-13T21:49:10.767Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 1 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/@serialport/stream/dist/index.js:118:18
2024-01-13T21:49:10.786Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 4
2024-01-13T21:49:10.786Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
2024-01-13T21:49:10.787Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 4 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/@serialport/stream/dist/index.js:118:18
2024-01-13T21:49:11.783Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 4
2024-01-13T21:49:11.783Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
2024-01-13T21:49:11.784Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 4 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/@serialport/stream/dist/index.js:118:18
2024-01-13T21:49:12.044Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 5
2024-01-13T21:49:12.044Z zigbee-herdsman:adapter:ezsp:erro Reset error Error: Failure to connect
    at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:277:19)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:139:9)
    at Driver.onReset (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:111:17)
2024-01-13T21:49:12.044Z zigbee-herdsman:adapter:ezsp:driv Pause 60sec before try 7
2024-01-13T21:49:12.044Z zigbee-herdsman:adapter:ezsp:driv Reset connection. Try 6
2024-01-13T21:49:12.044Z zigbee-herdsman:adapter:ezsp:driv Stop driver
2024-01-13T21:49:12.044Z zigbee-herdsman:adapter:ezsp:ezsp Stop ezsp
2024-01-13T21:49:12.044Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2024-01-13T21:49:12.044Z zigbee-herdsman:adapter:ezsp:driv Close driver
2024-01-13T21:49:13.045Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
2024-01-13T21:49:13.049Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 1 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/@serialport/stream/dist/index.js:118:18
2024-01-13T21:49:13.620Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 4
2024-01-13T21:49:13.620Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2024-01-13T21:49:13.715Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 2
2024-01-13T21:49:13.715Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
2024-01-13T21:49:13.716Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 2 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/@serialport/stream/dist/index.js:118:18
2024-01-13T21:49:13.792Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
[... cut ...]
2024-01-13T21:49:16.785Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 5
2024-01-13T21:49:16.786Z zigbee-herdsman:adapter:ezsp:erro Reset error Error: Failure to connect
    at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:277:19)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:139:9)
    at Driver.onReset (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:111:17)
2024-01-13T21:49:16.786Z zigbee-herdsman:adapter:ezsp:driv Pause 60sec before try 3
```