Koenkk/zigbee2mqtt #20762: IKEA Vindstyrka Reporting

MaxMSchneider, 2024-01-13:
### What happened?

Not able to set the reporting intervals for the measured value PM25 in the frontend. So far this sensor is spanning into the Zigbee-Network multiple times  second causing delays in other actions in the ZigBee network.

### What did you expect to happen?

Reporting action value being accepted, reporting less often.

### How to reproduce it (minimal and precise)

Setting reporting action value to any number via the frontwnd

### Zigbee2MQTT version

 1.35.0-dev commit: b17c22b

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Add-On in HA, RasPi 4B 8GB

### Debug log

When setting the Attribute measuredValue:

`
Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x943469fffe7c1626/1 pm25Measurement([{"attribute":"measuredValue","minimumReportInterval":60,"maximumReportInterval":120,"reportableChange":10}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE')
`

After checking the IKEA.ts [https://github.com/koenkk/zigbee-herdsman-converters/blob/master/src%2Fdevices%2Fikea.ts#L1146-L1170](url) trying to set for measuredValueIkea:

`Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x943469fffe7c1626/1 pm25Measurement([{"attribute":"measuredValueIkea","minimumReportInterval":60,"maximumReportInterval":120,"reportableChange":10}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4476,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

dedors, 2024-01-17:
I get 1 update every second.
When trying to change the intervall I get the same responses as above.

sanyafifa, 2024-01-19:
`Failed to configure '0x287681fffe6f2bb1', attempt 3 (Error: ConfigureReporting 0x287681fffe6f2bb1/1 pm25Measurement([{"attribute":"measuredValueIkea","minimumReportInterval":60,"maximumReportInterval":120,"reportableChange":2}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4476,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:331:28) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:684:22) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/ikea.ts:1168:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))`

rippiedoos, 2024-01-22:
I have the same problem. Is this a thing of defining the right values somewhere? I want to use this but it's flooding my HA-install, MQTT and logfiles with values in logfiles and history.

Koenkk, 2024-01-22:
@sjorge do you know if it's possible to change this?

sjorge, 2024-01-22:
I think the PR/issue file on zh having to do with the data type for type pm25 might be related to this.

I’ll see if I can find my vindstyrke, to check but mine never worked very well to begin with.

sanyafifa, 2024-01-22:
I can test it on my device :)

sjorge, 2024-01-22:
I can confirm I see the error during pairing on mine too:

```
Zigbee2MQTT:error 2024-01-22 21:20:26: Failed to configure '0x90ab96fffefa5d91', attempt 1 (Error: ConfigureReporting 0x90ab96fffefa5d91/1 pm25Measurement([{"attribute":"measuredValueIkea","minimumReportInterval":60,"maximumReportInterval":120,"reportableChange":2}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4476,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/Users/sjorge/Documents/repos/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28)
    at Endpoint.configureReporting (/Users/sjorge/Documents/repos/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:672:22)
    at Object.configure (/Users/sjorge/Documents/repos/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/ikea.ts:1168:13)
    at Configure.configure (/Users/sjorge/Documents/repos/zigbee2mqtt/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/Users/sjorge/Documents/repos/zigbee2mqtt/lib/eventBus.ts:174:17))
Zigbee2MQTT:info  2024-01-22 21:20:27: Configuring '0x90ab96fffefa5d91'
```

Which definitely was not the case before, probably related to https://github.com/Koenkk/zigbee-herdsman/issues/828

Koenkk, 2024-01-23:
So if I understand correctly it works after changing to:

```js
measuredValueIkea: {ID: 0x0000, type: DataType.singlePrec},
```

sjorge, 2024-01-23:
> So if I understand correctly it works after changing to:
> 
> ```js
> measuredValueIkea: {ID: 0x0000, type: DataType.singlePrec},
> ```

Yes,  but then it overlaps with the regular measuredValue which has the same id but different datatype

Koenkk, 2024-01-23:
What if we swap them like this:

```
            measuredValue: {ID: 0x0000, type: DataType.uint16},
            measuredValueIkea: {ID: 0x0000, type: DataType.singlePrec, manufacturerCode: ManufacturerCode.IKEA_OF_SWEDEN},
            //default cluster spec: values reported as uint16
```

sjorge, 2024-01-23:
I don’t think that will work, yesterday i tried with just measuredValue commented and it didn’t pick the ikea one. I’m guessing because there is no manufacturerCode send by the device.

sjorge, 2024-01-23:
I think the best way to fix is probably removing the ikea one from cluster.ts and specifying the data type during the configure reporting. For decoding the dt doesn’t seem to matter ?

Koenkk, 2024-01-24:
> I think the best way to fix is probably removing the ikea one from cluster.ts and specifying the data type during the configure reporting. For decoding the dt doesn’t seem to matter ?

Sounds good! (if the received value is still correct)

sjorge, 2024-01-28:
This should hopefully be fixed in the february release (https://github.com/Koenkk/zigbee-herdsman-converters/pull/6972)

sanyafifa, 2024-02-01:
Zigbee2MQTT version
1.35.2-dev commit: [6621f6fd](https://github.com/Koenkk/zigbee2mqtt/commit/6621f6fd)

`Error 2024-02-01 22:49:43Failed to configure '0xa4c13828302637fa', attempt 2 (Error: Read 0xa4c13828302637fa/1 hvacUserInterfaceCfg(["programmingVisibility"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28) at Endpoint.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:469:22) at setupAttributes (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:63:13) at configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:84:17) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))`

FranciscoSGPa, 2024-03-13:
ZigBeee2MQTT version 1.36 ufortunately the problem still happens so the february release (https://github.com/Koenkk/zigbee-herdsman-converters/pull/6972) does not seem to help. Any idea on hoy to solve How to write an external definition to solve this issue?
Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x943469fffe814aa4/1 pm25Measurement([{"attribute":"measuredValue","minimumReportInterval":60,"maximumReportInterval":120,"reportableChange":2}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE')'

Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x943469fffe814aa4/1 pm25Measurement([{"attribute":"measuredValueIkea","minimumReportInterval":60,"maximumReportInterval":120,"reportableChange":2}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Don't know value type for 'undefined')'

FranciscoSGPa, 2024-03-22:
Reporting configuration resolved with MQTT message according https://github.com/Koenkk/zigbee2mqtt/issues/21807

sjorge, 2024-04-28:
In the dev release / next release in may, it should now be possible to configure reporting via the frontend again.

closed 2024-11-25 by github-actions[bot]