Koenkk/zigbee2mqtt #20763: TS011F_plug_1 not reporting power

gianlucasullazzo, 2024-01-13:
### What happened?

Hi guys, I have several TuYa based plug, [TS011F_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1)
It seems like the power isn't being reported correctly even if appliances are plugged on.
[image]
I see other old issues similar to this but they have been marked as solved several weeks ago.
My Zigbee-herdsman-converters version is 18.9.0.
I just commented on https://github.com/Koenkk/zigbee2mqtt/issues/13222#issuecomment-1889000067 but the issue is closed and I don't know if you read comments of closed issues. If yes, please close this one; we can continue on https://github.com/Koenkk/zigbee2mqtt/issues/13222

### What did you expect to happen?

Power should be reported correctly

### How to reproduce it (minimal and precise)

Pair plug, connect appliances and see data from dashboard.

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Sonoff Dongle USB Zigbee più-ZBDongle-E 3.0

### Setup

Add-on on HassOS on RPi4

tannisroot, 2024-02-04:
I have the same problem with my plugs. They do sometimes report the power measurement but then just stop.
In case this is a bug with the firmware and not Z2M, since amperage and voltage are reported correctly, can't Z2M just apply a hack to calculate wattage from that? Sounds pretty trivial, in fact I already do just that using a Home Assistant helper.

francisp2, 2024-03-18:
I have one that does not even report voltage

[image]

IngAley, 2024-03-27:
Same problem here, and I just have bought them on Amazon, and have got the round ones.
Here "Voltage" and "sum of consumed energy" is shown, but not Power, and Current. Stays 0 although value of the "sum of consumed energy is" going up... Strange!

ingoratsdorf, 2024-04-10:
Sorry, just noticed that the other issue is closed although later while this new one is the same but open.
Here I posted screenshots and comparison between Tuya app and Z2M, the plug is fully functional in Tuya but not in Z2M, so it's a software issue, not a hardware issue.

https://github.com/Koenkk/zigbee2mqtt/issues/13222#issuecomment-2048470328

paolapersico1, 2024-07-07:
The issue is still present on version 1.39.0.

garagara93, 2024-09-27:
Same problem here. Someone help us

confuu, 2024-12-30:
I have same problem too. I have multiple Tuya Smart Plugs that report only Energy properly. Power/Current/Voltage is being reported only sometimes randomly no matter how I config the reporting settings. They used to report everything properly one or two years ago, but then at somepoint they stopped doing it after updating z2m. 

My plugs are recognized as 
Zigbee Model: TS011F
Zigbee Manufacturer: _TZ3000_gjnozsaz
Manufacturer: [Tuya]
Model: [TS011F_plug_1]

gianlucasullazzo, 2025-03-02:
The issue is still present on 2.1.2

closed 2025-05-11 by github-actions[bot]