Koenkk/zigbee2mqtt #20764: Aqara T1 water leak sensor (lumi.flood.agl02) - No Battery status and error while report on request

marianob85, 2024-01-13:
### What happened?

Bought new Aqara T1 water leak sensor [SJCGQ12LM]
The device reports firmware 28 by reading the cluster.
Battery status is not reported ( Voltage and % ), and when pressing the button to report actual status nothing happend. In debug log I can see that _No converter available for 'SJCGQ12LM'_ 

### What did you expect to happen?

Report battery status, report on demad

### How to reproduce it (minimal and precise)

Pair to Zigbee2MQTT with Home Assistant SkyConnect USB Stick

### Zigbee2MQTT version

1.35.1 commit: 9eaaa0f

### Adapter firmware version

 7.3.2.0 build 212

### Adapter

Home Assistant SkyConnect USB Stick

### Setup

Docker, rPI5

### Debug log

2024-01-13 13:11:00Received Zigbee message from '0x54ef4410009acd35', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.flood.agl02"}' from endpoint 1 with groupID 0
Debug 2024-01-13 13:11:00SJCGQ12LM: Processed data into payload {}
Debug 2024-01-13 13:11:00Received Zigbee message from '0x54ef4410009acd35', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,197,11,3,40,25,4,33,168,19,5,33,17,0,6,36,2,0,0,0,0,8,33,28,1,10,33,0,0,12,32,1,100,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2024-01-13 13:11:00No converter available for 'SJCGQ12LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"247":{"data":[1,33,197,11,3,40,25,4,33,168,19,5,33,17,0,6,36,2,0,0,0,0,8,33,28,1,10,33,0,0,12,32,1,100,16,0],"type":"Buffer"}}'

crivchri, 2024-01-22:
Hi,
I have a similar issue with the Aqara T1 water leak sensor [SJCGQ12LM], where the Battery % and Voltage are not reported.
By the way, the firmware version is incorrectly found as “2019www.”.
Thank you in advance!

LightMoon, 2024-01-25:
I third this one! Experiencing the same!

Koenkk, 2024-01-26:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/bedcf4cc4f3c63058349151571e7a82a): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

marianob85, 2024-01-26:
Works fine:
`2024-01-26 18:59:15Received Zigbee message from 'Xiaomi/WaterLeakDetector/Mariano-Łazienka-Góra', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.flood.agl02"}' from endpoint 1 with groupID 0
Debug 2024-01-26 18:59:15SJCGQ12LM: Processed data into payload {}
Debug 2024-01-26 18:59:15Received Zigbee message from 'Xiaomi/WaterLeakDetector/Mariano-Łazienka-Góra', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,241,11,3,40,25,4,33,168,19,5,33,22,0,6,36,2,0,0,0,0,8,33,28,1,10,33,0,0,12,32,1,100,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2024-01-26 18:59:15SJCGQ12LM: Processed buffer into data {"1":3057,"3":25,"4":5032,"5":22,"6":2,"8":284,"10":0,"12":1,"100":0}
Debug 2024-01-26 18:59:15SJCGQ12LM: unknown key 6 with value 2
Debug 2024-01-26 18:59:15SJCGQ12LM: Processed data into payload {"voltage":3057,"battery":100,"device_temperature":25,"power_outage_count":21,"state":"OFF"}
Debug 2024-01-26 18:59:15SJCGQ12LM: Processed data into payload {"voltage":3057,"battery":100,"device_temperature":25,"power_outage_count":21,"state":"OFF"}
Info 2024-01-26 18:59:15MQTT publish: topic 'zigbee2mqtt/Xiaomi/WaterLeakDetector/Mariano-Łazienka-Góra', payload '{"battery":100,"battery_low":false,"device_temperature":25,"linkquality":255,"power_outage_count":21,"state":"OFF","tamper":false,"update":{"installed_version":28,"latest_version":18,"state":"idle"},"update_available":false,"voltage":3057,"water_leak":false}'`

crivchri, 2024-01-27:
Hi @Koenkk,

Thank you a lot, it seems to work with this external converter (see below), the battery % and voltage are now shown.

However, the OTA updates are now gone for this device and the "softwareBuildID" is recognized as "2019\u0000www." , which seems weird. Can this be resolved?

Thank you in advance!

{
    "battery": 100,
    "battery_low": false,
    "device_temperature": 25,
    "last_seen": "2024-01-27T09:27:05+01:00",
    "linkquality": 255,
    "power_outage_count": 70,
    "state": "OFF",
    "tamper": false,
    "voltage": 3061,
    "water_leak": false,
    "device": {
        "applicationVersion": 28,
        "dateCode": "20200316",
        "friendlyName": "Capteur fuite Cave",
        "hardwareVersion": 1,
        "ieeeAddr": "0x############",
        "manufacturerID": 4447,
        "manufacturerName": "LUMI",
        "model": "SJCGQ12LM",
        "networkAddress": 47057,
        "powerSource": "Battery",
        "softwareBuildID": "2019\u0000www.",
        "stackVersion": 2,
        "type": "EndDevice",
        "zclVersion": 3
    }
}

Koenkk, 2024-01-28:
Integrated the fix!

The `2019\u0000www.` is what actually comes from the device, so we cannot fix it from z2m but might be fixed in a future OTA update of the device.

closed 2024-01-28 by Koenkk

crivchri, 2024-01-28:
Thanks a lot!
When will this fix be integrated in stable release of Zigbee2MQTT?

LightMoon, 2024-01-28:
@Koenkk 
Will the fix merge into master branch soon or using the external converter is only way to get this working?

Koenkk, 2024-01-29:
Next release is on 1 Feb

asan84, 2024-02-16:
I have version Z2M 1.35.3 but this problem has not been solved. One of the 4 sensors does not display battery charge.
Battery N/A%
Voltage N/AmV

Adding or removing a sensor does not solve the problem.

dx0G, 2024-07-21:
Z2M version 1.39 still no battery status

z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/master_bathroom_water_leak sensor_t1', payload '{"battery":null,"battery_low":false,"linkquality":160,"tamper":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null,"water_leak":false}'

Technol7, 2024-08-16:
> Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/bedcf4cc4f3c63058349151571e7a82a):
>
> * save this as file next to `configuration.yaml` as [...]

> Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/bedcf4cc4f3c63058349151571e7a82a):
>
> * save this as file next to `configuration.yaml` as [...]

Hello Koenkk,
Is this still the only solution (concerter externe) to have the battery value of the SJCGQ12LM returned?
I have exactly the same problem as Marianob85.
Thank you in advance,
Have a nice day.