Koenkk/zigbee2mqtt #20765: TuYa TS011F_plug_1: OTA Update unavailable

ConfigMgrPro, 2024-01-13:
### What happened?

I have 6 pcs [TS011F_plug](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html) all having the same issue (sold by the same seller on [AliExpress).](https://nl.aliexpress.com/item/1005005665161815.html)

The device cannot be updated, as it has no OTA cluster available.

```
Info 2024-01-13 13:12:24Checking if update available for 'Powermonitor Quooker cube'
Info 2024-01-13 13:12:24MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Powermonitor Quooker cube"},"error":"Failed to check if update available for 'Powermonitor Quooker cube' (Failed to find endpoint which support OTA cluster)","status":"error","transaction":"ok9ma-6"}'
Error 2024-01-13 13:12:24Failed to check if update available for 'Powermonitor Quooker cube' (Failed to find endpoint which support OTA cluster)
```

On ZHA i also had no OTA cluster visible for these devices.

### What did you expect to happen?

I want to be able to update the devices, as i having troubles that they shut down randomly.

### How to reproduce it (minimal and precise)

Click on the OTA update button.

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

 6.7.8.0 build 373

### Adapter

Used both the Sonoff zigbee dongle and Sonoff ZHABridge

### Setup

Rpi4

Koenkk, 2024-01-13:
Fixed! Note that even with this it might complain about not having a fw for this device. In that case you need to follow [this](https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html)

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow.

closed 2024-01-13 by Koenkk

ConfigMgrPro, 2024-01-15:
Unfortunately still same issue, no OTA cluster for Tuya device. (Z2M 1.35.1-dev commit: [2bd7df8](https://github.com/Koenkk/zigbee2mqtt/commit/2bd7df8))

ConfigMgrPro, 2024-01-23:
I had some issues updating Z2M, it's now finally updated, so i was able to test.

The original Error message is gone,  but i have a New one.

Failed to check if update available for 'Powerplug xxx' (OTA: Device didn't respond to OTA request)

I already reconfigured the device.

ConfigMgrPro, 2024-01-23:
MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Powerplug xxx"},"error":"Failed to check if update available for 'Powerplug xx' (OTA: Device didn't respond to OTA request)","status":"error","transaction":"3d3hq-2"}'

Also i see an software version of -1 in the log

MQTT publish: topic 'zigbee2mqtt/Powerplug xxx', payload '{"child_lock":"UNLOCK","current":0,"energy":1.6,"indicator_mode":"off/on","linkquality":91,"power":0,"power_outage_memory":"on","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":236}'

Koenkk, 2024-01-24:
@ConfigMgrPro can you check if OTA is supported with the original hub? I expect that your device may not support OTA at all.

ConfigMgrPro, 2024-01-25:
Hi @Koenkk,

I don't have an original hub. Can i just buy any tuya zigbee hub? Thanks!

If OTA is not supported, can i still update using the manual flash method as mentioned [here](https://www.zigbee2mqtt.io/guide/adapters/flashing/alternative_flashing_methods.html)?

Koenkk, 2024-01-26:
> I don't have an original hub. Can i just buy any tuya zigbee hub? Thanks!

yes

> If OTA is not supported, can i still update using the manual flash method as mentioned [here](https://www.zigbee2mqtt.io/guide/adapters/flashing/alternative_flashing_methods.html)?

No, that's for flashing the coordinator fw (not the TuYa device)

b2un0, 2024-06-30:
same for me for 2 of `_TZ3000_okaz9tjs `

SantaFox, 2024-10-15:
Same problem... 
```
Device type: Router
Zigbee Model: TS011F
Zigbee Manufacturer: _TZ3000_okaz9tjs
Description: Smart plug (with power monitoring by polling)
Support status: Supported

Firmware build date: 20210625
Firmware version: 500+TZSKT31BS102
Manufacturer: [Tuya](https://www.zigbee2mqtt.io/supported-devices/#v=Tuya)
Model: [BSD29_1](https://www.zigbee2mqtt.io/devices/BSD29_1.html#tuya-bsd29_1)
```
But:
```
{
    "current": 0.02,
    "energy": 0.37,
    "linkquality": 164,
    "power": 0,
    "power_outage_memory": "restore",
    "state": "ON",
    "voltage": 241,
    "child_lock": null,
    "indicator_mode": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}
```