Koenkk/zigbee2mqtt #20766: Since the last update: IKEA ICPSHC24-30EU-IL-1 does not remember brightness

z-master42, 2024-01-13:
### What happened?

I use the IKEA ICPSHC24-30EU-IL-1 together with IKEA E1743, which are also directly linked in Z2M. Since the last update, the brightness is now always set to the lowest level when switching on. Previously, the lamp was switched on again at the previously set brightness. I have also bound Lidl HG06492B to this push-button, these "remember" their last brightness value.
Übersetzt mit DeepL https://www.deepl.com/app/?utm_source=android&utm_medium=app&utm_campaign=share-translation

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

Add-on on Home Assistent OS

danishru, 2024-01-13:
I think this error is related to this discussion https://github.com/Koenkk/zigbee2mqtt/issues/19211

Мое обходное решение https://github.com/Koenkk/zigbee2mqtt/issues/19211#issuecomment-1871092838

@Koenkk Please link these problems.

tjorim, 2024-01-14:
I have the same issue since a few months: https://github.com/Koenkk/zigbee2mqtt/discussions/20783

closed 2024-08-04 by z-master42