Koenkk/zigbee2mqtt #20767: Lidl 14158704L (Livarno Home LED Floor Lamp RGBW) not functioning properly

Nyarly360, 2024-01-13:
### What happened?

I just added a Lidl 14158704L floor lamp to my setup (now 29 devices), but it doesn't work properly. I think it's the same problem as with the Lidl light bar discussed and solved [here](https://github.com/Koenkk/zigbee2mqtt/issues/13421). For example: I can hardly control brightness, the white LEDs won't turn off, if I switch to colors and, thus, the colors are hardly visible and I can't turn on/off the lamp via Z2M. @Koenkk: Could you please transfer the fixes for the light bar to the floor lamp as well?

### What did you expect to happen?

I expected to be able to control the floor lamp like any other RGBW light I currently use.

### Zigbee2MQTT version

1.35.1 commit: 9eaaa0f

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle P

### Setup

Docker container on Intel NUC

Koenkk, 2024-01-15:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/a3d7b568e070cd58150d15b4ed0e870d): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Nyarly360, 2024-01-16:
Hi, thanks for the file. I did what you requested, but the converter is not picked up. The log says: "Failed to load external converter file 'ext_converter.js' (tzLocal is not defined)" and "Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version".

Koenkk, 2024-01-17:
Updated https://gist.github.com/Koenkk/a3d7b568e070cd58150d15b4ed0e870d

Nyarly360, 2024-01-17:
Thanks. Sadly, things don't appear to be so quick to fix. I still can't switch the lamp off via Z2M and can't change brightness or colors. The converter makes it even worse than before, as I can't eve change white hues anymore. When I try to change colors, Z2M shows the following errors each time I try to change the color:
```
Zigbee2MQTT:error 2024-01-17 20:41:04: Publish 'set' 'color' to '0x9035eafffe5aeb38' failed: 'ReferenceError: colorModeLookup is not defined'
```
Can I do something to get the lamp supported properly?

Koenkk, 2024-01-17:
Updated https://gist.github.com/Koenkk/a3d7b568e070cd58150d15b4ed0e870d again.

Nyarly360, 2024-01-17:
I still get errors:
```
Zigbee2MQTT:error 2024-01-17 20:57:38: Publish 'set' 'color' to '0x9035eafffe5aeb38' failed: 'TypeError: Cannot read properties of undefined (reading 'HS')'
Zigbee2MQTT:error 2024-01-17 20:57:59: Publish 'set' 'color' to '0x9035eafffe5aeb38' failed: 'TypeError: Cannot read properties of undefined (reading 'HS')'
Zigbee2MQTT:error 2024-01-17 20:58:03: Publish 'set' 'brightness' to '0x9035eafffe5aeb38' failed: 'TypeError: Cannot read properties of undefined (reading 'ColorTemp')'
Zigbee2MQTT:error 2024-01-17 20:58:13: Publish 'set' 'color_temp' to '0x9035eafffe5aeb38' failed: 'TypeError: Cannot read properties of undefined (reading 'ColorTemp')'
```

Nyarly360, 2024-01-19:
Thanks. However, I still can't control the lamp. Here's the log output from trying to change brightness, white temps and colors:
```
Zigbee2MQTT:error 2024-01-19 07:28:35: Publish 'set' 'brightness' to '0x9035eafffe5aeb38' failed: 'ReferenceError: libColor is not defined'
Zigbee2MQTT:error 2024-01-19 07:28:37: Publish 'set' 'color_temp' to '0x9035eafffe5aeb38' failed: 'ReferenceError: libColor is not defined'
Zigbee2MQTT:error 2024-01-19 07:28:42: Publish 'set' 'color' to '0x9035eafffe5aeb38' failed: 'ReferenceError: libColor is not defined'
```
Switching the lamp off and on via Z2M does not produce a log entry.

Koenkk, 2024-01-19:
Updated https://gist.github.com/Koenkk/a3d7b568e070cd58150d15b4ed0e870d again, hopefully this is it.

Nyarly360, 2024-01-19:
Sorry, but after start-up there's already an error in the log:
```
Zigbee2MQTT:error 2024-01-19 20:05:21: Failed to load external converter file 'ext_converter.js' (Cannot find module 'zigbee-herdsman-converters/lib/libColor'
Zigbee2MQTT:error 2024-01-19 20:05:21: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
```
I still can't control the lamp's on/off status and the white LEDs won't turn off. Although the colors "shine through" the white light more than before. Changing brightness appears to change color temp also. But the lamp gets brighter and darker, if I change the brightness setting. So - progress, but not there yet.

Koenkk, 2024-01-20:
Updated again!

Nyarly360, 2024-01-20:
Hi, it appears that the file didn't change. At least the error message is the same (as well as the file size and control behavior):
```
Zigbee2MQTT:error 2024-01-20 12:56:15: Failed to load external converter file 'ext_converter.js' (Cannot find module 'zigbee-herdsman-converters/lib/libColor'
Zigbee2MQTT:error 2024-01-20 12:56:15: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
```

Koenkk, 2024-01-21:
@SmartHomeNewbie since you added this device in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4059, does it work fine for you?

SmartHomeNewbie, 2024-01-22:
> @SmartHomeNewbie since you added this device in [Koenkk/zigbee-herdsman-converters#4059](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4059), does it work fine for you?

Sorry i dont have the device anymore so iam unable to test

Nyarly360, 2024-01-24:
@Koenkk Can I do something to help get this device integrated correctly?

Koenkk, 2024-01-25:
@Nyarly360 I'm afraid the only thing to do here is to [sniff](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html) the traffic when controlling it through the original gateway

Nyarly360, 2024-01-26:
I just read that the lamp works with deconz, hue and - obviously - the lidl gateway. I don't have either of those, but maybe I'll get a hue or lidl gateway and another zigbee controller for sniffing. But I first have to decide, whether I'll keep the lamp or return it.

Koenkk, 2024-01-27:
They are already using the exact same code (if you are not using an external converter). It could also be that your device is defective (since you are the only one to report this issue yet)

Nyarly360, 2024-01-30:
You were right! I bought another lamp and the new one works as intended. I never would've guessed that the problems I had were due to a defect. Thank you very much!