Koenkk/zigbee2mqtt #20774: TuYa TS0601_air_quality_sensor doesnt corretly represents negative temperature

onhala, 2024-01-13:
### What happened?

If the temperature goes under 0C the values are not corretly parsed. It looks like an error in number representation. 

Current temperature is around -4 °C  but it shows 6550.9 °C.

[image]

### What did you expect to happen?

Correct the implementation to show parse the value properly.

### How to reproduce it (minimal and precise)

Connect TuYa TS0601_air_quality_sensor to Zigbee2MQTT and place the device in room with temperature under 0 °C.

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20221226

### Adapter

smlight-slzb-05 over ehternet

### Setup

Part of Loxberry -> Zigbee2Mqtt Plugin / running on Pi

MikeTolkachev, 2024-01-17:
1.35.1-dev commit: [ab9124a] same behavior

MikeTolkachev, 2024-01-17:
Ok, 6553.5 = -0.1

Koenkk, 2024-01-18:
Should be fixed now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-10-07 by github-actions[bot]