Koenkk/zigbee2mqtt #20778: Exception while calling fromZigbee converter - z2mqtt stops working

diaz66, 2024-01-13:
### What happened?

- After switching to version 1.35.1-1 I started to encounter the error in question.
- From this moment on, zigbee2mqtt no longer sends anything to home assistant (Mosquitto broker)
- The device generating the error is a [PJ-MGW1203](https://www.zigbee2mqtt.io/devices/PJ-MGW1203.html#tuya-pj-mgw1203)
- The system block occurs 2/3 times a day it's more , randomly

My configuration
homeassistant: true
mqtt:
server: mqtt://core-mosquitto
user: ***
password: ***
serial:
port: /dev/ttyACM0
adapter: ezsp
frontend:
port: 8099
advanced:
homeassistant_legacy_entity_attributes: false
legacy_api: false
legacy_availability_payload: false
log_syslog:
app_name: Zigbee2MQTT
eol: /n
host: localhost
localhost: localhost
path: /dev/log
pid: process.pid
port: 123
protocol: tcp4
type: '5424'
channel: 25
log_level: info
device_options:
legacy: false
devices:
'0xe0798dfffea826ff':
friendly_name: Router ZBDongle-E
'0x003c84fffef0ecca':
friendly_name: ZigBee Termosifone Camera Mat
'0xa4c1380483ecb1b5':
friendly_name: ZigBee Termometro Sala
'0xa4c13878a044e674':
friendly_name: ZigBee Termometro Soggiorno
'0xa4c138fa78091d52':
friendly_name: ZigBee Presa Frigorifero
'0xa4c138bab31104b2':
friendly_name: ZigBee Luce Terrazzo Camera Matrimoniale
'0xa4c13831eaf7dbba':
friendly_name: ZigBee Tapparella1 Sala
'0xa4c138c3b27cee34':
friendly_name: ZigBee Lampada Luna
'0xa4c138bcec51d379':
friendly_name: ZigBee Luce Sala
'0xa4c1383aad8d029e':
friendly_name: ZigBee Presa Sala TV
'0xa4c138e52b567c58':
friendly_name: ZigBee Presa Camera Grande Play
'0xa4c138fe60c0d996':
friendly_name: ZigBee Presa Camera Grande PC
'0xa4c138ddd7cffecf':
friendly_name: ZigBee Presa Soggiorno TV
'0xa4c1388b23a3bf79':
friendly_name: ZigBee Porta Frigo
'0xa4c1382e0495eafe':
friendly_name: ZigBee Porta Congelatore
'0xa4c138a06155b4d5':
friendly_name: ZigBee Luce Camera Grande
power_precision: 1
[... 150 lines cut ...]
    at /app/node_modules/@serialport/stream/dist/index.js:118:18
2024-01-13T21:49:15.768Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 2
2024-01-13T21:49:15.768Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
2024-01-13T21:49:15.768Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 2
2024-01-13T21:49:15.769Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
2024-01-13T21:49:15.769Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 2 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/@serialport/stream/dist/index.js:118:18
2024-01-13T21:49:15.770Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 2 error: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/@serialport/stream/dist/index.js:118:18
2024-01-13T21:49:15.788Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 5
2024-01-13T21:49:15.789Z zigbee-herdsman:adapter:ezsp:erro Reset error Error: Failure to connect
    at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:277:19)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:139:9)
    at Driver.onReset (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:111:17)
2024-01-13T21:49:15.789Z zigbee-herdsman:adapter:ezsp:driv Pause 30sec before try 2
2024-01-13T21:49:16.785Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 5
2024-01-13T21:49:16.786Z zigbee-herdsman:adapter:ezsp:erro Reset error Error: Failure to connect
    at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:277:19)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:139:9)
    at Driver.onReset (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:111:17)
2024-01-13T21:49:16.786Z zigbee-herdsman:adapter:ezsp:driv Pause 60sec before try 3

danieledwardgeorgehitchcock, 2024-01-14:
Hi,

Instead of using the devices symlinked path `/dev/ttyACM0` try using its full address path from `/dev/serial/by-id/`

diaz66, 2024-01-14:
Thanks for the reply.

 ```
port: >-
  /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221031111922-if00
```

...and updated version to 1.35.1-1

Let's see what happens.

However, I confirm that with version 1.34.0-1 everything is fine.

diaz66, 2024-01-15:
> Hi,
> 
> From a brief look over your logs, the issue seems more to to with Z2M being unable to communicate with your coordinator adapter.
> 
> Instead of using the devices symlinked path `/dev/ttyACM0` try using its full address path from `/dev/serial/by-id/`
> 
> It also seems strange that you are getting a `cannot lock port` error - this usually occurs if you have a competing resource trying to use the same adapter..

Okay. since I changed `/dev/ttyACM0 `to `/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221031111922-if00 `the problem no longer occurred.

diaz66, 2024-01-19:
It looks quite similar to this [one](https://github.com/Koenkk/zigbee2mqtt/issues/20868)

droans, 2024-01-19:
> It looks quite similar to this [one](https://github.com/Koenkk/zigbee2mqtt/issues/20868)

Seems likely. Does `dmesg` show similar messages as in my issue?

Yves-Be, 2024-04-27:
I've been running 1.36.1 for a week now, and I didn't have this issue anymore.

diaz66, 2024-05-01:
> I've been running 1.36.1 for a week now, and I didn't have this issue anymore.

not so good for me.
Now:

```
error 2024-04-30 22:22:11: Exception while calling fromZigbee converter: ZCL command 0x4c5bb3fffe2eeaf7/1 manuSpecificTuya.mcuGatewayConnectionStatus({"payloadSize":1,"payload":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)}
error 2024-04-30 22:22:11: Exception while calling fromZigbee converter: ZCL command 0x4c5bb3fffe2eeaf7/1 manuSpecificTuya.mcuGatewayConnectionStatus({"payloadSize":1,"payload":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)}
error 2024-04-30 22:22:11: Exception while calling fromZigbee converter: ZCL command 0x4c5bb3fffe2eeaf7/1 manuSpecificTuya.mcuGatewayConnectionStatus({"payloadSize":1,"payload":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)}
error 2024-04-30 22:22:11: Exception while calling fromZigbee converter: ZCL command 0x4c5bb3fffe2eeaf7/1 manuSpecificTuya.mcuGatewayConnectionStatus({"payloadSize":1,"payload":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)}
error 2024-04-30 22:22:11: Exception while calling fromZigbee converter: ZCL command 0x4c5bb3fffe2eeaf7/1 manuSpecificTuya.mcuGatewayConnectionStatus({"payloadSize":1,"payload":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)}
error 2024-04-30 22:22:11: Exception while calling fromZigbee converter: ZCL command 0x4c5bb3fffe2eeaf7/1 manuSpecificTuya.mcuGatewayConnectionStatus({"payloadSize":1,"payload":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)}
```
....**and the device stops sending data**....it starts sending data again only after device reconfiguration....
`

Yves-Be, 2024-05-01:
Strange, everything else seems the same, apart from me using channel 11, and my ZB-GW04 USB dongle that i use on /dev/ttyUSB0 instead of /dev/ttyACM0. It's also still on 6.10.3.0 build 297.
I didn't apply any calibrations, but I think that shouldn't matter either.

Yves-Be, 2024-05-22:
I did get a crash 2 days later, but I've since been using 1.37.x with the ember adapter and 7.4.2 FW instead of ezsp.
No issues after more than two and a half weeks, so I'd give that a go.
Flashing the stick was pretty easy, and the network still worked without having to re-pair anything.
Good guide [here](https://smarthomescene.com/guides/how-to-use-ember-driver-with-efr32mg21-zigbee-adapters/).

closed 2024-12-30 by github-actions[bot]