Koenkk/zigbee2mqtt #20787: Child lock disapeared

Blackduke77, 2024-01-14:
### What happened?

I am new to Z2M and Ha, I have been working on UK power sockets and have tested most that suit my usecases. I settled on a mixture of ClickSmart+ and Aone's working with Z2M as this gave the best experience and usability. however when I came to re-test the click smart I noticed the child lock had gone. I am sure it was there during my testing but now it is not available. I thought I was going mad and researched again, these devices sport it and there is an old feature request to add the support that was merged.

I am not sure if this was recently broken. The old feature request is https://github.com/Koenkk/zigbee2mqtt/issues/15684
This was for the double socket, I am using the single at the moment, but will also deploy some doubles. 

This is the socket https://www.zigbee2mqtt.io/devices/CMA30035.html#clicksmart%252B-cma30035

Would someone mind helping get this fixed.

### What did you expect to happen?

Option available to disable local operation of the button switch

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonof

### Setup

Z2M HA addon

Blackduke77, 2024-01-25:
I have created a converter that seems to work, this has power, brightness and child lock. Not sure how this gets into main support
https://github.com/Blackduke77/Home-Assistant/blob/main/Zigbee2MQTT/Converters/DoubleSocket50AU.js

Koenkk, 2024-01-27:
Could you create a PR for this? To do so click [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/aurora_lighting.ts)

closed 2024-08-26 by github-actions[bot]