Koenkk/zigbee2mqtt #20789: Bticino's OTA updates not work

leoogermenia, 2024-01-14:
### What happened?

I'm trying to update my switch (K4003C), but I recived multiple errors. Sometimes I can start the update, but when it arrives to 98/99% it fails

### How to reproduce it (minimal and precise)

Start update for the device

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ITead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

### Setup

Add-on on HA (Proxmox VM)

### Debug log

Update of 'Lampadario Cameretta' failed (OTA: Timeout, device did not request any image blocks)

syntesys87, 2024-02-06:
I also have the same problem.

leoogermenia, 2024-02-07:
> I also have the same problem.

Hi, at the end I was able to update it. Just try to click lots of time the update button and it updates

syntesys87, 2024-02-07:
I tried one month clicking the update button of every device without solution..

closed 2024-09-05 by github-actions[bot]