Koenkk/zigbee2mqtt #20791: TS130F doesn't use calibration

ivrlx, 2024-01-14:
### What happened?

I have four TS130F connected to Z2M and HA. They're calibrated to 10 seconds, but every move they make is always 10secs move no matter what position you say them to go. If they're closed and I tell them to go to 50% they move the entire calibration time (10 secs), if I say to go to 75% after that they move another 10 secs.

### What did you expect to happen?

They should move the correct seconds in accordance with the position they are in that moment. If I have the device calibrated to 10 secs and I want to close to 50% the should move 5 secs.

### How to reproduce it (minimal and precise)

Giving position through HA or Z2M

### Zigbee2MQTT version

 1.35.1

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 USB Dongle-P

### Setup

Add-on on HA OS

### Debug log

Info 2024-01-14 18:53:52Successfully changed options
Info 2024-01-14 18:53:52MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"ybdmf-1"}'
Debug 2024-01-14 18:54:00Received MQTT message on 'zigbee2mqtt/Persiana Despacho/set' with data '{"position":50}'
Debug 2024-01-14 18:54:00Publishing 'set' 'position' to 'Persiana Despacho'
Info 2024-01-14 18:54:00MQTT publish: topic 'zigbee2mqtt/Persiana Despacho', payload '{"backlight_mode":"ON","calibration":"OFF","calibration_time":12,"indicator_mode":"off","linkquality":75,"motor_reversal":"OFF","moving":"STOP","position":50,"state":"CLOSE"}'
Debug 2024-01-14 18:54:00Received Zigbee message from 'Persiana Despacho', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50,"tuyaMovingState":0}' from endpoint 1 with groupID 0
Info 2024-01-14 18:54:00MQTT publish: topic 'zigbee2mqtt/Persiana Despacho', payload '{"backlight_mode":"ON","calibration":"OFF","calibration_time":12,"indicator_mode":"off","linkquality":75,"motor_reversal":"OFF","moving":"UP","position":50,"state":"OPEN"}'
Debug 2024-01-14 18:54:12Received Zigbee message from 'Persiana Despacho', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":50,"tuyaMovingState":1}' from endpoint 1 with groupID 0
Info 2024-01-14 18:54:12MQTT publish: topic 'zigbee2mqtt/Persiana Despacho', payload '{"backlight_mode":"ON","calibration":"OFF","calibration_time":12,"indicator_mode":"off","linkquality":24,"motor_reversal":"OFF","moving":"STOP","position":50,"state":"OPEN"}'

ivrlx, 2024-01-22:
I've updated the firmware version to the last version (20230507) and it's still happening.

tajnymag, 2024-05-22:
How have you solved it?

blackwiz4rd, 2024-06-11:
Hello,
I am not sure if it works because I still don't have a motor to test, but it seems that if you turn on calibration:
`zigbee2mqtt/FRIENDLY_NAME/set`
`{"calibration": "ON"}`
then suppose you had UP/DOWN, bring the shutter on the opposite side
then turn off calibration
`zigbee2mqtt/FRIENDLY_NAME/set`
`{"calibration": "OFF"}`

it should set the calibration time on it's own.

closed 2024-12-30 by github-actions[bot]