Koenkk/zigbee2mqtt #20795: ZNCLBL01LM unreliability

marcgarciamarti, 2024-01-14:
### What happened?

I purchased 4 ZNCLBL01LM devices during last black friday campaign. They have not performed reliably since I paired them to my coordinator. They seem to work fine for a couple of days, but then it looks like they drop out of the network, because they no longer seem to react to home assistant commands

This is the log of a freshly paired device. Compare this with the logs from devices in error, as shown below:

```
Info 2024-01-14 21:48:11MQTT publish: topic 'zigbee2mqtt/curtain1xiaomi', payload '{"action":null,"battery":100,"charging":false,"device_temperature":25,"hand_open":true,"hooks_lock":"LOCK","hooks_state":"locked","illuminance_lux":0,"linkquality":240,"motor_state":"stopped","position":10,"power_outage_count":0,"power_source":"battery","running":false,"state":"STOP","target_position":100,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3000}'
Info 2024-01-14 21:48:11MQTT publish: topic 'zigbee2mqtt/curtain1xiaomi', payload '{"action":null,"battery":100,"charging":false,"device_temperature":25,"hand_open":true,"hooks_lock":"LOCK","hooks_state":"locked","illuminance_lux":0,"linkquality":240,"motor_state":"stopped","position":10,"power_outage_count":0,"power_source":"battery","running":false,"state":"STOP","target_position":21,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3000}'
Info 2024-01-14 21:48:11MQTT publish: topic 'zigbee2mqtt/curtain1xiaomi', payload '{"action":null,"battery":100,"charging":false,"device_temperature":25,"hand_open":true,"hooks_lock":"LOCK","hooks_state":"locked","illuminance_lux":0,"linkquality":240,"motor_state":"stopped","position":21,"power_outage_count":0,"power_source":"battery","running":false,"state":"OPEN","target_position":21,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3000}'
```

### What did you expect to happen?

devices continue to operate normal

### How to reproduce it (minimal and precise)

pair a device and wait for a couple of days

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Home Assistant Add-On

### Debug log

Error 2024-01-14 21:48:19Publish 'set' 'state' to 'curtain3xiaomi' failed: 'Error: Command 0x54ef4410004fe847/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":100}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 33808 - 1 - 46 - 258 - 11 after 10000ms)'
Info 2024-01-14 21:48:19MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'curtain3xiaomi' failed: 'Error: Command 0x54ef4410004fe847/1 closuresWindowCovering.goToLiftPercentage({\"percentageliftvalue\":100}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 33808 - 1 - 46 - 258 - 11 after 10000ms)'","meta":{"friendly_name":"curtain3xiaomi"},"type":"zigbee_publish_error"}'
Error 2024-01-14 21:48:18Publish 'set' 'state' to 'curtain2xiaomi' failed: 'Error: Command 0x54ef4410006bf2f3/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 62576 - 1 - 45 - 258 - 11 after 10000ms)'
Info 2024-01-14 21:48:18MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'curtain2xiaomi' failed: 'Error: Command 0x54ef4410006bf2f3/1 closuresWindowCovering.goToLiftPercentage({\"percentageliftvalue\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 62576 - 1 - 45 - 258 - 11 after 10000ms)'","meta":{"friendly_name":"curtain2xiaomi"},"type":"zigbee_publish_error"}'Error 2024-01-14 21:48:20Publish 'set' 'state' to 'curtain4xiaomi' failed: 'Error: Command 0x54ef4410004ffde9/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":100}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 46403 - 1 - 47 - 258 - 11 after 10000ms)'
Info 2024-01-14 21:48:20MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'curtain4xiaomi' failed: 'Error: Command 0x54ef4410004ffde9/1 closuresWindowCovering.goToLiftPercentage({\"percentageliftvalue\":100}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 46403 - 1 - 47 - 258 - 11 after 10000ms)'","meta":{"friendly_name":"curtain4xiaomi"},"type":"zigbee_publish_error"}'

marcgarciamarti, 2024-01-21:
@Koenkk in the past you've been very helpful and key to solving any issue. I wonder if you could take a look at this issue and let me know if further information may be needed?

marcgarciamarti, 2024-01-23:
Just rebooted my instance and clicked on device closing on HA front-end and got this error in Z2M add-on log messages: 

```
Error 2024-01-23 21:37:22Publish 'set' 'state' to 'curtain2xiaomi' failed: 'Error: Command 0x54ef4410006bf2f3/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 62576 - 1 - 104 - 258 - 11 after 10000ms)'
```

marcgarciamarti, 2024-02-03:
@Koenkk any idea what may the issue with my 4 devices? I have everything running latest versions but these devices and only these devices drop out of my network after a couple of days upon pairing. Any hint as to how to troubleshoot this case would be most appreciated!

Koenkk, 2024-02-03:
@marcgarciamarti this may be to an incompatible router, you can try to keep them very close to the coordinator and see if they still drop.

marcgarciamarti, 2024-02-03:
> @marcgarciamarti this may be to an incompatible router, you can try to keep them very close to the coordinator and see if they still drop.

The 4 curtain motors are all within a 3m range from my coordinator but upon checking the network map, I see that a freshly paired curtain motor connects to the networks through a router that sits in between it and the coordinator. 

This is interesting because I have like 35 devices and all of them work beautifully, and many of them connect through that same potentially problematic router.

@Koenkk is there any way I can tell these devices to avoid a certain router? I'm positive they could do without a router and simply attempt to connect to the coordinator directly.

Thanks a lot for your support. It looks like this could definitely be the case.

Koenkk, 2024-02-03:
> is there any way I can tell these devices to avoid a certain router? 

unfortunately not possible in Zigbee

marcgarciamarti, 2024-02-03:
> > is there any way I can tell these devices to avoid a certain router?
> 
> unfortunately not possible in Zigbee

What a bummer! Thanks again for your support. I may try to work this around by putting yet another router even closer to these devices hoping to have them use the new router instead. 

Thanks again!!

marcgarciamarti, 2024-02-09:
@Koenkk I disconnected and removed the router this device was hanging from, and none of the end devices that were getting on the network through that router seem to connect to the network anymore. I rebooted the host and Unplugged the coordinator, but still none of them would work anymore. 

I was expecting all these end devices would transparently transition to the next best performing router once the allegedly problematic one was removed from the network, but that did not happen.

Have you experienced this behavior before? Do you have any suggestion?

Thanks!

Koenkk, 2024-02-11:
For end device it can take some time until they switch router, it may help to power cycle them (re-insert battery)

marcgarciamarti, 2024-02-23:
> For end device it can take some time until they switch router, it may help to power cycle them (re-insert battery)

It looks like you hit the nail right on its head. I removed the noisy router from the network and these devices seem to work more reliably.

I'm impressed @Koenkk !!

marcgarciamarti, 2024-07-22:
> > is there any way I can tell these devices to avoid a certain router?
> 
> unfortunately not possible in Zigbee

HI Koenk,

still wrestling with these devices picking up the wrong router. I revisited this issue because I wanted to re-read your statements, and I'm at a loss. None of my other devices has any issues with any of my current routers, however, these curtain devices act up frequently. Most of the time 3 of the 4 simply exit the network and remain in a black hole until I get fed up with them and power cycle them.

Question: Given that I can't chose what router my devices pick up, nor can I tell my problematic devices to avoid a certain router, I wonder if there is any way I can at least mark the problematic routers in a way in which they simply cease to work as such?

If the answer is consistent with the above, the only I have is to gradualy ditch my existing routers and that feels disproportionate... am I missing any other option?

thanks in advance!!

Koenkk, 2024-07-23:
> I wonder if there is any way I can at least mark the problematic routers in a way in which they simply cease to work as such?

This is not possible, the coordinator (Zigbee2MQTT) cannot change the device type of a device

marcgarciamarti, 2025-01-06:
@Koenkk I know this is an old thread but issues continue to happen for me. I've been monitoring these devices for a while and I found they end up abandoning the network. But not all of them. There is typically one of them that sticks to the network, but the other 3 (I have 4) randomly fall off the network. See for example this snapshot from my network map. Any idea as to why these devices, tend to either connect with the wrong router or jump off the network.

[image]

Hotelk52339, 2025-02-23:
One possible solution to your issue with Zigbee devices frequently disconnecting is to change the Zigbee channel in Zigbee2MQTT. The problem might be that your Zigbee channel overlaps with your Wi-Fi network, causing interference.  

If you have a mesh Wi-Fi network with multiple routers, there's a higher chance of channel overlap, which could be affecting the stability of your Zigbee devices. Try switching to a less congested Zigbee channel and see if that helps

closed 2025-05-03 by github-actions[bot]