Koenkk/zigbee2mqtt #20797: Lidl light bulb HG06492C/HG08130C/HG09154C - Do not disturb option not work

antti365, 2024-01-14:
### What happened?

Lidl light bulb HG06492C/HG08130C/HG09154C - when the Do not disturb option is enabled, the light bulb continues to turn on after a power failure

### Zigbee2MQTT version

1.34.0 commit: 56589dc

### Adapter firmware version

20210319

### Adapter

CC2652P

### Setup

plain

wuuy, 2024-01-17:
The same for Lidl **HG06463A** and **HG06463B** (Lidl filament bulbs E27) visible as

TS0501A
_TZ3000_nosnx7im
Livarno Lux E27 G95 filament bulb
and 
TS0501A
_TZ3000_j2w1dw29
Livarno Lux E27 ST64 filament bulb

Zigbee2MQTT Current version: 1.35.1-1 
(didn't work properly with the older version either) 

Regards
Tommy

BernardOtto, 2024-01-19:
Same for [HG08010]  Lidl Flood light.
TS0505B
_TZ3210_umi6vbsz

Zigbee2MQTT [1.35.1]

Is there maybe a trick to activate this do not disturb mode? 
After switichg the toggle, like turning the light on/off a few times? Which we miss?

triggertreyy, 2024-01-30:
Yep im facing the same issue. Strange is that the RGB Version TS0505B do not disturb is working AS it should

antti365, 2024-01-31:
@Koenkk Hi, maybe you can watch it when you have time, and at the same time this too... https://github.com/Koenkk/zigbee2mqtt/issues/21085#issue-2104185322

Koenkk, 2024-01-31:
Note sure if this device support it, is `do_not_disturb` supported when using the original hub?

antti365, 2024-01-31:
Unfortunately, I don’t have the original hub, maybe one of the guys has it?

BernardOtto, 2024-01-31:
Me neither, but I asume that people would complain in the lidl e-shop reviews, if the light would stay on after a power failure. I will look for an used hub on e-bay.

slima, 2024-02-01:
Same for TS0502A / [HG06492A/HG08130A](https://www.zigbee2mqtt.io/devices/HG06492A_HG08130A.html#lidl-hg06492a%252Fhg08130a)

wuuy, 2024-02-02:
Hi,
 
Sorry for the delay, the message landed in the spam folder. Thx for your reply.
I don't have the original hub.
 
I just checked the manual, it doesn't mention anything about the behavior after power failure. But it's short and advises to install the app without describing the app's action. So there's no prove if do_no_disturb should work.
 
I have 6 of such Lidl bulbs. All behave the same way. Resetting them (switching power on and off) doesn't change their do_no_disturb behavior.
 
That's all I can tell. 
 
Regards
Tommy

antti365, 2024-02-04:
But at least it should support transition... or not? This is not supported for me either.

BernardOtto, 2024-02-29:
So, I have managed to buy a discounted Lidl HUB, connected the [HG08010] Lidl Flood light, and I can't find any "do not disturb" or "power on" features.  I also see that there is no firmware update for this Light, the current one is V1.0.0

We should write their support, that this is a mandatory function, maybe they will release an firmware update.

KentuckyMC, 2024-04-23:
Seems like also not working on the TS0505A Ledstrip.

closed 2024-11-20 by github-actions[bot]