Koenkk/zigbee2mqtt #20801: [New device support]: tuya dimmer tuch switch

ssrsmart, 2024-01-14:
### Link

https://x.alibaba.com/AvV1Ji?ck=pdp

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0xf082c0fffe1895ef","nwkAddr":41641,"manufId":4098,"manufName":"_TZE204_9qhuzgo0","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_9qhuzgo0","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1705273727921,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,
 I bought tuya zigbee 1gang dimmer wall switch (touch type), the dives discovered quickly but with no entity and it's showing as not supported, I tried to support it by external converter with no success

### External defintion

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_9qhuzgo0',
    description: 'Automatically generated definition',
    extend: [],
};

module.exports = definition
```

Koenkk, 2024-01-15:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/c98bfafb7d4f8e22af403db39cbb2862): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

ssrsmart, 2024-01-16:
now the device is supported, the on/off working well with a feedback but dimmer control not work, when I change the brightness the light just flicker once, also there is a problem with the min/max brightness , when I tried to set it the dimmer control works well from the physical buttons but after repowered the device or trying to change the brightness from home assistant it loss the settings and the physical brightness control become only one step on/off

zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0xf082c0fffe1895ef with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,2,111]}}': type='commandDataReport', datatype='value', value='623', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","nousHumidity","tthHumidity","tIlluminanceLux","evanellMode","AM02PercentControl","x5hMode","connecteMode","tshpscSensitivity","alectoSmokeValue","zsMode"]

ssrsmart, 2024-01-16:
> Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/c98bfafb7d4f8e22af403db39cbb2862):
>
> * save this as file next to `configuration.yaml` as [...]

now the device is supported, the on/off working well with a feedback but dimmer control not work, when I change the brightness the light just flicker once, also there is a problem with the min/max brightness , when I tried to set it the dimmer control works well from the physical buttons but after repowered the device or trying to change the brightness from home assistant it loss the settings and the physical brightness control become only one step on/off

zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint https://github.com/Koenkk/zigbee2mqtt/issues/2 from 0xf082c0fffe1895ef with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,2,111]}}': type='commandDataReport', datatype='value', value='623', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","nousHumidity","tthHumidity","tIlluminanceLux","evanellMode","AM02PercentControl","x5hMode","connecteMode","tshpscSensitivity","alectoSmokeValue","zsMode"]

Koenkk, 2024-01-17:
Updated https://gist.github.com/Koenkk/c98bfafb7d4f8e22af403db39cbb2862, can you try again?

ssrsmart, 2024-01-18:
> Updated https://gist.github.com/Koenkk/c98bfafb7d4f8e22af403db39cbb2862, can you try again?

now everything works fine on/off and dimmer control with feedback, the only thing I noticed is the Power-on behavior always off regardless the mode being set

Koenkk, 2024-01-18:
Integrated the changes, you can remove the external converter after updating to the next release.

closed 2024-01-18 by Koenkk

ssrsmart, 2024-01-18:
> Integrated the changes, you can remove the external converter after updating to the next release.

I couldn't have gotten the device to work without your help,
thank you.