Koenkk/zigbee2mqtt #20802: NWK_TABLE_FULL after upgrading to CC2652RB_coordinator_20230507.hex

Piscator74, 2024-01-14:
### What happened?

I just decided to upgrade the FW on my slae.sh stick today from CC2652RB_coordinator_20221226.hex to CC2652RB_coordinator_20230507.hex. Now, on every command I run i get a NWK_TABLE_FULL error. I can still track all zigbee activity, but it is no longer possible to give any commands. Also, when I downgrade back to 20221226, I have the same problem now.
I have 92 devices running.

Any suggestions how I can fix this? 
```
Zigbee2MQTT:error 2024-01-15 00:23:40: Publish 'set' 'state' to 'bseedSwitch_gf_garage_front2WL' failed: 'Error: Command 0xa4c138d8af309db2/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":44695,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'
Zigbee2MQTT:info  2024-01-15 00:23:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'bseedSwitch_gf_garage_front2WL' failed: 'Error: Command 0xa4c138d8af309db2/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SREQ '--> ZDO - extRouteDisc - {\"dstAddr\":44695,\"options\":0,\"radius\":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'","meta":{"friendly_name":"bseedSwitch_gf_garage_front2WL"},"type":"zigbee_publish_error"}'
```

### Zigbee2MQTT version

1.35.0

### Adapter firmware version

20230507

### Adapter

Slaesh's CC2652RB stick

### Setup

In a docker container

Piscator74, 2024-01-15:
I managed to fix it by trial and error.

I restored a working setup by doing the following.
* Reflash firmware with CC2652RB_coordinator_20221226.hex
* Use a backup of zigbee2mqtt setup (without this, it is not working).

I was reading up on pan_id's and since I only have an ext_pan_id defined in my yaml, I thought that perhaps this was the problem.
* Enabled log_level debug.
* Got the pan_id and configured it in yaml.
* Reflash firmware with CC2652RB_coordinator_20230507.hex
--> **Not ok**. all zigbee commands again give some kind of error. Mostly: "Timeout" and "No network route (205):.

Next thing I tried:
* Restored the setup (as described above).
* Made a read of the NVRAM (using ZigStarGW-MT)
* Reflash firmware with CC2652RB_coordinator_20230507.hex
* Made a write of the NVRAM (using ZigStarGW-MT)
--> Now everything works fine with the upgraded firmware

Anyone has an idea why this works and why this is needed?

Koenkk, 2024-01-15:
Soon I will publish a new firmware which should fix the `NWK_TABLE_FULL` (or at least reduce them). Keep an eye on https://github.com/Koenkk/Z-Stack-firmware/discussions

Piscator74, 2024-01-15:
Thanks, I will be keeping an eye out for it.

closed 2024-08-13 by github-actions[bot]