Koenkk/zigbee2mqtt #20803: TS011F_plug_3 loses connection after energy reset command

VladNord, 2024-01-15:
### What happened?

I have TS011F_plug_3 and automation, which should reset energy sensor every day
here is part code
```
service: mqtt.publish
  data:
    topic: >-
      zigbee2mqtt/teapot/1/set
    payload_template: >-
      {
        "command": {
          "cluster": 0,
          "command": 0,
          "payload": {}
          }
      }
```

But after this command plug loses connection with coordinator and i should re-pair it.
Wherein,  Energy Data remains the same.
I did the same thing through the interface (according to the manual) same result.

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

0x26720700

### Adapter

ConBee_II 

### Setup

HASS

Koenkk, 2024-01-15:
That command is actually a factory reset, I guess the behaviour of a factory resets differs per TuYa plug (TLDR; not fixable from z2m)

VladNord, 2024-01-16:
> That command is actually a factory reset, I guess the behaviour of a factory resets differs per TuYa plug (TLDR; not fixable from z2m)

ahh, damn. Question is closed. Very disappointing :(

Maybe  you can  advise something: is there some resource where there is a list of zigbee devices that can be used without compromise?
The further into the world of Zigbee, the more I am convinced that there is nothing better than esphome with esp...

Koenkk, 2024-01-16:
> Maybe you can advise something: is there some resource where there is a list of zigbee devices that can be used without compromise?

That's a very general question, so I cannot given an answer to that.

closed 2024-08-14 by github-actions[bot]