Koenkk/zigbee2mqtt #20807: No devices pair or logs

BinaryShrub, 2024-01-15:
### What happened?

I had an issue with my network, rebuilt my configuration.yaml to create a new network and nothing will pair

### What did you expect to happen?

```
Zigbee2MQTT:info  2024-01-14 21:23:50: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2024-01-14 21:23:51: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"4cvid-3"}'
```

would be followed by pair requests but nothing comes through an any devices even right next to the coordinator

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Docker

### Debug log

info  2024-01-14 21:21:14: Logging to console and directory: '/app/data/log/2024-01-14.21-21-14' filename: log.txt
info  2024-01-14 21:21:14: Starting Zigbee2MQTT version 1.35.1 (commit #9eaaa0f)
info  2024-01-14 21:21:14: Starting zigbee-herdsman (0.30.0)
info  2024-01-14 21:22:11: zigbee-herdsman started (reset)
info  2024-01-14 21:22:11: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20230507,"transportrev":2},"type":"zStack3x0"}'
info  2024-01-14 21:22:11: Currently 0 devices are joined:
warn  2024-01-14 21:22:11: `permit_join` set to  `true` in configuration.yaml.
warn  2024-01-14 21:22:11: Allowing new devices to join.
warn  2024-01-14 21:22:11: Set `permit_join` to `false` once you joined all devices.
info  2024-01-14 21:22:11: Zigbee: allowing new devices to join.
info  2024-01-14 21:22:12: Connecting to MQTT server at mqtt://mqtt
info  2024-01-14 21:22:12: Connected to MQTT server
info  2024-01-14 21:22:12: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
info  2024-01-14 21:22:12: Started frontend on port 8080
info  2024-01-14 21:22:12: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x00124b0024c0c3b5/connection_state/config', payload '{"device":{"hw_version":"zStack3x0 20230507","identifiers":["zigbee2mqtt_bridge_0x00124b0024c0c3b5"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.1"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"1.35.1","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0x00124b0024c0c3b5_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
info  2024-01-14 21:22:12: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x00124b0024c0c3b5/restart_required/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":"zStack3x0 20230507","identifiers":["zigbee2mqtt_bridge_0x00124b0024c0c3b5"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.1"},"device_class":"problem","enabled_by_default":false,"entity_category":"diagnostic","name":"Restart required","object_id":"zigbee2mqtt_bridge_restart_required","origin":{"name":"Zigbee2MQTT","sw":"1.35.1","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0x00124b0024c0c3b5_restart_required_zigbee2mqtt","value_template":"{{ value_json.restart_required }}"}'
info  2024-01-14 21:22:12: MQTT publish: topic 'homeassistant/button/1221051039810110150109113116116_0x00124b0024c0c3b5/restart/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/bridge/request/restart","device":{"hw_version":"zStack3x0 20230507","identifiers":["zigbee2mqtt_bridge_0x00124b0024c0c3b5"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.1"},"device_class":"restart","name":"Restart","object_id":"zigbee2mqtt_bridge_restart","origin":{"name":"Zigbee2MQTT","sw":"1.35.1","url":"https://www.zigbee2mqtt.io"},"payload_press":"","unique_id":"bridge_0x00124b0024c0c3b5_restart_zigbee2mqtt"}'
info  2024-01-14 21:22:12: MQTT publish: topic 'homeassistant/select/1221051039810110150109113116116_0x00124b0024c0c3b5/log_level/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_template":"{\"options\": {\"advanced\": {\"log_level\": \"{{ value }}\" } } }","command_topic":"zigbee2mqtt/bridge/request/options","device":{"hw_version":"zStack3x0 20230507","identifiers":["zigbee2mqtt_bridge_0x00124b0024c0c3b5"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.1"},"entity_category":"config","name":"Log level","object_id":"zigbee2mqtt_bridge_log_level","options":["info","warn","error","debug"],"origin":{"name":"Zigbee2MQTT","sw":"1.35.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0x00124b0024c0c3b5_log_level_zigbee2mqtt","value_template":"{{ value_json.log_level | lower }}"}'
info  2024-01-14 21:22:12: MQTT publish: topic 'homeassistant/sensor/1221051039810110150109113116116_0x00124b0024c0c3b5/version/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":"zStack3x0 20230507","identifiers":["zigbee2mqtt_bridge_0x00124b0024c0c3b5"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.1"},"entity_category":"diagnostic","icon":"mdi:zigbee","name":"Version","object_id":"zigbee2mqtt_bridge_version","origin":{"name":"Zigbee2MQTT","sw":"1.35.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0x00124b0024c0c3b5_version_zigbee2mqtt","value_template":"{{ value_json.version }}"}'
info  2024-01-14 21:22:12: MQTT publish: topic 'homeassistant/sensor/1221051039810110150109113116116_0x00124b0024c0c3b5/coordinator_version/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":"zStack3x0 20230507","identifiers":["zigbee2mqtt_bridge_0x00124b0024c0c3b5"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.1"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:chip","name":"Coordinator version","object_id":"zigbee2mqtt_bridge_coordinator_version","origin":{"name":"Zigbee2MQTT","sw":"1.35.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0x00124b0024c0c3b5_coordinator_version_zigbee2mqtt","value_template":"{{ value_json.coordinator.meta.revision }}"}'
info  2024-01-14 21:22:12: MQTT publish: topic 'homeassistant/sensor/1221051039810110150109113116116_0x00124b0024c0c3b5/network_map/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":"zStack3x0 20230507","identifiers":["zigbee2mqtt_bridge_0x00124b0024c0c3b5"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.1"},"enabled_by_default":false,"entity_category":"diagnostic","json_attributes_template":"{{ value_json.data.value | tojson }}","json_attributes_topic":"zigbee2mqtt/bridge/response/networkmap","name":"Network map","object_id":"zigbee2mqtt_bridge_network_map","origin":{"name":"Zigbee2MQTT","sw":"1.35.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/bridge/response/networkmap","unique_id":"bridge_0x00124b0024c0c3b5_network_map_zigbee2mqtt","value_template":"{{ now().strftime('%Y-%m-%d %H:%M:%S') }}"}'
info  2024-01-14 21:22:12: MQTT publish: topic 'homeassistant/sensor/1221051039810110150109113116116_0x00124b0024c0c3b5/permit_join_timeout/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":"zStack3x0 20230507","identifiers":["zigbee2mqtt_bridge_0x00124b0024c0c3b5"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.1"},"device_class":"duration","entity_category":"diagnostic","name [... cut ...] ansaction":"4cvid-1"}'
info  2024-01-14 21:23:47: Zigbee: disabling joining new devices.
info  2024-01-14 21:23:48: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"4cvid-2"}'
info  2024-01-14 21:23:50: Zigbee: allowing new devices to join.
info  2024-01-14 21:23:51: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"4cvid-3"}'

mnn, 2024-01-20:
Happened to me on 1.35.1 and ConBee2, had to rollback to 1.34.0. Not sure if exactly the same (I think I didn't even see the MQTT log line, just `allowing new devices to join`). It showed notification in Web UI, but button didn't reflect it. When setting it in config and restarting, I think it didn't even finish loading (Web UI down, from logs it looked like it didn't  connect to MQTT, probably frozen after the pairing log message).

lukaszblhome, 2024-02-10:
I've got the same issue:

Zigbee2MQTT Current version: 1.35.3-1
usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E85A5B-if00

[12:36:20] INFO: Preparing to start...
[12:36:20] INFO: Socat not enabled
[12:36:22] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-02-10 12:36:26: Logging to console and directory: '/config/zigbee2mqtt/log/2024-02-10.12-36-26' filename: log.txt
Zigbee2MQTT:info  2024-02-10 12:36:26: Starting Zigbee2MQTT version 1.35.3 (commit #unknown)
Zigbee2MQTT:info  2024-02-10 12:36:26: Starting zigbee-herdsman (0.33.8)
Zigbee2MQTT:info  2024-02-10 12:36:27: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2024-02-10 12:36:27: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:info  2024-02-10 12:36:27: Currently 0 devices are joined:
Zigbee2MQTT:warn  2024-02-10 12:36:27: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2024-02-10 12:36:27: Allowing new devices to join.
Zigbee2MQTT:warn  2024-02-10 12:36:27: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2024-02-10 12:36:27: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2024-02-10 12:36:27: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:info  2024-02-10 12:36:27: Connected to MQTT server
Zigbee2MQTT:info  2024-02-10 12:36:27: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2024-02-10 12:36:28: Started frontend on port 8099
Zigbee2MQTT:info  2024-02-10 12:36:28: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x00124b0018e85a5b/connection_state/config', payload '{"device":{"hw_version":"zStack12 20190608","identifiers":["zigbee2mqtt_bridge_0x00124b0018e85a5b"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.3"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"1.35.3","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0x00124b0018e85a5b_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Zigbee2MQTT:info  2024-02-10 12:36:28: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x00124b0018e85a5b/restart_required/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":"zStack12 20190608","identifiers":["zigbee2mqtt_bridge_0x00124b0018e85a5b"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.3"},"device_class":"problem","enabled_by_default":false,"entity_category":"diagnostic","name":"Restart required","object_id":"zigbee2mqtt_bridge_restart_required","origin":{"name":"Zigbee2MQTT","sw":"1.35.3","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0x00124b0018e85a5b_restart_required_zigbee2mqtt","value_template":"{{ value_json.restart_required }}"}'
Zigbee2MQTT:info  2024-02-10 12:36:28: MQTT publish: topic 'homeassistant/button/1221051039810110150109113116116_0x00124b0018e85a5b/restart/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/bridge/request/restart","device":{"hw_version":"zStack12 20190608","identifiers":["zigbee2mqtt_bridge_0x00124b0018e85a5b"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.3"},"device_class":"restart","name":"Restart","object_id":"zigbee2mqtt_bridge_restart","origin":{"name":"Zigbee2MQTT","sw":"1.35.3","url":"https://www.zigbee2mqtt.io"},"payload_press":"","unique_id":"bridge_0x00124b0018e85a5b_restart_zigbee2mqtt"}'
Zigbee2MQTT:info  2024-02-10 12:36:28: MQTT publish: topic 'homeassistant/select/1221051039810110150109113116116_0x00124b0018e85a5b/log_level/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_template":"{\"options\": {\"advanced\": {\"log_level\": \"{{ value }}\" } } }","command_topic":"zigbee2mqtt/bridge/request/options","device":{"hw_version":"zStack12 20190608","identifiers":["zigbee2mqtt_bridge_0x00124b0018e85a5b"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.35.3"},"entity_category":"config","name":"Log level","object_id":"zigbee2mqtt_bridge_log_level","options":["info","warn","error","debug"],"origin":{"name":"Zigbee2MQTT","sw":"1.35.3","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0x00124b0018e85a5b_log_level_zigbee2mqtt","value_template":"{{ value_json.log_level | lower }}"}'
Zigbee2MQTT:info  2024-02-10 12:36:28: MQTT publish: topic 'homeassistant/sensor/1221051039810110150109113116116_0x00124b0018e85a5b/version/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":"zStack12 20190608","identifiers":["zigbee2mqtt_bridge_0x00124b0018e85a5b"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge [... cut ...] ct_id":"zigbee2mqtt_bridge_permit_join","origin":{"name":"Zigbee2MQTT","sw":"1.35.3","url":"https://www.zigbee2mqtt.io"},"payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0x00124b0018e85a5b_permit_join_zigbee2mqtt","value_template":"{{ value_json.permit_join | lower }}"}'
Zigbee2MQTT:info  2024-02-10 12:36:28: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2024-02-10 12:36:28: Zigbee2MQTT started!

closed 2024-09-08 by github-actions[bot]