Koenkk/zigbee2mqtt #20808: SNZB-06P is failing to configure, with nothing exposed to HA

aatifk, 2024-01-15:
### What happened?

I just tried to set up SONOFF SNZB-06P. The device gets identified properly but then I get this error message in the logs:

```
Error 2024-01-15 02:19:13Exception while calling fromZigbee converter: Expected one of: 0, 1, got: 'undefined'}
Error 2024-01-15 02:19:13Failed to configure '0x3410f4fffee3a50a', attempt 1 (Error: Read 0x3410f4fffee3a50a/1 64529([8193], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28) at Endpoint.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22) at setupAttributes (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:62:9) at configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:72:9) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:167:17))
```
As of now the device shows as identified and even configured but doesn't report any values (all Null except LQI which is static at 153). What's interesting is that despite this error message, I managed to update firmware to 1.0.5 (it was on 1.0.3 out of the box). I've tried removing the sensor, resetting it and adding it back without any luck. Nothing is getting exposed to Home Assistant either. I'm on the most recent available version of Z2M.

### What did you expect to happen?

For the sensor to work properly as it's a supported device. 

### Zigbee2MQTT version

1.35.1 commit: 3c962042

### Adapter firmware version

20230507

### Adapter

SONOFF USB Zigbee Dongle P

### Setup

Native install on Debian

1337hium, 2024-01-17:
I have the same problem:

Device: SNZB-06P
FW Version: 1.0.6

Failed to configure '0x842712fffeff64c0', attempt 1 (Error: Read 0x842712fffeff64c0/1 64529([8193], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:62:9) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:72:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:17))

Koenkk, 2024-01-17:
Fixed in https://github.com/Koenkk/zigbee-herdsman-converters/issues/6829

closed 2024-01-17 by Koenkk