Koenkk/zigbee2mqtt #20810: Human presence sensor (ZG-205Z/A) doesn't change presence state

Rincovanderbeek, 2024-01-15:
### What happened?

After the last update, the presence state doesn't change back fro 'home' state.
Only if you change motion distance setting, it goes back to 'not detected'
But once it detects presence, it keeps this state and does not go back.

### What did you expect to happen?

When noting is detected, the sensor should go back to 'no presence'

### How to reproduce it (minimal and precise)

Install de ZG-205Z/A sensor with zigbee2mqtt as usual.
Have it detect human presence and make sure, afterwards, thesre is noting to detect.
It will detect 'presence'.  Wait 5-10 minutes and check again, it still will state 'presence'

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zb Dongle-E

### Setup

Direct as add-on in HA

Rincovanderbeek, 2024-01-15:
Sorry, issue still exists, I tried a new device but it did exactly the same

janmannsbart23, 2024-02-13:
same for me

Ciqsky, 2024-03-05:
Same for me ....2 sensors

Toxo666, 2024-03-07:
same here. 2 sensors. any workarrounds?

PrinceofValium, 2024-09-15:
Hello,

i own 2 of this devices and the issue still exist.
Somebody have an workaround for me?

Rincovanderbeek, 2024-09-15:
Not sure if there were changes in Z2M but so far, the sensors are still not changung state. Only if you change a setting (sensitivity),  it changes but after one detection it again stays in the state it has.

closed 2025-01-07 by github-actions[bot]

J0nnyBr4v0, 2025-01-09:
Did someone found a solution?

PrinceofValium, 2025-01-09:
Me Not. Sometimes its working sometimes not.