Koenkk/zigbee2mqtt #20812: LED slow red/blue blinking SP-EUC01 Z2M

CorentinPDO, 2024-01-15:
### What happened?

Dear community,

I have an Aqara Smart Plug SP-EUC01 and I have noticed that my aqara smart plug EU (rated for 2300w) is blinking blue/red (slow) after a couple of minutes under a 600w load.

It is not a rapid quick red blinking as per overload limit. I have tested it out and overload protection is working if I set the limit to 500w for example.
I can still switch on/off the switch remotely (Z2M via Home assistant)
The documentation states that the Zigbee network is not connected, but I can still operate the switch so I am bit confused..

Should I turn off/on the plug, the led get back to solid blue.

The plug is not warm.

If I use low power devices, such as a 20w lamp, I have no issues.
There is no error being thrown by Z2M and the switch does not switch off automatically...

As the plug is rated for 2300W I do not understand what is going on...

SP-EUC01 firmware version: 01-13-2023
HA version: 2023.11.1
Z2M version: 1.33.2 commit 9996c93
Coordinator type: zStack3x0
Coordinator revision: 20210708

Thank you in advance for your support.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20210708

### Adapter

Zigbee 3.0 - Sonoff USB Dongle

### Setup

Add-on on HA Docker on RPi4 8gb

dankocrnkovic, 2024-02-01:
I have the same issue.
Also device is not reporting Voltage, Current or consumer.
[image]

imammedo, 2024-04-20:
wrt voltage/current/power I've just opened issue: https://github.com/Koenkk/zigbee2mqtt/issues/22269

as for slow red/blue blinking, I also had it (load ~5w), but after updating SONOFF ZB Dongle-P  to 20230507 it seems to be gone for the last 1hr after update. 
with state: on and rest of knobs: false

trozman, 2024-05-04:
Same here - I removed a rogue Tuya smart socket and Aqara EU-SP01 started to work normally (no blue-red blinking).

illixion, 2024-06-28:
I've had this issue as well (CC2531), but it was because my Z2M version was out of date, Home Assistant stopped checking updates for it and I was stuck on a version from 2021. Updating to latest edge version solved the blinking and added many attributes that were previously missing.

GurbEHV, 2024-09-24:
My SP-EUCO1 plugs all started doing this after updating them to the Aqara firmware of 19-7-2024. I did not find a solution yet (Power cycling doesn't appear to help, power outage memory was already set for them). I'll try re-paring one of them and see if that does anything.

b3ko91, 2024-10-01:
same issue for me

infinicode, 2024-10-01:
Same issue here after upgrading SP-EUC01 firmware on 19-7-2024.

Txus6630, 2024-10-03:
Same issue here, firmware 19-7-2024

MikeDeltaHH, 2024-10-11:
Same here with one of my 11 plugs...

Cooluck, 2024-11-03:
Same issue

martcs, 2024-11-18:
i get the same issue, after update blinking red led. I solve it by restoring overload value in plug settings zigbee2mqtt, after firmware update this value changed by itself to low value (700W or sou i dont remember),  restore to 1900W (water boiler) and all work again...

deB4SH, 2024-11-27:
>i get the same issue, after update 07-19-2024 blinking red led. I solve it by restoring overload value in plug settings [Overload protection] zigbee2mqtt, after firmware update this value changed by itself to low value (700W or sou i dont remember), restore to 1900W (water boiler) and all work again...

Sadly it didn't work for me. 
Socket keep blinking after setting overload protection.

b3ko91, 2024-11-27:
> Sadly it didn't work for me. Socket keep blinking after setting overload protection.

Me too
Not resolv for me

rommess, 2024-12-06:
same issue with red/blue light blinking slowly.
load of 10W
everything was reinstall from scratch.

rommess, 2025-01-05:
updated
ha to 2025.1.0
z2m to 2.0.0
zonoff E to 7.4.5
-> no more led blinking :)
voltage and current are still set to Null

closed 2025-03-15 by github-actions[bot]