Koenkk/zigbee2mqtt #20814: PIR Sensor Tuya 809WZT won't connect

juandelarocha, 2024-01-15:
### What happened?

Description of the issue

I have just bought a new Tuya 809WZT and there's no way I can manage to attach it to the zigbee network. The sensor has a working battery, it blinks when pressing the button, and the Zigbee2mqtt is obviously in "permit to join" mode. I have already more than 25 sensors in the network working flawlessly. The Zigbee adaptor is a sonoff dongle zbdongle. 

I have tried in different ways, pressing the button with the pin included in the box, pressing for more than 10 secs (as the instructions said) pressing only 5 secs, pressing right before changing the battery, but there's no trace of the device in the log of zigbe2mqtt. 

I confirm the device is a zigbee 3.0 sensor. 

Addon version
Current version: 1.35.1-1

Core 2024.1.3
Supervisor 2023.12.0
Operating System 11.4
Frontend 20240104.0

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

20220219

### Adapter

Sonoff CC2652P

### Setup

Add-on on HomeAssistant setup in Intel NUC

juandelarocha, 2024-01-17:
Solved. Don't know why, but managed to open the sensor and it worked when out of the plastic case.

closed 2024-01-17 by juandelarocha

Col-Hogan, 2024-01-19:
Hi,
ich hatte heute das gleiche problem.
Ich denke das problem liegt am Minuspin. Die Ausstanzung trückt auf die Platine und deshalb kommt es zu falschem Minus Potenzial.

[image]
[image]

andrefcarvalho, 2024-02-18:
i have the same problem, but with a broken component..
[image]
i have to remove both, but one its broken, because of the metal clip
any one know what component is?
[image]

andrefcarvalho, 2024-02-18:
litle sherch, can be a 100r resistor?
https://developer.tuya.com/en/docs/iot/wifi-tyde30-sch?id=K96gqp3brde85

[image]
[image]

Soukyuu, 2024-06-20:
@Col-Hogan: you are spot on. Bending the punchout pin away from the board on both my devices made them finally pair correctly, needed 4-6 tries before that.