Koenkk/zigbee2mqtt #20815: [New device support]: KOJIMA THS-ZG-LITE temperature and humidity sesnsos

WAlekseev, 2024-01-15:
### Link

https://kojima.ru/catalog/klimat/kojima-ths-zg-lite-umnyy-zigbee-datchik-temperatury-i-vlazhnosti-s-alisoy-i-google-assistant-mini-/#desc

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c1386e02cea548","nwkAddr":37225,"manufId":4098,"manufName":"_TZE200_ysm4dsb1","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,1026,1029],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2620}},"msRelativeHumidity":{"attributes":{"measuredValue":2780}},"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ysm4dsb1"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":131,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1705310612144,"defaultSendRequestWhen":"immediate"}

### Comments

from CLUSTERS tab of this device:
`
0xa4c1386e02cea548
Endpoint 1
Input clusters
genBasic
genPowerCfg
msTemperatureMeasurement
msRelativeHumidity
`

z2m log:
`
2024-01-15 12:33:25MQTT publish: topic 'zigbee2mqtt_1/0xa4c1386e02cea548', payload '{"last_seen":"2024-01-15T09:33:26.604Z","linkquality":172}'
Debug 2024-01-15 12:33:26Received Zigbee message from '0xa4c1386e02cea548', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2550}' from endpoint 1 with groupID 0
Debug 2024-01-15 12:33:26Skipping message, still interviewing
Info 2024-01-15 12:33:26MQTT publish: topic 'zigbee2mqtt_1/0xa4c1386e02cea548', payload '{"last_seen":"2024-01-15T09:33:27.166Z","linkquality":176}'
Debug 2024-01-15 12:33:26Received Zigbee message from '0xa4c1386e02cea548', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":1870}' from endpoint 1 with groupID 0
Debug 2024-01-15 12:33:26Skipping message, still interviewing
Info 2024-01-15 12:33:26MQTT publish: topic 'zigbee2mqtt_1/0xa4c1386e02cea548', payload '{"last_seen":"2024-01-15T09:33:27.414Z","linkquality":172}'
`

### External defintion

```shell
const {batteryPercentage, temperature, humidity} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_ysm4dsb1',
    description: 'Automatically generated definition',
    extend: [batteryPercentage(), temperature(), humidity()],
};

module.exports = definition;
```

usv, 2024-01-26:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ysm4dsb1']),
    model: 'RSH-HS06',
    vendor: 'TuYa',
    description: 'Temperature and humidity sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({queryOnDeviceAnnounce: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await device.getEndpoint(1).command('manuSpecificTuya', 'dataQuery', {});
    },
    exposes: [e.temperature(), e.humidity(), tuya.exposes.batteryState()],
    meta: {
        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10],
            [2, 'humidity', tuya.valueConverter.raw],
            [4, 'battery_state', tuya.valueConverter.batteryState],
        ],
    },
    whiteLabel: [
        tuya.whitelabel('TuYa', 'RSH-HS06', 'Temperature and humidity sensor', ['_TZE200_ysm4dsb1']),
    ],
};

module.exports = definition;
```

usv, 2024-01-26:
[image]

Koenkk, 2024-01-27:
Try: `[3, 'battery_state', tuya.valueConverter.batteryState],`

usv, 2024-01-27:
> Try: `[3, 'battery_state', tuya.valueConverter.batteryState],`

Battery state - None

Koenkk, 2024-01-28:
What if you use:

```js
[3, 'battery_state', tuya.valueConverter.raw],
[4, 'battery', tuya.valueConverter.raw],
```

and wait for 24 hours, is anything reported?

usv, 2024-01-28:
> What if you use:
> 
> ```js
> [3, 'battery_state', tuya.valueConverter.raw],
> [4, 'battery', tuya.valueConverter.raw],
> ```
> 
> and wait for 24 hours, is anything reported?

[image]
[image]

usv, 2024-01-28:
[image]

usv, 2024-01-29:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ysm4dsb1']),
    model: 'RSH-HS06',
    vendor: 'TuYa',
    description: 'Temperature and humidity sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({queryOnDeviceAnnounce: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await device.getEndpoint(1).command('manuSpecificTuya', 'dataQuery', {});
    },
    exposes: [e.battery(), e.temperature(), e.humidity()],
    meta: {
        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10],
            [2, 'humidity', tuya.valueConverter.raw],
            [4, 'battery', tuya.valueConverter.raw],
        ],
    },
    whiteLabel: [
        tuya.whitelabel('TuYa', 'RSH-HS06', 'Temperature and humidity sensor', ['_TZE200_ysm4dsb1']),
    ],
};

module.exports = definition;
```
[image]

Koenkk, 2024-01-29:
Great so battery works, could you submit a PR?

usv, 2024-01-30:
> Great so battery works, could you submit a PR?

What is PR?

Koenkk, 2024-01-30:
Added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-30 by Koenkk