Koenkk/zigbee2mqtt #20819: Cant change MQTT server both on WebUI and Configuration.yaml

Jezza34000, 2024-01-15:
### What happened?

I can't change MQTT SERVER both on WebUI on in Configuration.yaml
I got an error on WebUI, and dosen't take effect when i change it into yaml file.

[image]

### What did you expect to happen?

Changing my MQTT server 
mqtt > server (in configuration.yaml)
MQTT server* (in WebUI)

### How to reproduce it (minimal and precise)

Change MQTT SERVER into WebUI or into Configuration.yaml

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

itead SonOff Zigbee 3.0 Dongle PLUS

### Setup

HAOS

### Debug log

When i try ot change in WebUI i got thoses messages :
`Cannot read properties of null (reading 'length')`
`Request 'zigbee2mqtt/bridge/request/options' failed with error: 'Cannot read properties of null (reading 'length')'`

When i try to change over configuration.yaml it still use old configuration...

[image]

```
Zigbee2MQTT:info  2024-01-15 14:30:51: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2024-01-15 14:30:51: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:info  2024-01-15 14:30:51: Connected to MQTT server
Zigbee2MQTT:info  2024-01-15 14:30:51: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
```

jwvanderbeck, 2024-04-22:
Same problem :(
I want to switch it to use my own internal broker I have running on my Synology that I send other messages to from my own devices/code but I get this error.

mihsu81, 2024-05-09:
An alternative is to modify the Add-On configuration. But I do agree that a real fix is needed.

```yaml
base_topic: zigbee2mqtt
server: mqtt://192.168.45.201:1883
user: xxxxxxx
password: xxxxxxxxxxxx
include_device_information: true
```

weljajoh, 2024-05-25:
If you use the Home Assistant zigbee2mqqt add-on, you can change it via the add-on's 'Configuration' tab if you switch to 'Edit in YAML' (editing /config/zigbee2mqtt/configuration.yaml directly doesn't work in the HA case)

jwvanderbeck-Unity, 2024-07-13:
> An alternative is to modify the Add-On configuration. But I do agree that a real fix is needed.
> 
> ```yaml
> base_topic: zigbee2mqtt
> server: mqtt://192.168.45.201:1883
> user: xxxxxxx
> password: xxxxxxxxxxxx
> include_device_information: true
> ```

I feel like I tried this before coming here, but I'll try it again just to be sure.  Been a bit now since I last looked at this.

closed 2025-01-07 by github-actions[bot]