Koenkk/zigbee2mqtt #20822: [Wrong device]:

Flavien06, 2024-01-15:
### Link

/

### Model

TYZG-001-RF

### Description

Dry Contact Smart Switch

### Vendor

TuYa

### Picture (link)

https://ae01.alicdn.com/kf/S710e3df3b7594213ba6d680f67eb7b6ba.jpg?width=1000&height=1000&hash=2000

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0xa4c138bb9befdb97","nwkAddr":4958,"manufId":4417,"manufName":"_TZ3000_g8n1n7lg","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3000_g8n1n7lg","powerSource":1,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c16cdb","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":153093895},"lastSeen":1705339501202,"defaultSendRequestWhen":"immediate"}

Jordan87, 2024-03-06:
What do you mean by wrong device?

Flavien06, 2024-03-06:
I see this [image]
[image]
[image]

Jordan87, 2024-03-06:
I think because they used same internal parts and ID but maybe im wrong ;)

Jordan87, 2024-03-07:
Here you can also find different device which shows same model TS0001 but have different case.
https://github.com/zigpy/zha-device-handlers/issues/1118

closed 2024-10-04 by github-actions[bot]