Koenkk/zigbee2mqtt #20827: New Edge not starting anymore with ezsp 7.4.0

wastez, 2024-01-15:
### What happened?

It doesn’t start anymore. 
no error message, just writing „starting“ in the logs. 
the gui doesn’t get available. 

### Zigbee2MQTT version

latest edge

### Adapter firmware version

7.4.0

### Adapter

EZSP Sonoff Dongle E

### Setup

Proxmox

Koenkk, 2024-01-15:
- What zigbee herdsman version shows up under frontend -> settings -> about
- Can you provide the herdsman debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

CC: @Nerivec

wastez, 2024-01-15:
The version from 13.1 is working without problems.
The version from today isn't working anymore.
So it seem to be related to the last changes in zigbee-herdsmann.

Nerivec, 2024-01-15:
I unfortunately don't have test equipment at the moment...
@wastez, following the fix, assuming everything works, I'd appreciate it if you could report your observations in a couple of days (here will be fine, even if it's closed by then, unless you encounter a new issue of course). 😉

wastez, 2024-01-15:
@Nerivec 
Of course i will tell you how it will work.

@Koenkk 
Would it possible to migrate it into the dev channel?
Would be really great.

wastez, 2024-01-15:
@Nerivec 
Do you have a idea if I could migrate this manual to the container?

mamrai1, 2024-01-16:
If you haven't deleted the old latest-dev image:
1)Stop z2m container and delete the latest-dev image
2) docker tag sha256:0d024c7064d81f74864b6327d35144716c33f936e0b74cc23660db224e088d31 koenkk/zigbee2mqtt:latest-dev
3) start the container

Koenkk, 2024-01-16:
Should be fixed now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

wastez, 2024-01-16:
@mamrai1 
Yeah that was the problem deleted it and made no backup of it in Hass. 
So I thought to implement the changes directly. But this was not so easy.  
So i had to restore my backup from the last day on my proxmox server. 

@Koenkk 
Thank you will test it.

wastez, 2024-01-16:
Ok updated it and now it seem to work. 
So far so good, will keep you up to date.

wastez, 2024-01-19:
@Nerivec 
Just to inform you, so it seems to work but there are some waitress problems. 

For example:
```
Zigbee2MQTT:error 2024-01-16 22:31:25: Failed to configure 'Vorhaus - Bewegungsmelder', attempt 2 (Error: Read 0x54ef441000777cc6/1 aqaraOpple([338], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 33566 - 1 - 173 - 64704 - 1 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:471:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:478:28)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:1766:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Koenkk, 2024-01-20:
@wastez thats not because of the EZSP adapter, seems the device just doesn't respond.

wastez, 2024-01-20:
Yes but since this changes I have it with every device.

Koenkk, 2024-01-21:
@wastez looks like an issue in the EZSP firmware, did you already try downgrading it?

wastez, 2024-01-21:
Yes could be and of course I could downgrade. 
But it wasn't before this changes I think, even with this firmware. Are you sure the changes could not cause this problem?

wastez, 2024-01-21:
Here much more about it:

```

[18:03:27] INFO: Preparing to start...
[18:03:27] INFO: Socat not enabled
[18:03:28] INFO: Starting Zigbee2MQTT...
Assertion failed: Command (setConfigurationValue(1, 255)) returned unexpected state: {"_cls_":"setConfigurationValue","_id_":83,"_isRequest_":false,"status":55}
Assertion failed: exportKey returned unexpected status: 45
Zigbee2MQTT:warn  2024-01-20 18:03:43: Failed to ping 'Bad - Deckenbeleuchtung' (attempt 1/1, Read 0x84b4dbfffe2171c7/1 genBasic(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 15786 - 1 - 8 - 0 - 1 after 10000ms))
Zigbee2MQTT:warn  2024-01-20 18:04:55: OTA: Images currently unavailable for device 'TS0601', hardwareVersion '1', manufacturerName _TZE200_hue3yfsn, {"fieldControl":0,"manufacturerCode":4098,"imageType":5634,"fileVersion":69}'
Zigbee2MQTT:error 2024-01-20 18:12:56: Failed to configure 'Vorhaus - Bewegungsmelder', attempt 1 (Error: Read 0x54ef441000777cc6/1 genPowerCfg(["batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 33566 - 1 - 42 - 1 - 1 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:478:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:478:28)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:1764:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:error 2024-01-20 18:12:59: Failed to configure 'Hobbyzimmer - Bewegungsmelder', attempt 1 (Error: Read 0x54ef44100053d18c/1 genPowerCfg(["batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 50564 - 1 - 43 - 1 - 1 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:478:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:478:28)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:1764:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:error 2024-01-20 18:13:38: Failed to configure 'Hobbyzimmer - Bewegungsmelder', attempt 2 (Error: Read 0x54ef44100053d18c/1 genPowerCfg(["batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 50564 - 1 - 44 - 1 - 1 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:478:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:478:28)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:1764:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:error 2024-01-20 18:13:47: Failed to configure 'Vorhaus - Bewegungsmelder', attempt 2 (Error: Read 0x54ef441000777cc6/1 genPowerCfg(["batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 33566 - 1 - 49 - 1 - 1 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:478:47)
[... cut ...] Undiv":false}) failed (Timeout - 37115 - 1 - 96 - 0 - 1 after 10000ms))
Zigbee2MQTT:warn  2024-01-21 05:23:33: Failed to ping 'Bad - Lueftersteuerung' (attempt 2/2, Read 0x00124b0024c336a3/1 genBasic(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37115 - 1 - 97 - 0 - 1 after 10000ms))

```

wastez, 2024-01-21:
But looks it only happens after a fresh start of z2m

Nerivec, 2024-01-21:
Can you capture one with herdsman debug ON?

wastez, 2024-01-22:
Ok here it is:
```

Zigbee2MQTT:error 2024-01-22 00:54:43: Failed to configure 'Hobbyzimmer - Bewegung
smelder', attempt 1 (Error: Read 0x54ef44100053d18c/1 aqaraOpple([338], {"timeout"
:10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":tr
ue,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"tran
sactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 50564 - 1 - 34 
- 64704 - 1 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23
)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsm
an/src/adapter/ezsp/adapter/ezspAdapter.ts:478:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/requ
est.ts:79:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpo
int.ts:478:28)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.
ts:1767:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts
:140:25)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:debug 2024-01-22 00:54:46: Saving state to file /config/zigbee2mqtt/st
ate.json
Zigbee2MQTT:debug 2024-01-22 00:54:50: Received Zigbee message from 'Kueche - Lamp
enschirm', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from e
ndpoint 1 with groupID 0
Zigbee2MQTT:info  2024-01-22 00:54:50: MQTT publish: topic 'zigbee2mqtt/Kueche - L
ampenschirm', payload '{"brightness":254,"color_mode":"color_temp","color_options"
:
```

Nerivec, 2024-01-22:
Can you include the stuff before `Zigbee2MQTT:error` ? That's the part I want.

wastez, 2024-01-22:
Do you search for that:

```

Zigbee2MQTT:info  2024-01-22 00:54:20: Configuring 'Hobbyzimmer - Bewegungsmelder'
Zigbee2MQTT:debug 2024-01-22 00:54:20: Received Zigbee message from 'Hobbyzimmer - Bewegungsmelder', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,196,11,3,40,24,4,33,0,0,5,33,2,0,8,33,10,1,10,33,57,97,12,32,1,19,32,0,20,32,0,100,16,0,101,33,21,0,105,32,30,106,32,3,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-22 00:54:20: RTCGQ14LM: Processed buffer into data {"1":3012,"3":24,"4":0,"5":2,"8":266,"10":24889,"12":1,"19":0,"20":0,"100":0,"101":21,"105":30,"106":3,"107":0}
Zigbee2MQTT:debug 2024-01-22 00:54:20: RTCGQ14LM: unknown key 19 with value 0
Zigbee2MQTT:debug 2024-01-22 00:54:20: RTCGQ14LM: unknown key 20 with value 0
Zigbee2MQTT:debug 2024-01-22 00:54:20: RTCGQ14LM: Processed data into payload {"voltage":3012,"battery":100,"device_temperature":24,"power_outage_count":1,"illuminance":21,"detection_interval":30,"motion_sensitivity":"high","trigger_indicator":false}
Zigbee2MQTT:debug 2024-01-22 00:54:20: RTCGQ14LM: Processed data into payload {"voltage":3012,"battery":100,"device_temperature":24,"power_outage_count":1,"illuminance":21,"detection_interval":30,"motion_sensitivity":"high","trigger_indicator":false}
Zigbee2MQTT:info  2024-01-22 00:54:20: MQTT publish: topic 'zigbee2mqtt/Hobbyzimmer - Bewegungsmelder', payload '{"battery":100,"detection_interval":30,"device_temperature":24,"illuminance":21,"last_seen":"2024-01-22T00:54:20+01:00","linkquality":224,"motion_sensitivity":"high","occupancy":false,"power_outage_count":1,"trigger_indicator":false,"update":{"installed_version":10,"latest_version":10,"state":"idle"},"update_available":null,"voltage":3012}'
Zigbee2MQTT:debug 2024-01-22 00:54:21: Received Zigbee message from 'Hobbyzimmer - Bewegungsmelder', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":29}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-01-22 00:54:21: MQTT publish: topic 'zigbee2mqtt/Hobbyzimmer - Bewegungsmelder', payload '{"battery":33,"detection_interval":30,"device_temperature":24,"illuminance":21,"last_seen":"2024-01-22T00:54:21+01:00","linkquality":224,"motion_sensitivity":"high","occupancy":false,"power_outage_count":1,"trigger_indicator":false,"update":{"installed_version":10,"latest_version":10,"state":"idle"},"update_available":null,"voltage":2900}'
Zigbee2MQTT:debug 2024-01-22 00:54:22: Received Zigbee message from 'Hobbyzimmer - Bewegungsmelder', type 'readResponse', cluster 'aqaraOpple', data '{"258":30}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-22 00:54:22: RTCGQ14LM: Processed data into payload {"detection_interval":30}

```

Nerivec, 2024-01-22:
I'm looking for the lines like you had in your [first log](https://github.com/Koenkk/zigbee2mqtt/issues/20827#issuecomment-1892718947), `zigbee-herdsman:adapter:ezsp:ezsp`, `zigbee-herdsman:adapter:ezsp:uart` that kind. Seems you don't have herdsman debug ON here.

wastez, 2024-01-22:
Sorry less cuted it:

```

2024-01-21T23:54:03.733Z zigbee-herdsman:adapter:ezsp:driv handleRouteRecord: nwk=28854, ieee=0�����, lqi=208, rssi=-48, relays=�
2024-01-21T23:54:03.780Z zigbee-herdsman:adapter:ezsp:uart <-- [719ab1a97d312a15b259914a25aa5592499c16f77b5bbe9874fe896389f8ee547e]
2024-01-21T23:54:03.780Z zigbee-herdsman:adapter:ezsp:uart --> parsed 719ab1a9112a15b259914a25aa5592499c16f77b5bbe9874fe896389f8ee547e
2024-01-21T23:54:03.780Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,1,0): 719ab1a9112a15b259914a25aa5592499c16f77b5bbe9874fe896389f8ee547e
2024-01-21T23:54:03.780Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2024-01-21T23:54:03.780Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2024-01-21T23:54:03.780Z zigbee-herdsman:adapter:ezsp:uart <-- ACK (1): 719ab1a9112a15b259914a25aa5592499c16f77b5bbe9874fe896389f8ee547e
2024-01-21T23:54:03.780Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: d890014500000000050000000000000058d0d0b670ffff034f000004
2024-01-21T23:54:03.780Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":0,"sequence":88,"clusterId":5,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":0},"lastHopLqi":208,"lastHopRssi":-48,"sender":28854,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[79,0,0]}}
2024-01-21T23:54:03.780Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":0,"sequence":88,"clusterId":5,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":0},"lqi":208,"rssi":-48,"sender":28854,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[79,0,0]}}
2024-01-21T23:54:08.294Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2024-01-21T23:54:08.294Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2024-01-21T23:54:08.294Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2024-01-21T23:54:08.294Z zigbee-herdsman:adapter:ezsp:uart --> DATA (1,0,0): d900010500
2024-01-21T23:54:08.294Z zigbee-herdsman:adapter:ezsp:uart --> [109b21a9512a04b07e]
2024-01-21T23:54:08.295Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (2)
2024-01-21T23:54:08.298Z zigbee-herdsman:adapter:ezsp:uart <-- [029ba1a9512a484c7e]
2024-01-21T23:54:08.298Z zigbee-herdsman:adapter:ezsp:uart --> parsed 029ba1a9512a484c7e
2024-01-21T23:54:08.298Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,2,0): 029ba1a9512a484c7e
2024-01-21T23:54:08.298Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2024-01-21T23:54:08.298Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2024-01-21T23:54:08.298Z zigbee-herdsman:adapter:ezsp:uart <-- ACK (2): 029ba1a9512a484c7e
2024-01-21T23:54:08.298Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: d980010500
2024-01-21T23:54:08.299Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_cls_":"nop","_id_":5,"_isRequest_":false}
2024-01-21T23:54:08.299Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (2) success
2024-01-21T23:54:18.302Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2024-01-21T23:54:18.302Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2024-01-21T23:54:18.303Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2024-01-21T23:54:18.303Z zigbee-herdsman:adapter:ezsp:uart --> DATA (2,1,0): da00010500
2024-01-21T23:54:18.303Z zigbee-herdsman:adapter:ezsp:uart --> [219821a9512a804e7e]
2024-01-21T23:54:18.303Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (3)
2024-01-21T23:54:18.307Z zigbee-herdsman:adapter:ezsp:uart <-- [7d3398a1a9512af9ba7e]
2024-01-21T23:54:18.307Z zigbee-herdsman:adapter:ezsp:uart --> parsed 1398a1a9512af9ba7e
2024-01-21T23:54:18.307Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,3,0): 1398a1a9512af9ba7e
2024-01-21T23:54:18.307Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2024-01-21T23:54:18.307Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2024-01-21T23:54:18.307Z zigbee-herdsman:adapter:ezsp:uart <-- ACK (3): 1398a1a9512af9ba7e
2024-01-21T23:54:18.307Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: da80010500
2024-01-21T23:54:18.307Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_cls_":"nop","_id_":5,"_isRequest_":false}
2024-01-21T23:54:18.308Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (3) success
2024-01-21T23:54:20.312Z zigbee-herdsman:adapter:ezsp:uart <-- [2398b1a90d2a9177d5451925ba7d317d5d1d7c9a26928ce87d5d7e]
2024-01-21T23:54:20.312Z zigbee-herdsman:adapter:ezsp:uart --> parsed 2398b1a90d2a9177d5451925ba117d1d7c9a26928ce87d7e
2024-01-21T23:54:20.312Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,3,0): 2398b1a90d2a9177d5451925ba117d1d7c9a26928ce87d7e
2024-01-21T23:54:20.312Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (3)
2024-01-21T23:54:20.312Z zigbee-herdsman:adapter:ezsp:uart --> [83401b7e]
2024-01-21T23:54:20.312Z zigbee-herdsman:adapter:ezsp:uart <-- ACK (3): 2398b1a90d2a9177d5451925ba117d1d7c9a26928ce87d7e
2024-01-21T23:54:20.313Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: da9001590084c58cd153001044ef54e0d4013961
2024-01-21T23:54:20.313Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x59: {"_cls_":"incomingRouteRecordHandler","_id_":89,"_isRequest_":false,"source":50564,"longId":{"type":"Buffer","data":[84,239,68,16,0,83,209,140]},"lastHopLqi":224,"lastHopRssi":-44,"relay":{"type":"Buffer","data":[57]}}
2024-01-21T23:54:20.313Z zigbee-herdsman:adapter:ezsp:driv handleRouteRecord: nwk=50564, ieee=T�DSь, lqi=224, rssi=-44, relays=9
[... cut ...]
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:478:28)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:1767:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

```

Nerivec, 2024-01-22:
@kirovilya Could this be a side-effect of the removal of `setExtendedTimeout`? https://github.com/Koenkk/zigbee-herdsman/pull/871

```log
<== 0x80: {"_cls_":"incomingRouteErrorHandler","_id_":128,"_isRequest_":false,"status":66,"target":50564}
handleRouteError: nwk=50564, status=66
```

https://docs.silabs.com/d/zigbee-stack-api/7.4.0/message#ember-incoming-route-error-handler
> A status of [EMBER_MAC_INDIRECT_TIMEOUT](https://docs.silabs.com/d/zigbee-stack-api/7.4.0/status-codes#ember-mac-indirect-timeout) indicates that a message sent to the target end device could not be delivered by the parent because the indirect transaction timer expired. Upon receipt of the route error, the stack sets the extended timeout for the target node in the address table, if present. It then calls this handler to indicate receipt of the error.

If I understand this correctly, the default timeout starts at 3sec, and can increase up to 30s, so after having failed a few times, the timeout has increased internally on its own, and it stops failing, which would explain the reported behavior (on start)...

kirovilya, 2024-01-22:
@Nerivec I'm not sure if this is the reason, but anything is possible.
I removed the extended timeout because I saw how in bellows send it only to end devices; for ordinary routers such a command is not executed.

if these commands are sent to a sleeping sensor, then of course they will not reach - you need to wake up the sensor

wastez, 2024-01-22:
@kirovilya 
This make sense it happens on end devices.....

kirovilya, 2024-01-22:
@wastez they need to be “awakened” by briefly pressing the pairing button

wastez, 2024-01-22:
@kirovilya 
Yes of course I do this when I set something. 
But this happens while not changing anything.

wastez, 2024-01-22:
@kirovilya 
If you tell me this is no problem then it's ok to me. 
Just wanted to report problems which I don't saw before.

kirovilya, 2024-01-22:
@wastez "Failed to configure" - means that the device has not yet been configured (perhaps it was not configured when the device was added) - which means it needs to be woken up for configuration.
Configuration is done at the start of z2m for all unconfigured devices.

wastez, 2024-01-22:
@kirovilya 
This is strange, I'm using the devices for a long time and they are configured and they are working.

kirovilya, 2024-01-22:
@wastez One of the possible reasons is that previously this device did not have a “configure” code, but now it does, and therefore it needs to be executed.

wastez, 2024-01-22:
@kirovilya 
It's ok for me. 
And this make sense that this happen to all my aqara motion sensors? (At least)

kirovilya, 2024-01-22:
@wastez I don’t know why this happened... try waking up one of the sensors and performing configuration (you can force the button at the bottom of the sensor page) - the error for this sensor should disappear.

wastez, 2024-01-22:
@kirovilya 
Ok will test it and will report.

Nerivec, 2024-01-22:
@wastez Just a side note from what I read over time since you mentioned them:
Aqara has always been plagued with issues regarding their Zigbee protocol, it was supposed to get better with the T1 versions, but considering the number of reports I have seen, even recently, I still suspect they didn't follow the Zigbee standard properly.
This manifests with devices being extremely annoying in how they pick routers. Some will drop from the network rather than connect to a router "they don't like". I guess if you are lucky enough to have routers they like, then you wouldn't see any issue... There were quite a few threads about these (several on the Hubitat forums if memory serves) if you want more details.

wastez, 2024-01-22:
Hmmmmmm....
Ok for now I can see that some thing where not refreshed some time (battery)
Looks like reconfiguration makes a difference. 
Will keep you up to date.

RichardPar, 2024-01-23:
I have the same issue... does not start on boot... using a RESTART and it then works..  I am using EDGE and the latest ezsp multiprotocol. 

[code]Assertion failed: Command (setConfigurationValue(1, 255)) returned unexpected state: {"_cls_":"setConfigurationValue","_id_":83,"_isRequest_":false,"status":55}
[/code]

Nerivec, 2024-01-23:
@RichardPar do you have any more logs (try starting with [herdsman debug](https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging) ON)? This config ID assertion is not an issue per say, it's just not supported anymore, the code needs cleanup; anyway, it does not prevent startup.

RichardPar, 2024-01-23:
> @RichardPar do you have any more logs (try starting with [herdsman debug](https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging) ON)? This config ID assertion is not an [...]

It hangs after a boot at this point.. it however starts when I do a RESTART

''[16:26:15] INFO: Preparing to start...
[16:26:16] INFO: Socat not enabled
[16:26:20] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-01-23 16:26:26: Logging to console and directory: '/config/zigbee2mqtt/log/2024-01-23.16-26-26' filename: log.txt
Zigbee2MQTT:info  2024-01-23 16:26:26: Starting Zigbee2MQTT version 1.35.1-dev (commit #de7bc7c)
Zigbee2MQTT:info  2024-01-23 16:26:26: Starting zigbee-herdsman (0.32.7)
Assertion failed: Command (setConfigurationValue(1, 255)) returned unexpected state: {"_cls_":"setConfigurationValue","_id_":83,"_isRequest_":false,"status":55}
Assertion failed: Command (setValue) returned unexpected state: {"_cls_":"setValue","_id_":171,"_isRequest_":false,"status":55}
Assertion failed: exportKey returned unexpected status: 45
``

kirovilya, 2024-01-23:
"Assertion failed" - not a critical situation for ezsp adapter. I can hide them or put them in debug logs.

Nerivec, 2024-01-23:
It's good to have them as `console.assert` I think. It will help figure out values supported for each adapter, in case it becomes needed to init based on detected adapter in the future.

RichardPar, 2024-01-23:
It hangs on Boot :D There has to be a race condition with something else.. but to clarify - it does not start at all on first boot.. 

[image]
****

As to how to enable debug, where would herdsman be set? I added under the Socat stanza - didnt seem to do anything

[image]

Nerivec, 2024-01-23:
Check the `Show unused optional configuration options`, and it will pop up as a toggle.

wastez, 2024-01-23:
You have to change to yaml edit.

RichardPar, 2024-01-23:
Thanks... got it. Didnt realise the top changed :D was looking the other way...

From the end bits... looks like the Z2MQTT is started before SILabs Multiprotocol and times-out before it works

''
2024-01-23T21:06:27.259Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (2) success
2024-01-23T21:06:27.769Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 3
2024-01-23T21:06:27.770Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with core-silabs-multiprotocol:9999
2024-01-23T21:06:27.773Z zigbee-herdsman:adapter:ezsp:uart Socket error
2024-01-23T21:06:27.774Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 3 error: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:150:24)
    at Socket.emit (node:events:529:35)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)
2024-01-23T21:06:27.774Z zigbee-herdsman:adapter:ezsp:uart Port closed
2024-01-23T21:06:27.774Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2024-01-23T21:06:27.774Z zigbee-herdsman:adapter:ezsp:driv Close driver
2024-01-23T21:06:31.361Z zigbee-herdsman:adapter:ezsp:driv Reset connection. Try 1
2024-01-23T21:06:31.361Z zigbee-herdsman:adapter:ezsp:driv Stop driver
2024-01-23T21:06:31.362Z zigbee-herdsman:adapter:ezsp:ezsp Stop ezsp
2024-01-23T21:06:31.362Z zigbee-herdsman:adapter:ezsp:uart closing
2024-01-23T21:06:31.362Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2024-01-23T21:06:31.362Z zigbee-herdsman:adapter:ezsp:driv Close driver
''

RichardPar, 2024-01-23:
Its looping

```
2024-01-23T21:20:33.598Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2024-01-23T21:20:33.599Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2024-01-23T21:20:33.599Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2024-01-23T21:20:33.599Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,6,0): 160005
2024-01-23T21:20:33.599Z zigbee-herdsman:adapter:ezsp:uart --> [665421ad29157e]
2024-01-23T21:20:33.600Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2024-01-23T21:20:33.601Z zigbee-herdsman:adapter:ezsp:uart <-- [6754a1ad]
2024-01-23T21:20:33.602Z zigbee-herdsman:adapter:ezsp:uart <-- [0c2a5c79557e]
2024-01-23T21:20:33.602Z zigbee-herdsman:adapter:ezsp:uart --> parsed 6754a1ad0c2a5c79557e
2024-01-23T21:20:33.602Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,7,0): 6754a1ad0c2a5c79557e
2024-01-23T21:20:33.602Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2024-01-23T21:20:33.603Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2024-01-23T21:20:33.603Z zigbee-herdsman:adapter:ezsp:uart <-- ACK (7): 6754a1ad0c2a5c79557e
2024-01-23T21:20:33.603Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 168005580049
2024-01-23T21:20:33.604Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_cls_":"nop","_id_":5,"_isRequest_":false}
2024-01-23T21:20:33.604Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2024-01-23T21:20:37.486Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 3
2024-01-23T21:20:37.486Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with core-silabs-multiprotocol:9999
2024-01-23T21:20:37.490Z zigbee-herdsman:adapter:ezsp:uart Socket error
2024-01-23T21:20:37.491Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 3 error: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:150:24)
    at Socket.emit (node:events:529:35)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)
2024-01-23T21:20:37.492Z zigbee-herdsman:adapter:ezsp:uart Port closed
2024-01-23T21:20:37.492Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2024-01-23T21:20:37.492Z zigbee-herdsman:adapter:ezsp:driv Close driver
2024-01-23T21:20:42.495Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 4
2024-01-23T21:20:42.496Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with core-silabs-multiprotocol:9999
2024-01-23T21:20:42.501Z zigbee-herdsman:adapter:ezsp:uart Socket error
2024-01-23T21:20:42.507Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 4 error: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:150:24)
    at Socket.emit (node:events:529:35)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)
2024-01-23T21:20:42.507Z zigbee-herdsman:adapter:ezsp:erro EZSP could not connect: Error: Error while opening socket
2024-01-23T21:20:42.509Z zigbee-herdsman:adapter:ezsp:erro Reset error Error: Failure to connect
2024-01-23T21:20:42.501Z zigbee-herdsman:adapter:ezsp:uart Socket error
2024-01-23T21:20:42.507Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 4 error: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:150:24)
    at Socket.emit (node:events:529:35)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)
2024-01-23T21:20:42.507Z zigbee-herdsman:adapter:ezsp:erro EZSP could not connect: Error: Error while opening socket
2024-01-23T21:20:42.509Z zigbee-herdsman:adapter:ezsp:erro Reset error Error: Failure to connect
    at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:293:19)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:143:13)
    at Driver.reset (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:116:24)
    at Driver.onReset (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:132:9)
2024-01-23T21:20:42.509Z zigbee-herdsman:adapter:ezsp:driv Pause 60sec before try 5
2024-01-23T21:20:42.509Z zigbee-herdsman:adapter:ezsp:uart Port closed
2024-01-23T21:20:42.509Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2024-01-23T21:20:42.510Z zigbee-herdsman:adapter:ezsp:driv Close driver
2024-01-23T21:20:43.599Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2024-01-23T21:20:43.599Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2024-01-23T21:20:43.600Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2024-01-23T21:20:43.600Z zigbee-herdsman:adapter:ezsp:uart --> DATA (7,7,0): 170005
2024-01-23T21:20:43.600Z zigbee-herdsman:adapter:ezsp:uart --> [775521ad73367e]
2024-01-23T21:20:43.601Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (0)
2024-01-23T21:20:43.602Z zigbee-herdsman:adapter:ezsp:uart <-- [7055]
2024-01-23T21:20:43.602Z zigbee-herdsman:adapter:ezsp:uart <-- [a1ad0c2a5c12ca7e]
2024-01-23T21:20:43.603Z zigbee-herdsman:adapter:ezsp:uart --> parsed 7055a1ad0c2a5c12ca7e
[... cut ...]
2024-01-23T21:20:53.607Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2024-01-23T21:20:53.607Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2024-01-23T21:20:53.608Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2024-01-23T21:20:53.608Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 180005
2024-01-23T21:20:53.608Z zigbee-herdsman:adapter:ezsp:uart --> [005a21ad4c5f7e]
2024-01-23T21:20:53.608Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
```

wastez, 2024-01-23:
Looks like it can't connect to the port auf the multiprotocol addon. 

Can you access it via telnet? To see if the port is reachable.

RichardPar, 2024-01-23:
> Looks like it can't connect to the port of the multiprotocol addon.
> 
> Can you access it via telnet? To see if the port is reachable.

If I click restart on the Z2MQTT  plugin, it immidiately connects.   Its there and running (as my THREAD devices are online)

wastez, 2024-01-23:
Maybee z2m try's connect to early while the multiprotocol addon isn't already up and ready for connections.

Nerivec, 2024-01-23:
Can you check the logs of the silabs addon? You might find some more information there...
Something seems to make the socket drop in some weird fashion.

RichardPar, 2024-01-23:
> Can you check the logs of the silabs addon? You might find some more information there... Something seems to make the socket drop in some weird fashion.

I'll dig for some more debug.. Silabs is listening 

Z2MQTT looks like its connecting to something..  (below is the same small logs from Silabs Multiprotocol) 
```
[21:16:06] INFO: Socat enabled
[21:16:07] INFO: Starting socat
Socat startup parameters:
Options:     -d -d
Master:      pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
Slave:       tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
[21:16:07] INFO: Starting socat process ...
2024/01/23 21:16:07 socat[47] N PTY is /dev/pts/0
2024/01/23 21:16:07 socat[47] N listening on AF=2 0.0.0.0:8485
[21:16:08] INFO: Zigbee Herdsman debug logging enabled
[21:16:11] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2024-01-23 21:16:23: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2024-01-23 21:16:23: Logging to console and directory: '/config/zigbee2mqtt/log/2024-01-23.21-16-22' filename: log.txt
Zigbee2MQTT:debug 2024-01-23 21:16:23: Removing old log directory '/config/zigbee2mqtt/log/2024-01-23.17-45-09'
Zigbee2MQTT:info  2024-01-23 21:16:23: Starting Zigbee2MQTT version 1.35.1-dev (commit #de7bc7c)
Zigbee2MQTT:info  2024-01-23 21:16:23: Starting zigbee-herdsman (0.32.7)
Zigbee2MQTT:debug 2024-01-23 21:16:23: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[79,21,20,244,142,185,35,167],"networkKey":"HIDDEN","panID":57138},"serialPort":{"adapter":"ezsp","path":"tcp://core-silabs-multiprotocol:9999","rtscts":false}}'
2024-01-23T21:16:23.529Z zigbee-herdsman:adapter:ezsp:debg Adapter concurrent: 8
2024-01-23T21:16:23.531Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[189,75,78,133,143,84,44,153,64,153,244,26,19,43,146,35],"panID":57138,"extendedPanID":[79,21,20,244,142,185,35,167],"channelList":[11]},"serialPort":{"rtscts":false,"path":"tcp://core-silabs-multiprotocol:9999","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2024-01-23T21:16:23.534Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with core-silabs-multiprotocol:9999
2024-01-23T21:16:23.554Z zigbee-herdsman:adapter:ezsp:uart Socket connected
2024-01-23T21:16:23.555Z zigbee-herdsman:adapter:ezsp:uart Socket ready
2024-01-23T21:16:23.556Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2024-01-23T21:16:23.558Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2024-01-23T21:16:23.560Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2024-01-23T21:16:23.561Z zigbee-herdsman:adapter:ezsp:uart -?- waiting reset
2024-01-23T21:16:23.566Z zigbee-herdsman:adapter:ezsp:uart <-- [c1020b0a527e]
2024-01-23T21:16:23.568Z zigbee-herdsman:adapter:ezsp:uart --> parsed c1020b0a527e
2024-01-23T21:16:23.569Z zigbee-herdsman:adapter:ezsp:uart <-- RSTACK c1020b0a527e
2024-01-23T21:16:23.570Z zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e
2024-01-23T21:16:23.572Z zigbee-herdsman:adapter:ezsp:uart -+- waiting reset success
2024-01-23T21:16:23.573Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":4}
2024-01-23T21:16:23.574Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":4}
2024-01-23T21:16:23.577Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 00000004
2024-01-23T21:16:23.578Z zigbee-herdsman:adapter:ezsp:uart --> [004221a850ed2c7e]
2024-01-23T21:16:23.579Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
2024-01-23T21:16:25.582Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":1} after 2000ms
2024-01-23T21:16:25.582Z zigbee-herdsman:adapter:ezsp:uart -!- break waiting (1)
2024-01-23T21:16:25.582Z zigbee-herdsman:adapter:ezsp:uart Can't send DATA frame (0,0,0): 00000004
2024-01-23T21:16:26.084Z zigbee-herdsman:adapter:ezsp:uart ->> DATA (0,0,1): 00000004
2024-01-23T21:16:26.084Z zigbee-herdsman:adapter:ezsp:uart --> [084221a850ef017e]
2024-01-23T21:16:26.085Z zigbee-herdsman:adapter:ezsp:uart -?- rewaiting (1)
2024-01-23T21:16:27.649Z zigbee-herdsman:adapter:ezsp:uart <-- [c102]
2024-01-23T21:16:27.650Z zigbee-herdsman:adapter:ezsp:uart <-- [0b0a527e0142a1a8592815c6ab047e]
2024-01-23T21:16:27.650Z zigbee-herdsman:adapter:ezsp:uart --> parsed c1020b0a527e
2024-01-23T21:16:27.652Z zigbee-herdsman:adapter:ezsp:uart <-- RSTACK c1020b0a527e
2024-01-23T21:16:27.652Z zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e
2024-01-23T21:16:27.652Z zigbee-herdsman:adapter:ezsp:uart --> parsed 0142a1a8592815c6ab047e
2024-01-23T21:16:27.653Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,1,0): 0142a1a8592815c6ab047e
2024-01-23T21:16:27.653Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2024-01-23T21:16:27.654Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2024-01-23T21:16:27.655Z zigbee-herdsman:adapter:ezsp:uart <-- ACK (1): 0142a1a8592815c6ab047e
[... cut ...]
Listening on port 9999 for connection...

Accepting connection.

Accepted connection 10.

Restarting
[2024-01-23T21:25:40.057648Z] Info : Endpoint socket #12: Client disconnected. 0 connections
[2024-01-23T21:25:40.057851Z] Info : Client disconnected
[2024-01-23T21:25:41.061237Z] Info : New client connection using library v4.4.0.0
[2024-01-23T21:25:41.064223Z] Info : Opened connection socket for ep#12
[2024-01-23T21:25:41.067195Z] Info : Endpoint socket #12: Client connected. 1 connection(s)
Reusing socket from previo
```

wastez, 2024-01-23:
Something bring the multiprotocol to crash and restart the port. 
I don't use it so I don't know all about it it. 
It has to have a debug mode too, this logs are to less to get the reasons why this happens.

RichardPar, 2024-01-23:
> Something bring the multiprotocol to crash and restart the port. I don't use it so I don't know all about it it. It has to have a debug mode too, this logs are to less to get the reasons why this happens.

Why when I restart Z2MQTT it all works fine? (Multiprotocol is not restarted)   I have reproduced on 3 differenent installes too.  
I can restatart Multiprotocol Addon and z2mqtt wont connect back first load...  I dont know which side it is, but whichever it has made Z2MQTT sadly very unstable.

Nerivec, 2024-01-24:
@RichardPar Can you try again once [0.33.2](https://github.com/Koenkk/zigbee-herdsman/pull/887) is available to update and let me know if the behavior is better? If it's an underlaying issue with socket or silabs, of course, that won't fix the problem, but at least Z2M should handle the fail properly now. Well, hopefully anyway...

RichardPar, 2024-01-25:
> @RichardPar Can you try again once [0.33.2](https://github.com/Koenkk/zigbee-herdsman/pull/887) is available to update and let me know if the behavior is better? If it's an underlaying issue with [...]

Installed and ran+connectesd first time ... I will put it through more paces and see how it purrs.

closed 2024-08-24 by github-actions[bot]