Koenkk/zigbee2mqtt #20831: [New device support]: _TZE200_lndsb16m Bingoelec thermostat

RoGu777, 2024-01-15:
### Link

https://de.aliexpress.com/item/1005006165678941.html

### Database entry

{         "applicationVersion": 70,         "dateCode": "",         "friendlyName": "Thermostat Küche",         "hardwareVersion": 1,         "ieeeAddr": "0xa4c138c3c7464c4a",         "manufacturerID": 4417,         "manufacturerName": "_TZE200_lndsb16m",         "model": "bingoelec_thermostat",         "networkAddress": 41921,         "powerSource": "Mains (single phase)",         "stackVersion": 0,         "type": "Router",         "zclVersion": 3     }

### Comments

Dear all,

I've purchased the Bingoelec thermostat, but it is currently not supportet.

With the help of #17962 I was able to define basic settings, which work fine so far.

Still to be finalized:
- say of the week is shown wrong on the device (Friday instead of Monday)
- program schedule can't be set
- calibration & limits (low priority, as this can be set directly on the device)

Please feel free to contribute to improve the integration.

### External defintion

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =
{
    fingerprint: [ {
        modelID: 'TS0601',
        manufacturerName: '_TZE200_lndsb16m',
    },
    ],
    model: 'bingoelec_thermostat',
    vendor: 'Bingoelec',
    description: 'TS0601 series thermostat',

    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({ timeStart: '2000' }),
    configure: tuya.configureMagicPacket,
    exposes: [
        e.child_lock(),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withRunningState(['idle', 'heat'], ea.STATE)
    ],
    meta: {
        tuyaDatapoints:
            [
                [1, 'system_mode', tuya.valueConverterBasic.lookup({'heat': true, 'off': false})],
                [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
                [3, 'local_temperature', tuya.valueConverter.divideBy10],
                [9, 'child_lock', tuya.valueConverter.lockUnlock],
                [16, 'current_heating_setpoint', tuya.valueConverter.raw],
                [102, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(0), 'idle': tuya.enum(1)})],
            ],
    },
};

module.exports = definition;
```

dadock, 2024-01-21:
Thanks for working on this!

> device is sending a 'commandMcuSyncTime' message every 10 seconds, no idea how to handle that

I‘ve tried
`fromZigbee: [tuya.fz.datapoints, fz.ignore_tuya_set_time], `

inspired by https://github.com/Koenkk/zigbee2mqtt/issues/12225

dadock, 2024-01-23:
> Still to be finalized:
> day of the week is shown wrong on the device (Friday instead of Monday)

Hmm... I was wondering where the on the display the day of the week is being displayed. On mine, I can only see the time (no date, no day etc.)
> program schedule can't be set

Same as above: Is it possible that Date/Time/week profile/schedule etc. are functions of the Tuya bridge, not of the device itself? Also, the device doesn't have any setup functions to set date, schedule etc., see the table I posted on the other thread

> device is sending a 'commandMcuSyncTime' message every 10 seconds, no idea how to handle that

I stopped this with the addition I've mentioned in the post above (fz.ignore_tuya_set_time). Perhaps sb. could tell us whether this is "permitted" or rather a brutal way to get rid of the message. The method mentioned in the tutorial

`onEvent:` tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'

didn't work for me.
> calibration & limits (low priority, as this can be set directly on the device)

I've added limits; I'm not sure whether calibration is exposed. As far as I've understood the tutorials up to now, if there are no error messages as "No converter available for xyz" anymore, there are no more data points beeing sent, or am I wrong?
This is what I'm using right now, and it seems to be working ok. Beside introducing limits, I made a couple of other changes aiming at its use with [fhem](http:www.fhem.de):
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const definition =
{
    fingerprint: [ {
        modelID: 'TS0601',
        manufacturerName: '_TZE200_lndsb16m',
    },
    ],
    model: 'bingoelec_thermostat',
    vendor: 'Bingoelec',
    description: 'TS0601 series thermostat',

    fromZigbee: [tuya.fz.datapoints, fz.ignore_tuya_set_time],

    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({ timeStart: '2000' }),
    configure: tuya.configureMagicPacket,
    exposes: [
	e.binary('system_mode', ea.STATE_SET, 'on', 'off').withDescription('Turn the thermostat on/off'),
        e.child_lock(),
        e.climate()
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withRunningState(['idle', 'heat'], ea.STATE),
	e.min_temperature().withValueMin(5).withValueMax(15).withDescription('minimum setpoint'),
	e.max_temperature().withValueMin(15).withValueMax(45).withDescription('maximum setpoint'),

    ],
    meta: {
        tuyaDatapoints:
            [
		[1, 'system_mode', tuya.valueConverterBasic.lookup({'on': true, 'off': false})],
                [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
                [3, 'local_temperature', tuya.valueConverter.divideBy10],
                [9, 'child_lock', tuya.valueConverter.lockUnlock],
		[15, 'max_temperature', tuya.valueConverter.raw],
		[16, 'min_temperature', tuya.valueConverter.raw],
		[19, 'local_temperature_calibration', tuya.valueConverter.localTemperatureCalibration],
                [102, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(0), 'idle': tuya.enum(1)})],
            ],
    },
};

module.exports = definition;
```
Regards

RoGu777, 2024-01-23:
Hi,

I believe the schedule should be on the device, as I've seen pictures, where the period number is shown in the lower part of the display.

I've now monitored the thermostat for a couple of days and I am worried, that it is not actually working correct, see this graph from HA:

[image]

It looks like once heating is on, the device is getting hot and it impacts the measured temperature as it immediately getting higher. As you see in the graph the delta between heating on & of is around 2° without the room temperature is changed.

Could you please check your device, if it behaves the same?

dadock, 2024-01-24:
Hi,

> As you see in the graph the delta between heating on & of is around 2° without the room temperature is changed.

Perhaps it's a feature, not a bug. Have a look at the table (transcription of the original "manual" that came with my device) at https://github.com/Koenkk/zigbee2mqtt/issues/17962#issuecomment-1833934434 - it seems that the warming of the device is normal and can be taken into account via settings, being the default the 2 degrees you are measuring:

> 05 | Start up temperature difference | 1~5°C | 2 -- | 

As I bought it not for controlling a heater, but for triggering a fan to cool down a departement where I run a Laser proyector when certain temperature is reached, I'm uncertain whether it fits my purpose. I was wondering whether there are more data points; as far as I would expect, there should be at least an option to switch the contact to "permanently on". As I don't have a Tuya gateway, I'll have to wait until sb. discovers "the truth" ;-)

RoGu777, 2024-01-24:
> Perhaps it's a feature, not a bug. Have a look at the table (transcription of the original "manual" that came with my device) at [#17962 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/17962#issuecomment-1833934434) - it seems that the warming of the device is normal and can be taken into account via settings, being the default the 2 degrees you are measuring:
> 
> > 05 | Start up temperature difference | 1~5°C | 2 -- |
>

no this is the so called "dead zone" - the temperature difference to start heating. e.g. if it is set to 1° and target temperature is 23° - the heating will start at 22°

dadock, 2024-01-31:
I've ordered a tuya Gateway. No sense in guessing data points etc. ...

closed 2024-08-29 by github-actions[bot]

emolenev, 2025-03-09:
Anyone could solve the problems?