Koenkk/zigbee2mqtt #20834: New Logging Level

22nra, 2024-01-16:
### Is your feature request related to a problem? Please describe

Looking back at previous posts I've seen people with similar problems to myself and think a new feature is maybe the easiest way to solve our problems.
I'm now retired and been playing with z2m for years and been using an old 1.17.1 build of z2m.
I use the "info" log level output from z2m to build data into my own hobby sensor system.
I've read on the forum that I am not alone and others like myself enjoy using z2m to help populate their hobby sensor projects.

When I used zigbee2mqtt version 1.17.1 I got details of all sensors being logged as shown below:
Wed Jan 10 10:36, Long-Id:, Id:zigbee2mqtt, Type:0xa4c138f86e040736, 1:applicationVersion,70 2:friendlyName,0xa4c138f86e040736 3:hardwareVersion,1 4:ieeeAddr,0xa4c138f86e040736 5:manufacturerID,4417 6:manufacturerName,_TZ3000_dowj6gyi 7:model,TS0201 8:networkAddress,14554 9:powerSource,Battery 10:stackVersion,0 11:type,EndDevice 12:zclVersion,3 13:humidity,75.22 14:linkquality,109 15:temperature,5.83

The problem is that when I upgraded to the latest version 1.35.1 the "info" logged sensor details have been trimmed back to a skeleton record with minimal details being logged:
Zigbee2MQTT:info 2024-01-14 11:09:31: MQTT publish: topic 'zigbee2mqtt/0xa4c138f86e040736', payload '{"battery":90,"humidity":51.33,"linkquality":5,"temperature":18.5}'

### Describe the solution you'd like

Would it be possible to create a new logging level option of "detail".
The "detail" option would populate the log files with full sensor details.
This would allow z2m to be used by low tech developers like myself for their hobby projects.

### Describe alternatives you've considered

I noticed that when I run a "map refresh" from the frontend it runs a "networkscan IncludeRoutes false" and all the devices are populated in the log with all the data inc networkAddress and model which I can then store and use in my hobby system.

[],"friendlyName":"Coordinator","ieeeAddr":"0x00124b000e8c3ec4","lastSeen":1705355163656,"networkAddress":0,"type":"Coordinator"},{"definition":{"description":"Temperature & humidity sensor","model":"IH-K009","supports":"battery, temperature, humidity, voltage, linkquality","vendor":"TuYa"},"friendlyName":"0xa4c138f86e040736","ieeeAddr":"0xa4c138f86e040736","lastSeen":1705355058545,"manufacturerName":"_TZ3000_dowj6gyi","modelID":"TS0201","networkAddress":26396,"type":"EndDevice"},{"definition":{"description":"Door sensor","model":"TS0203","supports":"contact, battery_low, battery, voltage, tamper, linkquality","vendor":"TuYa"},"friendlyName":"0xa4c1382fed7a7666","ieeeAddr":"0xa4c1382fed7a7666","lastSeen":1705348359412,"manufacturerName":"_TZ3000_oxslv1c9","modelID":"TS0203","networkAddress":53730,"type":"EndDevice"},{"definition":{"description":"Motion sensor","model":"809WZT","supports":"occupancy, battery_low, battery, voltage, tamper, linkquality","vendor":"TuYa"},"friendlyName":"0xa4c13814968da3c6","ieeeAddr":"0xa4c13814968da3c6","lastSeen":1705354105087,"manufacturerName":"_TZ3040_bb6xaihh","modelID":"TS0202","networkAddress":55945,"type":"EndDevice"}]}},"status":"ok","transaction":"eckpc-9"}

I have been unable to confirm the exact syntax to run a networkscan that will force z2m to publish all sensor details in the log.

From what I have read the command will look something like below:
mosquitto_pub -t zigbee2mqtt/bridge/networkmap -m '{"type":raw,"routes":false}'

The network scan would only be used if I detected a new sensor to get the new sensor details like model and networkAddress.

Running a network scan is heavy on resources and I still feel a new logging level of "detail" is a better solution to my problem.

Thanks in advance, hope someone is able to help.

### Additional context

N/A

sjorge, 2024-01-16:
Why not collect the date by subscribing to the bridge and device topics over mqtt? That's how I do it for my scraper that pushes data to influxdb.

22nra, 2024-01-16:
Wow, many thanks @sjorge , I am on the learning curve :)
I just came across mosquitto_sub -t 'zigbee2mqtt/bridge/devices'
This gives me all the sensor device details and it gets updated whenever a device leaves or joins.
Thanks again. I'll leave my request open and see if anyone would get benefit from my suggestion.

sjorge, 2024-01-16:
You can be 'smart' about what you subscribe too also, I grab all device data, filter the devices I want and subscribe to their topics.

This works well too if you for example change a friendly name (I target on IEEE adress), you just need to unsub/sub to the new topic once the devices topic gets a republish.

closed 2024-08-14 by github-actions[bot]