Koenkk/zigbee2mqtt #20841: LDSENK02S DonT display Voltage or Current

Martin82sf, 2024-01-16:
### What happened?

Hi.

The model LDSENK02S

Cant measure either voltage or Current.

`Publish 'get' 'voltage' to 'Enchufe TV' failed: 'Error: Read 0x5cc7c1fffe07e5fc/1 haElectricalMeasurement(["rmsVoltage"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

`Publish 'get' 'current' to 'Enchufe TV' failed: 'Error: Read 0x5cc7c1fffe07e5fc/1 haElectricalMeasurement(["rmsCurrent"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

I have two of them and this happens in both 

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

20221102

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Add ON Home assistant ON a NUC

xabylr, 2024-01-19:
It looks like it broke when the converter was modified because of a refactor. If I create a custom converter using the previous code, it works again!

https://github.com/Koenkk/zigbee-herdsman-converters/commit/5f3378765bdf1de5327a4859164fcee68dd0794e#diff-3e7c29010594010cba39762df3fec6fc1fb0bf958c48679cf19cf508cd8c59a2L309-L320

@Martin82sf In the meantime you can do the same. You only have to create a file `LDSENK02S.js` next to `configuration.yaml` with the following content:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['LDSENK02S'],
        model: 'LDSENK02S',
        vendor: 'ADEO',
        description: 'ENKI LEXMAN 16A EU smart plug',
        extend: extend.switch({exposes: [e.power(), e.energy()], fromZigbee: [fz.electrical_measurement, fz.metering]}),
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.onOff(endpoint);
            await reporting.activePower(endpoint);
            await reporting.currentSummDelivered(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
        },
    }
];
```

Then edit the `configuration.yaml` file to have an entry like this:

```yaml
external_converters:
  - LDSENK02S.js
```

Finally restart Zigbee2MQTT and you should get your metrics for the device.

Martin82sf, 2024-01-19:
@xabylr Thanks but if not working .

Not don't appear in the dashboard the options 

[image]

And continues reporting "null" in the voltage / current values

[image]

xabylr, 2024-01-19:
Yes, you are right, for me they don't appear as well. I haven't noticed that. 
However, for me it was an improvement because I didn't even have power or energy before that.

Martin82sf, 2024-03-02:
Still fails in lastest version [1.36.0]

chiva, 2024-06-22:
More in-depth analysis is being done here:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/7147

We can close this one to avoid doubling efforts

closed 2025-01-07 by github-actions[bot]