Koenkk/zigbee2mqtt #20842: Legrand error since 1.35.1

steve41200, 2024-01-16:
### What happened?

I run z2M on add on in HA

Version de Zigbee2MQTT
[1.35.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.1) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Type de coordinateur
zStack3x0 (CC1352P2)
Révision du coordinateur
20230507

device: Legrand [412170](https://www.zigbee2mqtt.io/devices/412170.html#legrand-412170)

Sometimes the teleruptor legrand did'nt respond (in HA or in Z2M) but when you click on it in HA he light on in realife. But don't respond in z2m or HA.I see in log 
Publish 'set' 'state' to 'lumières allée jardin' failed: 'Error: Command 0x000474000085569a/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42668 - 1 - 141 - 6 - 11 after 10000ms)'

I cannot light off then from HA or z2M but with my automation from node red it works ...

Sunday it works like this for 2 of my 3 devices legrand and today 1 of the 3 ...

And in z2m the last seen is 7H, just after the warning message and the light is lighthning in reality ....

[image]

Have you a tips ?

Thank you

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

CC1352P2

### Setup

Add on HA in a VM Proxmox on a NUC

### Debug log

Publish 'set' 'state' to 'lumières allée jardin' failed: 'Error: Command 0x000474000085569a/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42668 - 1 - 141 - 6 - 11 after 10000ms)'

steve41200, 2024-01-16:
When I try to reconfigure the device from z2M UI I have this error
`51Failed to configure 'lumières allée jardin', attempt 1 (Error: Bind 0x000474000085569a/1 genIdentify from '0x00124b001ca1aed9/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))`

steve41200, 2024-01-16:
another test.
Same error with 1 number of difference
`2024-01-16 21:22:33Publish 'set' 'state' to 'lumières allée jardin' failed: 'Error: Command 0x000474000085569a/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42668 - 1 - 79 - 6 - 11 after 10000ms)'
Error 2024-01-16 21:22:56Publish 'set' 'state' to 'lumières allée jardin' failed: 'Error: Command 0x000474000085569a/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42668 - 1 - 82 - 6 - 11 after 10000ms)'
Error 2024-01-16 21:23:29Publish 'set' 'state' to 'lumières allée jardin' failed: 'Error: Command 0x000474000085569a/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 42668 - 1 - 90 - 6 - 11 after 10000ms)'`

and another try to reconfigure
`Error 2024-01-16 21:26:07Failed to configure 'lumières allée jardin', attempt 2 (Error: Bind 0x000474000085569a/1 genIdentify from '0x00124b001ca1aed9/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))`

the other devices works fine

sjorge, 2024-03-16:
This is probably related to https://github.com/Koenkk/zigbee-herdsman/issues/979, until this is fixed you can theoretically manually comment the readResponse coming from inside zh. I've tried this and then the device pairs again.

closed 2024-10-14 by github-actions[bot]