Koenkk/zigbee2mqtt #20851: Send homeassistant configuration topic again

SimonBard, 2024-01-17:
### What happened?

I use mosquitto, zigbee2mqtt and homeassistant in docker container, but that should not matter. 
Problem: The zigbee2mqtt devices only partially appear in homeassistant. 
Really strange, sometimes the tradfi switch appears, sometimes the tradfi controller. 

[image]
[image]

### What did you expect to happen?

All devices from zigbee2mqtt should appear in homeassistant. 

Can I somehow invoke resending the discovery topics for homeassistant?
I had a look at the code, but cannot figure out how to do it. 

### How to reproduce it (minimal and precise)

I restarted the container several times. Reconfigures the devices from Tradfi again. 
They appear in zigbee2mqtt, the mqtt message is sent out, but they do not appear in homeassistant. 

The tradfi controller sometimes appears, but shows no entities:
[image]

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20201127

### Adapter

CC2652

### Setup

docker

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13961847/log.txt)

SimonBard, 2024-01-17:
I deleted the container and configuration and got this in the logs: 

```
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00158d0002730566/battery/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00158d0002730566/temperature/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00158d0002730566/humidity/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00158d0002730566/pressure/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00158d0002730566/voltage/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00158d0002730566/linkquality/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00124b00292c2a70/battery/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00124b00292c2a70/voltage/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00124b00292c2a70/linkquality/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00124b00254e0417/battery/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00124b00254e0417/voltage/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00124b00254e0417/linkquality/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00124b00251f7250/battery/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00124b00251f7250/temperature/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00124b00251f7250/humidity/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00124b00251f7250/voltage/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/sensor/0x00124b00251f7250/linkquality/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/binary_sensor/0x00124b00292c2a70/contact/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/binary_sensor/0x00124b00292c2a70/battery_low/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/binary_sensor/0x00124b00254e0417/contact/config', payload 'null'
Zigbee2MQTT:info  2024-01-17 21:50:56: MQTT publish: topic 'homeassistant/binary_sensor/0x00124b00254e0417/battery_low/config', payload 'null'
```

But still in homeassistant I cannot see anything. Also mqtt explorer does not show the messages, probably because the payload is empty.

mundschenk-at, 2024-01-20:
Null payloads clear the topics if they are retained.

SimonBard, 2024-01-25:
Thanks. I digged a bit deeper. 

When I restart the zigbee2mqtt docker container, I get a lot of config topics, here is an excerpt:
```
homeassistant/sensor/0x00158d0004470aea/linkquality/config {"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"configuratio>
homeassistant/sensor/0xa4c13830a91bd2e7/temperature/config {"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"configuratio>
homeassistant/sensor/0xa4c13830a91bd2e7/humidity/config {"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"configuration_u>
homeassistant/sensor/0xa4c13830a91bd2e7/battery/config {"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"configuration_ur>
homeassistant/sensor/0xa4c13830a91bd2e7/temperature_display_mode/config {"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{>
homeassistant/select/0xa4c13830a91bd2e7/temperature_display_mode/config {"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_t>
homeassistant/sensor/0xa4c13830a91bd2e7/linkquality/config {"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"configuratio>
homeassistant/sensor/0xa4c13830a91bd2e7/update_state/config {"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"configurati>
homeassistant/binary_sensor/0xa4c13830a91bd2e7/update_available/config {"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{">
homeassistant/update/0xa4c13830a91bd2e7/update/config {"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt>
homeassistant/binary_sensor/1221051039810110150109113116116_0x00124b0018e1b395/restart_required/config (null)
homeassistant/button/1221051039810110150109113116116_0x00124b0018e1b395/restart/config (null)
homeassistant/select/1221051039810110150109113116116_0x00124b0018e1b395/log_level/config (null)
homeassistant/sensor/1221051039810110150109113116116_0x00124b0018e1b395/version/config (null)
homeassistant/sensor/1221051039810110150109113116116_0x00124b0018e1b395/coordinator_version/config (null)
homeassistant/sensor/1221051039810110150109113116116_0x00124b0018e1b395/network_map/config (null)
homeassistant/sensor/1221051039810110150109113116116_0x00124b0018e1b395/permit_join_timeout/config (null)
homeassistant/switch/1221051039810110150109113116116_0x00124b0018e1b395/permit_join/config (null)
```

So this means the config topics with the payload are sent out, then same messages again but with payload null. 
**Is that the expected behaviour?**

Strange is that I cannot see the messages in homeassistant. The connection itself works, I get test message sent from another pc to mosquitto, which are received by homeassisant, as seen here:
[image]

But the config topics from the mosquitto logs shown above do not appear in the homeassistant menu seen on the screenshot. 
Is that also expected behaviour? So the function basically filters out config topics?

closed 2024-08-23 by github-actions[bot]