Koenkk/zigbee2mqtt #20857: Supported device not recognized

valentinpx, 2024-01-17:
### What happened?

Hey,
I tried to add my [Moes AM43-0.45/40-ES-EB](https://www.zigbee2mqtt.io/devices/AM43-0.45_40-ES-EB.html) to my devices today.
While the device is successfully added, it is not recognized and I can't control it:
[image]
Here is what is displayed when I'm adding the device:
[image]
[image]
[image]
Is there something I'm missing ? Do you know if it's bug ? Should I use the "support new device" page in order to use it ?

Thanks :)

### Zigbee2MQTT version

 1.35.1

### Adapter firmware version

20210708

### Adapter

Sonoff Universal Zigbee 3.0 USB Dongle

### Setup

Add-on on Home Assistant running on an Odroid-2+

FabriceC01, 2024-01-17:
Hello,
your problem seems similar to mine #20850

Chicken123321, 2024-01-30:
My old AM43 works fine so I bought another recently and have the same problem.  I think they changed something in software

Chicken123321, 2024-01-30:
@valentinpx did you managed to make it work?

valentinpx, 2024-01-30:
@Chicken123321 no I didn't. Maybe we can try [adding a new device](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-add-device-picture-to-zigbee2mqtt-io-documentation). Do you have access to what was supported on your old device ?

valentinpx, 2024-02-24:
Solved by the last update

closed 2024-02-24 by valentinpx