Koenkk/zigbee2mqtt #20863: z2m is reporting update to HA for SNZB-01P even it is not reporting that in its OTA and it is providing wrong versions

Hadatko, 2024-01-18:
### What happened?

In z2m OTA it looks everything ok:
[image]
But in HA:
[image]
[image]

### What did you expect to happen?

No update trigger in HA.

### How to reproduce it (minimal and precise)

Pair SNZB-01P device

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

7.1.1.0 build 273

### Adapter

skyconnect

### Setup

addon in HA

Great-Chart, 2024-01-18:
I have a similar issue that shows similar -1 value presented for 2 devices (although I have multiples of one of them).
These aren't Sonoff items as per above which suggests its not device specific.
No OTA updates visible in Z2M however.
Using Sonoff Dongle-P also with 1.35.1 Z2M
[image]
[image]
[image]

DittmannCz, 2024-01-18:
Any advice?

Koenkk, 2024-01-18:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/20658

closed 2024-01-18 by Koenkk