Koenkk/zigbee2mqtt #20875: [New device support]: Immax NEO Smart outdoor button, IP55, zigbee-Tuya

JJANOUSEK1, 2024-01-18:
### Link

https://www.immax.cz/immax-neo-smart-venkovni-tlacitko-ip55-zigbee-p17655/

### Database entry

{"id":35,"type":"EndDevice","ieeeAddr":"0x70ac08fffe9122d5","nwkAddr":14787,"manufId":4098,"manufName":"_TZ3000_krwtzhfd","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096,57345],"outClusterList":[25,10,3,4,6,8,4096],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1705606400238,"defaultSendRequestWhen":"immediate"}

### Comments

This new button was find but "unsupported", please help me how to use it - insert. tks 

### External defintion

```shell
const {batteryPercentage, onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS004F'],
    model: 'TS004F',
    vendor: '_TZ3000_krwtzhfd',
    description: 'Automatically generated definition',
    extend: [batteryPercentage(), onOff({powerOnBehavior: false}), identify()],
};

module.exports = definition;
```

Tezar, 2024-03-07:
Got hands on this unit today. It seems that's whitelabel of Tyua TS0041. Made it working using custom converter with copy-pasted defintion from tuya.ts
Not sure if the prefered way to add it is to use immax vendor file and some sort of extend (which is already used there, althought not for pure whitelbale products afaik), or if I should just add one more row to the the [tuya definition](https://github.com/Koenkk/zigbee-herdsman-converters/blob/ed7ed68b9d08a50ce0507183b7daa27b031cc07b/src/devices/tuya.ts#L2198)

Also note that this button has 2 modes (not sure if thats the same for other TS004... clones). It's either direct light control messages, or button state ones. You switch between modes with triple click.

Code for the converter in case anyone wants to use it before it shows up in release

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;

const definition =
{
    fingerprint: [
        {
            modelID: 'TS004F',
            manufacturerName: '_TZ3000_krwtzhfd'
    }],
    model: 'NEO Smart',
    vendor: 'Immax',
    description: 'Outdoor switch with 1 button',
    exposes: [e.battery(), e.action(['single', 'double', 'hold'])],
    fromZigbee: [fz.battery, fz.tuya_on_off_action],
    toZigbee: [],
    configure: tuya.configureMagicPacket,
};

module.exports = definition;

```

Koenkk, 2024-03-08:
@Tezar could you create a PR for this?

Tezar, 2024-03-09:
Planning on that, but just as I stated, I am not sure if I should add it under immax vendor or under tuya file as whitelabel. 

Can you please advise what's the preferred approach?

Koenkk, 2024-03-09:
Under Immax

closed 2024-10-06 by github-actions[bot]