Koenkk/zigbee2mqtt #20877: Ikea Vallhorn E2134 illuminance and illuminance_lux randomly get stuck

shrx, 2024-01-18:
### What happened?

Logs from journalctl:

```
Jan 18 10:40:01 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 10:40:01: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26384,"illuminance_lux":435,"linkquality":55,"occupancy":true}'
Jan 18 10:40:11 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 10:40:11: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26374,"illuminance_lux":434,"linkquality":57,"occupancy":true}'
Jan 18 10:40:21 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 10:40:21: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":55,"occupancy":true}'
Jan 18 10:40:31 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 10:40:31: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":57,"occupancy":false}'
Jan 18 11:25:32 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 11:25:32: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":55,"occupancy":false}'
Jan 18 11:40:07 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 11:40:07: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":55,"occupancy":false}'
Jan 18 12:25:18 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 12:25:18: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":55,"occupancy":false}'
Jan 18 12:39:53 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 12:39:53: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":55,"occupancy":false}'
Jan 18 13:25:04 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 13:25:04: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":52,"occupancy":false}'
Jan 18 13:39:39 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 13:39:39: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":55,"occupancy":false}'
Jan 18 14:24:49 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 14:24:49: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":60,"occupancy":false}'
Jan 18 14:39:24 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 14:39:24: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":57,"occupancy":false}'
Jan 18 15:24:34 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 15:24:34: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":55,"occupancy":false}'
Jan 18 15:39:09 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 15:39:09: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":55,"occupancy":false}'
Jan 18 16:24:19 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 16:24:19: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":57,"occupancy":false}'
Jan 18 16:38:54 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 16:38:54: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":57,"occupancy":false}'
Jan 18 17:24:05 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 17:24:05: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":55,"occupancy":false}'
Jan 18 17:38:40 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 17:38:40: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":60,"occupancy":false}'
Jan 18 18:23:51 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 18:23:51: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":60,"occupancy":false}'
Jan 18 18:38:26 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 18:38:26: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":60,"occupancy":false}'
Jan 18 19:23:37 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 19:23:37: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":57,"occupancy":false}'
Jan 18 19:38:12 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 19:38:12: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":57,"occupancy":false}'
Jan 18 20:23:23 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 20:23:23: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":57,"occupancy":false}'
Jan 18 20:37:59 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 20:37:59: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":55,"occupancy":false}'
Jan 18 21:23:10 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 21:23:10: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":26364,"illuminance_lux":433,"linkquality":55,"occupancy":false}'
Jan 18 21:23:11 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 21:23:11: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":60,"occupancy":false}'
Jan 18 21:23:11 raspberrypi npm[1744490]: Zigbee2MQTT:info  2024-01-18 21:23:11: MQTT publish: topic 'zigbee2mqtt/vallhorn', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":60,"occupancy":true}'
```

Not sure if this is an issue with zigbee2mqtt or the device itself.

### What did you expect to happen?

The illuminance measurement should not get stuck.

### How to reproduce it (minimal and precise)

Place the sensor outside in a bright area and observe the published updates.

### Zigbee2MQTT version

    1.35.0 commit: 3d8f0fc3

### Adapter firmware version

20211115

### Adapter

CC2531

### Setup

Raspberry Pi

shrx, 2024-02-07:
> I suspect this could be related to removing the "msOccupancySensing" reporting parameter in the device configuration. I have another Vallhorn device with the default reportining parameters and it doesn't display this issue. I have now reset the parameters on this device as well. Will report whether the reporting returns to normal or not.

I have not observed this behavior lately, so it indeed seems that msOccupancySensing parameter is somehow affecting the illuminance publishing frequency as well.

shrx, 2024-02-08:
@0ip ah, those are the default endpoint settings, so my observations appear to be a red herring.

paulz-gh, 2024-02-28:
I am having the same issue. In addition occupancy also seems stuck on true...
I'm on 1.35.3. Firmware in E2134 is 1.0.57.

thomaslunze, 2024-03-01:
Just had the same issue on 1 of my motion sensors. Reseating the batteries helped for now but a more stable solution would be great

paulz-gh, 2024-03-01:
> Just had the same issue on 1 of my motion sensors. Reseating the batteries helped for now but a more stable solution would be great

My sensor was completely unresponsive even for any button press. So I doubt it's something that can be 'fixed' in Z2M.I suspect that would mean a firmware update  by IKEA is needed.

MattWestb, 2024-03-03:
One OTA update is one the way for Vallhorn but only one minor one so it shall coming in some weeks in the Dirgere feed.

leitma, 2024-07-07:
Has anyone found a solution to this issue?
My Vallhorn sensors freeze every couple of days. After removing the batteries they work fine for a couple of days.

ncodee, 2024-08-23:
> Has anyone found a solution to this issue? My Vallhorn sensors freeze every couple of days. After removing the batteries they work fine for a couple of days.

First, check if you're using the recommended _1.2v (not 1.5v) IKEA LADDA rechargable batteries_, and see if the occupancy and illuminance are reporting correctly in Z2M.

If not, it's more than likely an issue with the sensor itself. I myself had many issues with sensor, and looking to get a replacement. Seems like a common issues with the older batches.

ncodee, 2024-08-23:
> Has anyone found a solution to this issue? My Vallhorn sensors freeze every couple of days. After removing the batteries they work fine for a couple of days.

First, check if you're using the recommended 1.2v (not 1.5v) IKEA LADDA NiMH rechargable batteries, and see if the occupancy and illuminance are reporting correctly in Z2M.

If not, it's more than likely an issue with the sensor or possibly the firmware it. I myself had many issues with sensor, and looking to get a replacement. Seems like a common issues with the older batches.

petrows, 2024-09-17:
> One OTA update is one the way for Vallhorn but only one minor one so it shall coming in some weeks in the Dirgere feed.

What is the "Dirgere feed"?

MattWestb, 2024-09-17:
Tradfri and Dirigera systems have different servers for OTA updates.
Was looking on both and the latest is  production is there so shall getting it  in Z2M (from ZHA Firmware: 0x01000064).

Paring can being on PIA but keep the  device alive the interview is normally working  well and some ZHA user cant getting some devices working OK but i dont having that problems but need taking out the battery dome times for getting then online bbut most bad if have bad not Zigbee 3 routers ass parent then  they can timing out for the network.

closed 2025-01-09 by github-actions[bot]

ncodee, 2025-01-09:
These sensors are a pile of junk from Ikea, will be throwing them away. There doesn't seem to be any new batches/revisions being made. Shame really, as all their other products seems to be fine.